dresden-elektronik/deconz-rest-plugin #3056: Sunricher SRP-ZG9105-50CC (iluminize 511.210T)

MichaelHeimann, 2020-07-19:
## Device

- Product name: Sunricher SRP-ZG9105-50CC (relabeled as iluminize "511.210T" or as "LEDXess EOS10 - ZigBee® - CC Controller+Netzteil 1-Kanal 50W")
- Manufacturer: Sunricher
- Model identifier: DIM Lighning
- Device type : Light

## Screenshots
[image]

[image]
[image]
[image]
On/Off Cluster:
[image]
Level Control Cluster:
[image]
Diagnostics Cluster:
[image]
[image]
OTAU Cluster:
[image]
[image]

These lights seem to work out of the box pretty well.
I have three of these, one bought from Iluminize and the other two from LEDXess.

I'm adding this since I was asked to on discord after asking a question. For whom it may concern, I do have sometimes issues with one of them: The links in deCONZ GUI/VNC disappear and it only reacts to group commands, the node seems gray in deCONZ and Phoscon. This seems to resolve itself but I'd like to know how to enforce the fix/workaround/repair that.
I had this with Paul Neuhaus Lights before...

Hope that helps :)

MichaelHeimann, 2020-08-09:
Push.
The Group Command thing seems to be related to sub-optimal reachability. Signal Meter Plugin gives 111 or 119 sometimes.
Nothing related to the device itself.
What's needed to get support?

Mimiix, 2020-08-10:
I've asked @SwoopX to check on this ;)

SwoopX, 2020-08-11:
I've added Sunricher mains powered devices to allow attribute reporting, maybe that helps. If not, I'd suspect routing issues, which resolve itself over time.

pur3asdf, 2020-09-02:
I'm also looking forward to see Sunricher remotes implemented into phoscon app. I can see my [SR-ZG2868](https://www.sunricher.com/dim-cct-rgb-3-in-1-zigbee-remote-sr-zg2868.html) in deconz but i'm unable to add it as a switch in phoscon app.

Mimiix, 2020-09-03:
@pur3asdf Please read #2684

closed 2020-10-04 by stale[bot]