dresden-elektronik/deconz-rest-plugin #3057: Axis Gear Zigbee Shade is not supported

clement-chung, 2020-07-19:
Appreciated if deconz can support Axis Gear Zigbee shade device.
Node Info
[image]

Smanar, 2020-07-19:
Hello, when you have created the issue, you haven't see a part called "Device Request" ?

We need some more information > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

seafoodrice, 2020-07-19:
Thanks for provide the doc for requesting device support page. I have captured all available cluster info as below. Please let me know if I can assist in anyway. Thanks in advance.

Basic Cluster Info
[image]

Windows Covering Cluster Info 1
[image]

Windows Covering Cluster Info 2
[image]

Windows Covering Cluster Info 3
[image]

Poll Control Cluster Info
[image]

Power Config Cluster
[image]

OTAU
[image]

Mimiix, 2020-07-20:
@seafoodrice I am really tempted on closing this issue right away and having you to open a new one, following the correct format as stated in the Issue template.  

I'll ask @Smanar to confirm if the above screenshots are okay. If he say's they are not, i'm going to close this and ask you to open a [Device Request](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=Device+Request&template=device-request.md&title=Device+name) 
This will automatically assign a label too.

seafoodrice, 2020-07-20:
Hi Dennis,

Thanks for your reply. If tagging a device request label is not able to be done manually due to any workflow limitation in github, I am happy to open a new device request for that.

I would also suggest to have an issue workflow implemented so that to avoid the request device support process is not being executed. Since the request device support information is available under mid part of wiki.
Maybe you can add the common process at the beginning of the readme of the repo.

Let me know if anything I can assist on information provided regarding adding support on the requested device.

Thanks in advance.

Mimiix, 2020-07-20:
@seafoodrice Hi! 
Thanks for your reply.  We have enabled issue templates upon creating items. 

Do you have more info on that issue workflows? It might be something helpful for us :)

Smanar, 2020-07-20:
I m seing your device support level control AND window covering.

Do you know wich one is used to make it moving ? You can make tries using deconz, if you select cluster 0008 or 0102 you will have some command to test.

Your device is not just an up/down device, but support position control ?

seafoodrice, 2020-07-21:
Hi Smanar,

That's correct, it is windows shade cover switch which supports different level of shading.
I will try to test out the command later today and get back to you with some info hopefully.

thanks Smanar.

Smanar, 2020-07-22:
Tell me if you have problem to try command, I just need to be sure your device use window covering device and not the level control one.

seafoodrice, 2020-07-24:
Hi Smanar,

Sorry for late reply, having a busy week at work. Got sometime to try our the command and as follow. Let me know if you need more details info.

Cluster 102 info
[cid:01a6160c-b013-440b-a9fb-809ddb3d65b8]
[cid:446e0587-fb4f-48a8-be82-963a973a66f6]
[cid:73771b7b-90c6-45df-bc91-293b97988ea7]
[cid:795847f4-f915-480b-a616-f93927ee536b]

Command
Up/Open: shade up - works
Down/Close: Shade down - works
Lift Value: go to specific position, but don't know the full range of integer value supported
Lift percentage: go to specific position, however percentage of the position is not quite right at all.
Tile Value/Percentage: UNSUP_CLUSTER_COMMAND

seafoodrice, 2020-07-24:
Level Control Cluster
[cid:e7dab559-119e-457b-95d3-3315a35b457e]

[cid:a2caa49e-d81d-4bad-bc34-cd8823aafc0f]
all commands response as is what's the command names as. Transition time and rate has no effect as expected. However,

I am not able to figure out the level and step size range reflected to the shade level quite yet.

Hope this helps.
________________________________
From: Clement .. <clement624@hotmail.com>
Sent: Friday, 24 July 2020 6:07 PM
To: dresden-elektronik/deconz-rest-plugin <reply@reply.github.com>
Subject: Re: [dresden-elektronik/deconz-rest-plugin] Axis Gear Zigbee Shade is not supported (#3057)

Hi Smanar,

Sorry for late reply, having a busy week at work. Got sometime to try our the command and as follow. Let me know if you need more details info.

Cluster 102 info
[cid:01a6160c-b013-440b-a9fb-809ddb3d65b8]
[cid:446e0587-fb4f-48a8-be82-963a973a66f6]
[cid:73771b7b-90c6-45df-bc91-293b97988ea7]
[cid:795847f4-f915-480b-a616-f93927ee536b]

Command
Up/Open: shade up - works
Down/Close: Shade down - works
Lift Value: go to specific position, but don't know the full range of integer value supported
Lift percentage: go to specific position, however percentage of the position is not quite right at all.
Tile Value/Percentage: UNSUP_CLUSTER_COMMAND

seafoodrice, 2020-07-24:
@Smanar , I replied via email, somehow the screenshot are not showing up on the thread. :(
Let me know if I should resend the screen shots....

Smanar, 2020-07-24:
No it s good I don't need capture, don't worry.
So you device use window coivering cluster without something special, but I don't undersand what the level cluster is for ?

>all commands response as is what's the command names as 

I don't understand this phrase

seafoodrice, 2020-07-24:
I am not sure the difference between the clusters, they behave the same to me on controlling the shade level and open/close command.

all commands response as is what's the command names as - what I mean is all command behave as expected except the tilt is not supported. Also the leveling input value not yet figure out the value range corresponding to the shade level..

Smanar, 2020-07-24:
Ok so first try:
- added the device in whitelist
- make it reconised as light
- disable reporting for all clusters except windows covering

To compile it replace the step 1 by
git clone --banch axis https://github.com/Smanar/deconz-rest-plugin.git

(and you can ignore "git checkout -b mybranch HEAD")

seafoodrice, 2020-07-25:
Hi Smanar, 
Thanks for reply.

I am using Hassio Deconz add-on, and I afraid I wont't be able to the step mentioned above. I can try to compile branch build using Win 10 on my laptop instead.
For compile the branch in step 3- qmake && make -j2, what should I run if I want to compile on Win 10?

Smanar, 2020-07-25:
Har, so we have a problem ^^.
You can compile only on linux machine.

seafoodrice, 2020-07-27:
Not sure if any member who has the device can help out. Seems I have to setup another Pi for this purpose.....

Mimiix, 2020-07-27:
@seafoodrice If you have another SD card around, that is a good option too 👍

seafoodrice, 2020-08-01:
@Smanar, was too busy at work and unable to get back to this till now. While I am getting a pi to test the branch build. Can you
provide some guidance on these as I am new?

- added the device in whitelist
- make it reconised as light
- disable reporting for all clusters except windows covering

For the app, as I was using HA, I believe I can use postman to send request and test out the api? Am I on correct path?

> Ok so first try:
> 
> * added the device in whitelist
> * make it reconised as light
> * disable reporting for all clusters except windows covering
> 
> To compile it replace the step 1 by
> git clone --banch axis https://github.com/Smanar/deconz-rest-plugin.git
> 
> (and you can ignore "git checkout -b mybranch HEAD")
> 
> I haven't asked, what are you using as app with deconz ? Else you have api command here #3063

Smanar, 2020-08-01:
Ok as you are using HA, better to use the second raspberry, and if you are using a test environement use a Desktop version.

All was done by the code (If I m right) you just need to install deconz on the raspberry, like a normal installation and follow the instruction on github page

https://github.com/dresden-elektronik/deconz-rest-plugin at "Install deCONZ development package (optional, Linux only)", just follow the step, and replace the step 1 by
>git clone --banch axis https://github.com/Smanar/deconz-rest-plugin.git

You can skip too "git checkout -b mybranch HEAD"

After that just include the device on deconz.

If you want to use your old network in same time, you can use the Phoscon feature in "gateway", to make a backup/restore to clone your previous network on the new one.

Smanar, 2020-08-01:
```
"5": {
        "etag": "87269755b9b3a046485fdae8d96b252c",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-08-01T16:22:05Z",
        "manufacturername": "AXIS",
        "modelid": "Gear",
        "name": "Window covering device 5",
        "state": {
            "bri": 0,
            "lift": 0,
            "on": false,
            "open": false,
            "reachable": true
        },
        "swversion": "100-5.3.5.1122",
        "type": "Window covering device",
        "uniqueid": "00:24:46:00:00:12:48:df-01"
    }
```

open the device
`curl http://192.168.1.182/api/8C2FF47893/lights/5/state -X PUT -d '{"open": true}'`
close the device
`curl http://192.168.1.182/api/8C2FF47893/lights/5/state -X PUT -d '{"open": false}'`
stop the device
`curl http://192.168.1.182/api/8C2FF47893/lights/5/state -X PUT -d '{"stop": true}'`
set a level (50%)
`curl http://192.168.1.182/api/8C2FF47893/lights/5/state -X PUT -d '{"lift": 127}'`

Need to test too state return.
IDK if this device can be used with a command, or with your smartphone using BT, but if you use it, the value in deconz need to be updated (in the Json)
Same in the Json the field "lift" need to be updated, other values too ofc.

To get device JSON
`http://192.168.1.182/api/8C2FF47893/lights/5`

for the battery I need the cluster 0x0001 like previous, but you haven't pressed the "read" button" (and make the device awake), because all values are empty.

ebaauw, 2020-08-01:
`lift` should run from 0 to 100 (%).

seafoodrice, 2020-08-02:
Please see the power config cluster as below. thanks.
[image]

seafoodrice, 2020-08-02:
> ```
> "5": {
>         "etag": "87269755b9b3a046485fdae8d96b252c",
>         "hascolor": false,
>         "lastannounced": null,
>         "lastseen": "2020-08-01T16:22:05Z",
>         "manufacturername": "AXIS",
>         "modelid": "Gear",
>         "name": "Window covering device 5",
>         "state": {
>             "bri": 0,
>             "lift": 0,
>             "on": false,
>             "open": false,
>             "reachable": true
>         },
>         "swversion": "100-5.3.5.1122",
>         "type": "Window covering device",
>         "uniqueid": "00:24:46:00:00:12:48:df-01"
>     }
> ```
> 
> open the device
> `curl http://192.168.1.182/api/8C2FF47893/lights/5/state -X PUT -d '{"open": true}'`
> close the device
> `curl http://192.168.1.182/api/8C2FF47893/lights/5/state -X PUT -d '{"open": false}'`
> stop the device
> `curl http://192.168.1.182/api/8C2FF47893/lights/5/state -X PUT -d '{"stop": true}'`
> set a level (50%)
> `curl http://192.168.1.182/api/8C2FF47893/lights/5/state -X PUT -d '{"lift": 50}'`
> 
> Need to test too state return.
> IDK if this device can be used with a command, or with your smartphone using BT, but if you use it, the value in deconz need to be updated (in the Json)
> Same in the Json the field "lift" need to be updated, other values too ofc.
> 
> To get device JSON
> `http://192.168.1.182/api/8C2FF47893/lights/5`
> 
> for the battery I need the cluster 0x0001 like previous, but you haven't pressed the "read" button" (and make the device awake), because all values are empty.

Test findings:

- state change 'put' api lights/5/state
body: {"on":true} - it closed the shade, supposed to open the shade?
body: {"on":false} - it opened the shade, supposed to close the shade?

body: {"lift": 100} - fully closed the shade, should it be the other way around?
body: {"lift": 80} - closed 80% of the shade 
body: {"lift": 25} - closed 25% of the shade

body: {"stop":true} - worked as expected.

- State return: The json state attributes does NOT change (all values remained the same) after I sent different commands to operate the shade (open/close/lift).
"state": {
            "bri": 0,
            "lift": 0,
            "on": false,
            "open": false,
            "reachable": true
        },

Thanks.

seafoodrice, 2020-08-02:
missing 2 test cases for state changes.

state change 'put' api lights/5/state
body: {"open":true} - working as expected
body: {"open":false} - working as expected

ebaauw, 2020-08-02:
Best not use `on` nor `bri`; they’ve been deprecated and will be removed.

The API plugin needs to setup attribute reporting for _Current Position Lift Percentage_.  Do the resource attributes update when you read this Zigbee attribute in the GUI?

There’s no way to expose the battery on a `/lights` resource, so this needs to be expose as a separate ZHABattery `/sensors` resource, cf the IKEA FYRTUR and KADRILJ.

Smanar, 2020-08-02:
I have added the ZHABattery  for battery, to use the use tha attribute 0x0021.

For reporting, I have updated the code, removed a part where I m ignoring the cluster 0008 to use only the 0102.

If it still don't work, as have said ebaauw, move your shutter at 50%, and on deconz check if the value are updated in those both cluster. If yes, can you check if the reporting and binding is working using deconz too ?

Mimiix, 2020-11-17:
Re-opened on request.

seafoodrice, 2020-11-17:
Once put the blind to 80%, the attribute does not get updated and stayed at 100 (closed).

[image]

Only after I did a force read on deconz, the attribute then got updated.
[image]

FYI, I have already done the binding with cluster 12 of the windows device to the deconz.
[image]

SwoopX, 2020-11-17:
How does the attribute reporting for 0x0008 look like? Double click on the value and then `read config`.

seafoodrice, 2020-11-17:
read config returns NOT_FOUND, if press on read it returns 80.
[image]

Smanar, 2020-11-17:
BTW the last return from websocket are
```
2020-11-16 17:28:25 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"7","r":"lights","state":{"bri":254,"lift":100,"on":true,"reachable":true},"t":"event","uniqueid":"00:24:46:00:00:12:48:df-01"} <--- this is fully closed
```
```
2020-11-16 17:30:07 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"7","r":"lights","state":{"bri":205,"lift":81,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"00:24:46:00:00:12:48:df-01"} <-- this is when the shade is opened
```

So open=false don't appear on first return but it seem the field "open" is working.
On the code the "on" value depend of "bri", need to have "bri" = 0 to have "on" = false.

It s like if you send "open" or "close" command the "bri" values is not updated.

seafoodrice, 2020-11-17:
As I mentioned in discord, there are 2 issues so far.
1. the state and attributes do not update automatically until do a read on device cluster in phoscon
2. once read was pressed and the value of state "on" always returns true, no matter the cover is opened or not.

Smanar, 2020-11-17:
For the 1 the state is updated on fiield "open" no ?

You have open = true when you have open it
And nothing when you have closed it, but I think it s because the state was unchanged.

seafoodrice, 2020-11-17:
No, the HA integration is mapped the state with “on”, but not “open” attribute.

Smanar, 2020-11-17:
Sure but we can do "magic" in deconz code ^^.
If you can confirm the field "open" is good (without force read attribute) when checking the device JSON, it can help us.

seafoodrice, 2020-11-17:
Unfortunately after sending command to the blind, as related to issue 1, force read is required in order to update the open field. Otherwise no field is updated.

Smanar, 2020-11-17:
Can you show again the 2 websockets you have after a command but without "forced read" ?

seafoodrice, 2020-11-18:
There is NO web socket received in HA until a forced read is issued.

Smanar, 2020-11-18:
Ok, sorry, was not sure understand, so we need a return, any of them.
So now it s possible to have log in phoscon (I hope you haven't too much device), can you show us the log after a command (on pastebin clone), or just check if you have something about the device ?
Else you realy need to enable a bind and a reporting from one of thoses attribute

0x0006 : attribute 0x0000
0x0008 : attribute 0x0000
0x0102 : attribute 0x0008 > return NOT_FOUND, so not this one

seafoodrice, 2020-11-19:
As I said earlier, I have binded already and it shown binded successfully..
[image]

After I send command change change the blind position. There is nothing shown up on Phoscon log. ONLY AFTER I did a forced read. The following showed up in the log.

12:24:01:595 binding for cluster 0x0102 of 0x0024460000145410 exists (verified by reporting)
12:24:01:596 skip configure report for cluster: 0x0102 attr: 0x0008 of node 0x0024460000145410 (seems to be active)
12:24:03:523 Bind response success for 0x0024460000145410 ep: 0x01 cluster: 0x0006
12:24:03:525 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x0024460000145410 (wait reading or unsupported)
12:24:03:733 Bind response success for 0x0024460000145410 ep: 0x01 cluster: 0x0008
12:24:03:733 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x0024460000145410 (wait reading or unsupported)

SwoopX, 2020-11-19:
NOT_FOUND does usually mean that attribute reporting is supported, but not yet defined. Otherwise, you'd get UNREPORTABLE_ATTRIBUTE.

Can you please try to configure it with the values 1, 300, 1 and double check again?

seafoodrice, 2020-11-19:
Just to clarify your ask,
[image]
Min Report Interval - 1
Max Report Interval - 300
Reportable Change - 1

is that what you mean?

seafoodrice, 2020-11-19:
once I set the config as above. When I change the blind position, the device reports with following log.
0x0024460000145410 is the device.

23:41:28:532 ZCL attribute report 0x0024460000145410 for cluster: 0x0008, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
23:41:32:458 [INFO] - No button handler for: Gear endpoint: 0x01 cluster: WINDOW_COVERING (0x0102) command: 0x0A payload[0]: 008
23:41:32:459 ZCL attribute report 0x0024460000145410 for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
23:41:34:631 ZCL attribute report 0x00158D0001A4CC99 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
23:41:39:302 [INFO] - No button handler for: Gear endpoint: 0x01 cluster: 0x0001 command: 0x0A payload[0]: 021
23:41:39:304 ZCL attribute report 0x0024460000145410 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
23:41:40:589 [INFO] - No button handler for: Gear endpoint: 0x01 cluster: WINDOW_COVERING (0x0102) command: 0x0A payload[0]: 008

SwoopX, 2020-11-19:
Yes, that's what I meant. So that part shows that the device is not sending the attribute reports. The values should now update every sec.

seafoodrice, 2020-11-19:
Thanks @SwoopX , so if I set Reportable Change to 3 means 3 secs which will be less chatty?

SwoopX, 2020-11-19:
Well, yes and no. Depends on what you really want. Min reporting is the actual frequency upon the defined change value. If nothing changes, max intervall applies.

So, if you arenot interested in 1% changes e.g., you could move up to 10. If you generally only want to have changes reported every 5 secs, you might move up the min interval.

I've set my smart plugs up to 20 sec min reporting, to throttle the signals in the air a bit and reduce the processing load of my poor Rpi zero a bit 😀

seafoodrice, 2020-11-19:
Makes perfect sense. From the log above, if there nothing wrong you can see. I will close the issue.
Thanks @SwoopX and @Smanar 

> 23:41:28:532 ZCL attribute report 0x0024460000145410 for cluster: 0x0008, ep: 0x01, frame control: 0x18, mfcode: 0x0000
> 23:41:32:458 [INFO] - No button handler for: Gear endpoint: 0x01 cluster: [...]

SwoopX, 2020-11-19:
Looks fine to me from here. We could however doublecheck if reporting is correctly enabled in the code. It can happen though that bindings and reporting configs can be missed.

Smanar, 2020-11-19:
I realy don't see problem, the code look fine.

During your tests have you tried to re-include the device ?

>12:24:01:596 skip configure report for cluster: 0x0102 attr: 0x0008 of node 0x0024460000145410 (seems to be active)
>12:24:03:525 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x0024460000145410 (wait reading or unsupported)

SwoopX, 2020-11-19:
@Smanar just realized that the mac address is too long and not correct in the logs 🤔

seafoodrice, 2020-11-19:
I have tried to added few more Axis device to the network and every new device added I have to manually go to deconz to,
- bind the cluster
- write config for 0x0008
Is that something can be fixed in the code?

Smanar, 2020-11-20:
Yes, and it need.
Just to be sure, can you compare thoses values and yours from your last device

Model id = "Gear"
Manufacture number = 0x1262 or 0x109a
Mac adress start by 0x002446

Or if you have the log just during the device inclusion ? (when the bind/reporting need to be done by the code)

seafoodrice, 2020-11-22:
My apologies @Smanar, I am not sure where to look for those values,
Model id = "Gear"
Manufacture number = 0x1262 or 0x109a
Mac adress start by 0x002446

can you provide more information?

Smanar, 2020-11-22:
And your Mac adress is  0x0024460000145410
So all is fine ....

>12:24:03:525 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x0024460000145410 (wait reading or unsupported)

But I don't understand to have this debug line
```
        else if (lightNode && rq.maxInterval != 0xffff /* disable reporting */)
        {
            // wait for value is created via polling
            DBG_Printf(DBG_INFO, "skip configure report for cluster: 0x%04X attr: 0x%04X of node 0x%016llX (wait reading or unsupported)\n",
                       bt.binding.clusterId, rq.attributeId, bt.restNode->address().ext());
        }
```

To have NOT_FOUND as return we need to have rq.maxInterval == 0xffff no ?

Can you try with this version pls ? It have added a debug line, then re-include your device and give us log pls.
`
git clone --bracnh axis https://github.com/Smanar/deconz-rest-plugin.git`

seafoodrice, 2020-11-22:
Hi Smanar, unfortunately my house is hosting few guests for coming 2 weeks and I tried to keep my smart devices stable. Since I have only 1 conbee II. I will have to test it out afterwards.  
If you have just added the debug log, feel free to merge it into MASTER so maybe I just test the log in the next release.

SwoopX, 2020-11-22:
> If you have just added the debug log, feel free to merge it into MASTER so maybe I just test the log in the next release.

Erm, no!

@Smanar maxinterval is initialized with 0 as far as I can tell. As I see it, it could take up to around 30 mins for the binding and reporting to automatically set up. However, if the device is a deep sleeper and either binding or reporting is missed, you end up in the situation experienced.

Smanar, 2020-11-22:
So I m seing to test to do :
- making the device awake during inclusion
- let the device alone with bad reporting during somes days to see if it finally set them.

seafoodrice, 2020-11-22:
@Smanar , to help on the findings, I have following experience as I have 4 devices.
- I have 1 device which has the the config setup properly after an hours or so.
- all others 3 I left it for few days before @SwoopX helped out to write config manually, and never setup properly by itself after added to the network.

SwoopX, 2020-11-22:
Hm, strange. I just double checked the device above and it has a poll control cluster so I'd assume it is a light sleeper then.

[5 more replies]

closed 2020-12-25 by stale[bot]