dresden-elektronik/deconz-rest-plugin #3058: Heiman water leakage sensor status unreachable

seafoodrice, 2020-07-19:
After adding the sensor, the status shown unreachable.
[image]

Environment
[image]

 - Host system: Rasberry PI
 - Running method: Home Assistent deCONZ Add-on
 - Device: ConBee II
 - Do you use an USB extension cable: yes

Please advise. Thanks.

Mimiix, 2020-07-20:
Hi! Can you provide a screenshot of the deCONZ gui (accessible with the VNC viewer, please see HA Addon documentation for that ).

Also, do you see it in HA?

seafoodrice, 2020-07-20:
hi, thanks for your reply.
Screenshot in deCONZ gui

[image]

I can see the water sensor as binary sensor however the state never change when testing by putting some water onto the sensor.
[image]

Mimiix, 2020-07-20:
Seems like it is not turned on. 

What happens if you click the button on the device? Does anything blink on the VNC? 

Otherwise is suggest re-pairing.

seafoodrice, 2020-07-20:
When I clicked on the button on the device, the device led light blink and the corresponding node for the water sensor blinked green as response on GUI.

seafoodrice, 2020-07-20:
I have tried re-pairing the device several times and came back with same unreachable result. I had multiple water sensors with the identical brand and model and they all returned unreachable after connected.

thanks.

Mimiix, 2020-07-20:
It's interesting as it seems that it does connect, but sleeps right after. 

Can you provide some logging? Also, i am wondering how your complete mesh looks like. Do you have routers? How many devices do you have?

Mimiix, 2020-07-20:
On the logging: If you pair it, please check the addon logs , i need those :)

seafoodrice, 2020-07-20:
Thanks for prompt reply. Can you guide me through how to gather the log? I am running Hassio addon.

here is the complete mesh.
[image]

Mimiix, 2020-07-20:
If you open the HA addon page, you have a tab called "log". 

That's what i want to have :)

Also, it would be best to capture that when adding the sensor.

Mimiix, 2020-07-20:
Oh, to speed things up: I'd like to invite you over to our Discord :)  https://discord.gg/vpQbXP

seafoodrice, 2020-07-20:
I removed and add the sensor again. I hope I get the log you want.

17:36:17:399 CTRL skip polling while permit join is set
17:36:18:141 CTRL skip polling while permit join is set
17:36:18:207 DB UPDATE device_descriptors SET data = x'0104010204000500000300000501000900011900', timestamp = 1595230578 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:50:43:c9:03:37:2c:35') AND endpoint = 1 AND type = 4
17:36:18:207 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104010204000500000300000501000900011900', 1595230578 FROM devices WHERE mac = '00:50:43:c9:03:37:2c:35'
17:36:18:225 don't close database yet, keep open for 900 seconds
17:36:18:225 FP indication 0x0000 / 0x8004 (0x005043C903372C35 / 0xC9E3)
17:36:18:225                       ...     (0x005043C903372C35 / 0xC9E3)
17:36:18:225 ZDP indication search sensors 0x005043C903372C35 (0xC9E3) cluster 0x8004
17:36:18:225 ZDP indication search sensors 0x005043C903372C35 (0xC9E3) clear timeout on cluster 0x8004
17:36:18:290 don't use deleted sensor and node 0x005043C903372C35 as candidate
17:36:18:290 [3.1] get IAS Zone type for 0x005043c903372c35
17:36:21:328 CTRL skip polling while permit join is set
17:36:21:500 CTRL skip polling while permit join is set
17:36:21:533 FP indication 0x0104 / 0x0500 (0x005043C903372C35 / 0xC9E3)
17:36:21:533                       ...     (0x005043C903372C35 / 0xC9E3)
17:36:21:533 Clear fast probe timeout for cluster 0x0500, 0x005043C903372C35
17:36:21:596 don't use deleted sensor and node 0x005043C903372C35 as candidate
17:36:21:597 [4] get basic cluster attr 0x0004 for 0x005043c903372c35
17:36:22:428 CTRL skip polling while permit join is set
17:36:22:523 CTRL skip polling while permit join is set
17:36:22:708 CTRL skip polling while permit join is set
17:36:22:744 FP indication 0x0104 / 0x0000 (0x005043C903372C35 / 0xC9E3)
17:36:22:744                       ...     (0x005043C903372C35 / 0xC9E3)
17:36:22:744 Clear fast probe timeout for cluster 0x0000, 0x005043C903372C35
17:36:22:810 don't use deleted sensor and node 0x005043C903372C35 as candidate
17:36:22:810 [4] get basic cluster attr 0x0005 for 0x005043c903372c35
17:36:23:747 CTRL skip polling while permit join is set
17:36:23:951 CTRL skip polling while permit join is set
17:36:23:992 write attribute of 0x005043C903372C35 ep: 0x01 cluster: 0x0500: 0x0010
17:36:23:993 don't close database yet, keep open for 900 seconds
17:36:23:993 SensorNode 24: Water 24 added
17:36:23:994 FP indication 0x0104 / 0x0000 (0x005043C903372C35 / 0xC9E3)
17:36:23:994                       ...     (0x005043C903372C35 / 0xC9E3)
17:36:23:994 Clear fast probe timeout for cluster 0x0000, 0x005043C903372C35
17:36:24:064 discard double entry in binding queue (size: 2) for for 0x005043C903372C35, cluster 0x0001
17:36:24:064 discard double entry in binding queue (size: 2) for for 0x005043C903372C35, cluster 0x0500
17:36:24:145 [4] get basic cluster attr 0x4000 for 0x005043c903372c35
17:36:25:282 MAC poll fastEnddeviceProbe() 0x005043C903372C35
17:36:25:282 wait response fastEnddeviceProbe() 0x005043C903372C35, elapsed 1137 ms
17:36:25:283 discard double entry in binding queue (size: 2) for for 0x005043C903372C35, cluster 0x0001
17:36:25:283 discard double entry in binding queue (size: 2) for for 0x005043C903372C35, cluster 0x0500
17:36:25:283 CTRL skip polling while permit join is set
17:36:25:487 MAC poll fastEnddeviceProbe() 0x005043C903372C35
17:36:25:487 wait response fastEnddeviceProbe() 0x005043C903372C35, elapsed 1342 ms
17:36:25:487 CTRL skip polling while permit join is set
17:36:25:556 FP indication 0x0104 / 0x0500 (0x005043C903372C35 / 0xC9E3)
17:36:25:556                       ...     (0x005043C903372C35 / 0xC9E3)
17:36:25:637 wait response fastEnddeviceProbe() 0x005043C903372C35, elapsed 1492 ms
17:36:25:638 Bind response success for 0x005043c903372c35 ep: 0x01 cluster: 0x0500
17:36:26:188 dev /dev/ttyAMA0
17:36:26:245 GW firmware version: 0x264a0700
17:36:26:245 GW firmware version shall be updated to: 0x26580700
17:36:26:587 MAC poll fastEnddeviceProbe() 0x005043C903372C35

seafoodrice, 2020-08-14:
issue hasn't resolved, keeping the thread active.

YKO-de, 2020-10-14:
Has not yet been solved. I have the same problem.

Mimiix, 2020-10-14:
@YKO-de Please let me re-open next time.

Let me ask devs to see whats up.

Smanar, 2020-10-14:
You have immediatly the state "unreachable" ?

There is somes thing I don't understand
- Sensor need some time before they are marqued as "unreachable"
- If the sensor is marqued as unreachable but if deconz see a request from it, it change the state.

closed 2020-11-24 by stale[bot]