dresden-elektronik/deconz-rest-plugin #3059: TuYa TS0001 valve

renne, 2020-07-19:
## Device

- Product name: TuYa TS0001/CR Smart Home TS0001/Lonsonho TS0001/Bandi BDS03G1
- Manufacturer: Available in zigbee2mqtt
- Model identifier: Available in zigbee2mqtt
- Device type :
  - Other: valve

Mimiix, 2020-07-20:
@renne This is really not how it works unfortunately. We do need the information as per deCONZ screens and it's respectable clusters. Are you able to provide the screenshots as requested?

fi-sch, 2020-07-22:
Did you check if it doesn't work indeed? I have TS0011 and it works perfectly out of the box.

renne, 2020-07-23:
It's still in shipping ...

Mimiix, 2020-07-23:
I'll be closing this. Please create another issue if you do have the device and required information.

closed 2020-07-23 by Mimiix