dresden-elektronik/deconz-rest-plugin #3061: Not all buttons available with Busch Jäger ZLL Switches

dotcom84, 2020-07-20:
## Describe the bug
Busch Jäger has 6 different wall switches in their Zigbee line:
- Battery powered (2 buttons, 4 buttons, 8 buttons)
- 230V powered (2 buttons, 4 buttons, 8 buttons)

The 230V powered ones can be connected to either a power supply unit, where they basically act like the battery powered switches or to a relay unit, where the first row (upper two buttons) directly control the relay and the others (for the 4 and 6 button models) can control other lights (called "scenes").

My issue is that when connecting these 4 and 6 button relay switches to a deCONZ controlled network in some cases not all buttons are available. My definition of available: Pressing them triggers a "button event" in the REST API which can be used for home automation.
In my scenario some (not all) of the 4 button and some of the 8 button models behaved strange: When connected to a relay unit only some of the (lower) button pairs would trigger an "button event", while others just blink three times being pressed. This is normally the default behaviour of the switches when paired directly with other switches when no "scene" is programmed for this button. However I got no clue how to get them to work in the intended mode. Factory reset and re-adding them does not seem to be the solution.

## Steps to reproduce the behavior
Add 4 or 6 brand-new button Busch Jäger ZLL switches to your deCONZ network and hope that one of them shows the symtoms. Definitely not all switches have the problem.

## Expected behavior
All buttons available through the REST-API as button events.

## Environment

 - Host system: Raspberry Pi 4 B (4GB)
 - Running method: Raspbian
 - Firmware version: 2.05.78 / 22.5.2020
 - deCONZ version: 26580700
 - Device: ConBee II
 - Do you use an USB extension cable: yes - for better reception (RasPi is mounted in server rack behind glass)

Mimiix, 2020-08-11:
@manup

manup, 2020-08-12:
@SwoopX had a good hint, it seems a branch in the code might cause this for battery powered switches, we're looking into this.

If my memory servers me right, for the 230V powered relay and dimmer switches the top row doesn't generate button events and serves only as a local control.

SwoopX, 2020-08-24:
I assume I got the root cause for this. Got a new control element and played a bit around. I also get one row that is kinda "not available". Having a look at the sniffer output, it seems as if the binding table is full and the device only supports 4 bindings (4-gang).

So, the first 3 endpoints each got a binding my my case (scenes), but one of them also got a `level control` binding. That being 4 in total, the last endpoint didn't get any binding at all and this one is making "trouble".

[image]
[image]

As I see it, the bindings in the code must and should follow the defined switch type (dimmer or scenes). I'd assume you could mix that since the device exposes 4 `ZHASwitch` resources if you define the mode per resource.

Still gotta test this with some modified code though.

jensflorian, 2020-11-27:
I have a Busch-Jäger 2-gang 230V with power supply unit. The level control binding is also behaving strange after binding it in Deconz to a group with a single light. Left button: Short press: ON, Long press: Dimming down!, Right Button: Short press: OFF, Long press Dimming up! You'd usually expect the dimming behavior other way round, i.e. dimming up with the on button and dimming down with the off button.