dresden-elektronik/deconz-rest-plugin #3062: Schedule time cannot be randomized

Thomas-Vos, 2020-07-20:
## Feature request type

When creating a schedule using the REST API the time cannot be randomized.

## Description

Send a POST request to `/api/<username>/schedules` with the following body:
```json
{
    "name": "SOn_28_Schedule 1_JbUwF",
    "description": "28",
    "localtime": "2020-07-20T19:30:00A00:10:00",
    "time": "2020-07-20T19:30:00A00:10:00",
    "command": {
        "address": "/api/<username>/groups/28/scenes/4/recall",
        "method": "PUT",
        "body": {}
    },
    "status": "enabled",
    "autodelete": false,
    "recycle": false
}
```
The request should be successful. However, the random part of the (local)time (`A00:10:00`) is ignored. That can be seen in the source code here:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/4154ad416996058bda9ab2d1af622fe5bef45cec/rest_schedules.cpp#L417-L425
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/4154ad416996058bda9ab2d1af622fe5bef45cec/rest_schedules.cpp#L833-L841

## Additional context

The Hue API supports randomized schedules.
The issue was originally reported here: https://community.hueessentials.com/t/randomize-schedule-and-phoscon-bridge/243

Mimiix, 2020-07-20:
Added backlog, otherwise bot is going to bother you again.

ChrisHae, 2020-08-06:
Randomized times will be available in the next deCONZ beta release.

closed 2020-08-06 by ChrisHae

Thomas-Vos, 2020-08-10:
Thanks @ChrisHae, that's awesome! Could you link the PR when you have it ready?

ChrisHae, 2020-08-19:
It is already in the master branch and should also been in latest deCONZ v.2.05.80 release
https://github.com/dresden-elektronik/deconz-rest-plugin/commit/8009689e5378749311db150a8d722885d961d4f1