dresden-elektronik/deconz-rest-plugin #3063: [Question] Window Covering device api

MadMonkey87, 2020-07-21:
I've seen that a KADRILJ roller blind has the following state:

      "state": {
        "bri": 0,
        "lift": 0,
        "on": false,
        "open": false,
        "reachable": true
      },

How am I supposed to use the api correctly? Can I just set lift to a value between 0 and 100? What do the other (bri, on, open) represent?

SwoopX, 2020-07-21:
Very valid question. There's some work on the API documentation required. Last update I proposed was meant to add all the available information real quick. We should make it clearer, what's decrecated and such.

I would aim for collecting some more cpmprehensive information over the next days/weeks.

ebaauw, 2020-07-21:
`bri` and `on` have been deprecated and shouldn't be used.  `reachable` is read-only.  `open` sends a _Close_ or _Open_ command; `lift` sends a _Go To Lift Percentage_ command.  Note that `lift` represents "percentage closed", so 0% is fully open and 100% is fully closed. There's also a write-only boolean `stop`, which sends a _Stop_ command when set to true.

Not all devices behave the same way, but the API should provide a standard interface, mapping any device-specific stuff.

MadMonkey87, 2020-07-21:
Thx a lot, this is exactly what I needed to know!

closed 2020-07-21 by MadMonkey87

rkupfer2020, 2020-07-24:
Hi,
I use multiple Heimann / _TYZB01_dazsid15 curtain switches with Conbee II and Deconz. I already upated the Docker Container to 2.05.79. But I am still not able to use the Phoscon GUI to open/close the curtains. Only with VNC they respond to the commands.
Under Home Assistant I can open and stop the curtains, but not close them. Anybody has an idea how to solve the issue in Home Assistant?
Best regards
Roger

SwoopX, 2020-07-24:
Please do not double post your in various issues, especially if an answer has been provided. Also, we cannot help you with any issues in HA. You may, however, join our Discord and raise your question there to profit from other user's experience. Thanks!