dresden-elektronik/deconz-rest-plugin #3064: Up to 80 localhost IP sockets in TIME_WAIT

HB-F, 2020-07-21:
## It seems to be deconz service on Raspberry opens permanently new ip sockets on port 80 using the localhost (127.0.0.1) address. 

## After starting deconz service it takes about 5 minutes and there are up to 80 ip sockets for localhost address in mode TIME_WAIT.

## Is it possible to prevent this behavior !

## Screenshots
[image]

## Environment

 - Host system: (Raspberry Pi 4)
 - Running method: (Raspbian)
 - Firmware version: (264A0700)
 - deCONZ version: (2.05.77)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)

### deCONZ Logs
Please see screen shot.

## Additional context

SwoopX, 2020-07-21:
I'll check it out on my systems.

Mimiix, 2020-08-12:
@SwoopX Did you check?

SwoopX, 2020-08-12:
Yes, got the same. But we need Manu on that one.

Mimiix, 2020-08-12:
@SwoopX Forwarded it to him.