dresden-elektronik/deconz-rest-plugin #3065: Cannont detect gledopto 2ar8f-gl-c-008 zigbee led dimmer

pippo73, 2020-07-21:
# Device gledopto

- Product name: gledopto 2ar8f-gl-c-008.
- Manufacturer: Gedopto.
- Device type : . 
  - Light/dimmer

I've seen that there are several treads about this producer, but I cannot make it work with my deconz (conbee 2 - 2.05.78 - fw 26580700)
My alexaplus detected it whitout any problem, but I cannot connect it with the web interface :-(
I've tried to do it as light and as switch, but both did not worked. Do I miss something?

Could someone help me to integrate it to my system pls?

thank you

Mimiix, 2020-07-21:
Hi! What do you mean with web interface? 

How did you connect alexaplus?

What does the restapi show?

pippo73, 2020-07-21:
> Hi! What do you mean with web interface?

I cannot find the dimmer with the Phoscon app

> How did you connect alexaplus?

on my smart phone i've started alexa app, and did a research of new devices... and "by magic" the dimmer was detected as new device
 
> What does the restapi show?

how to reach it? Sorry I don't know what you mean with this?

Mimiix, 2020-07-21:
Ah. Now i understand.

Before reading the rest of this message, please read: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies. This will give you some basic insights.

Phoscon needs to implement devices themselves as it is not opensource. However, the rest api is and that is what is exposing the device to alexa (probably).  That is the rest api doing that.

To check, you can use this to find out: https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

pippo73, 2020-07-21:
thank you , I will take a look and come back. :-)

pippo73, 2020-07-21:
> Hi! What do you mean with web interface?
> 
> How did you connect alexaplus?
> 
> What does the restapi show?

Here I'm back

this is what I get if I do a rest api lights request (hope is what you were asking me :-p )

{
    "1": {
        "etag": "52f450b23c98f45c562b5409b1c33ee5",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-07-21T18:41:26Z",
        "manufacturername": "dresden elektronik",
        "modelid": "ConBee II",
        "name": "Configuration tool 1",
        "state": {
            "reachable": true
        },
        "swversion": "0x264a0700",
        "type": "Configuration tool",
        "uniqueid": "00:21:2e:ff:ff:05:4e:69-01"
    },
    "2": {
        "etag": "1733eaf136d576bf3de2b6fb1d2951b7",
        "hascolor": false,
        "lastannounced": "2020-07-20T19:23:46Z",
        "lastseen": "2020-07-21T18:40:44Z",
        "manufacturername": "OSRAM",
        "modelid": "Classic A60 W clear - LIGHTIFY",
        "name": "cucina 3",
        "state": {
            "alert": "none",
            "bri": 254,
            "on": false,
            "reachable": true
        },
        "swversion": "V1.05.10",
        "type": "Dimmable light",
        "uniqueid": "7c:b0:3e:aa:0a:04:69:2b-03"
    },
    "3": {
        "etag": "58444e32ac02ba5700c8519674cc09b5",
        "hascolor": false,
        "lastannounced": "2020-07-21T16:34:22Z",
        "lastseen": "2020-07-21T18:41:57Z",
        "manufacturername": "LEDVANCE",
        "modelid": "A60 DIM Z3",
        "name": "ingresso 2",
        "state": {
            "alert": "none",
            "bri": 182,
            "on": true,
            "reachable": true
        },
        "swversion": "00103101",
        "type": "Dimmable light",
        "uniqueid": "f0:d1:b8:00:00:10:29:23-01"
    },
    "4": {
        "etag": "0cd2339867b1fec464ea50e8f4517a97",
        "hascolor": false,
        "lastannounced": "2020-07-21T16:34:22Z",
        "lastseen": "2020-07-21T18:42:12Z",
        "manufacturername": "OSRAM",
        "modelid": "Classic A60 W clear - LIGHTIFY",
        "name": "ingresso 3",
        "state": {
            "alert": "none",
            "bri": 254,
            "on": true,
            "reachable": true
        },
        "swversion": "V1.04.12",
        "type": "Dimmable light",
        "uniqueid": "7c:b0:3e:aa:0a:05:b6:a3-03"
    },
    "5": {
        "etag": "40e043c4227407ff697c743da8a4983f",
        "hascolor": false,
        "lastannounced": "2020-07-21T16:34:22Z",
        "lastseen": "2020-07-21T18:41:32Z",
        "manufacturername": "OSRAM",
        "modelid": "Classic A60 W clear - LIGHTIFY",
        "name": "ingresso 1",
        "state": {
            "alert": "none",
            "bri": 254,
            "on": true,
            "reachable": true
        },
        "swversion": "V1.05.10",
        "type": "Dimmable light",
        "uniqueid": "7c:b0:3e:aa:0a:03:5a:3e-03"
    },
    "6": {
        "etag": "010f388f53a1178a50b280aa6752ee54",
        "hascolor": false,
        "lastannounced": "2020-07-20T19:23:46Z",
        "lastseen": "2020-07-21T18:41:47Z",
        "manufacturername": "OSRAM",
        "modelid": "Classic A60 W clear - LIGHTIFY",
        "name": "cucina 2",
        "state": {
            "alert": "none",
            "bri": 254,
            "on": false,
            "reachable": true
        },
        "swversion": "V1.05.10",
        "type": "Dimmable light",
        "uniqueid": "7c:b0:3e:aa:0a:03:40:dc-03"
    },
    "7": {
        "etag": "12417cc758b941ecec71f9c7fc0f1b34",
        "hascolor": false,
        "lastannounced": "2020-07-20T19:23:46Z",
        "lastseen": "2020-07-21T18:39:36Z",
        "manufacturername": "OSRAM",
        "modelid": "Classic A60 W clear - LIGHTIFY",
        "name": "Cucina 1",
        "state": {
            "alert": "none",
            "bri": 254,
            "on": false,
            "reachable": true
        },
        "swversion": "V1.05.10",
        "type": "Dimmable light",
        "uniqueid": "7c:b0:3e:aa:0a:03:61:7a-03"
    }
}

Mimiix, 2020-07-22:
As far as i can tell it is not connected, but i have no clue how your alexa can switch it. Do you have other zigbee bridges? i found this manual: https://fccid.io/2AR8F-GL-C-008/User-Manual/user-manual-4146503 this shows how to reset it. 

Best plan to do:
1. make Phoscon search for 10 minutes for lights 
2. Reset the gledopto
3. t should be found.