dresden-elektronik/deconz-rest-plugin #3066: Aqara D1 Wireless Switch (Double Button) WXKG07LM

markuskorbel, 2020-07-21:
## Device

- Product name: Aqara D1 double key wireless wall switch
- Manufacturer: 
- Model identifier: lumi.remote.b286acn02
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

[image]
[image]
[image]

ekesgin, 2020-07-22:
I am also waiting for the support for this device. Beside, the support for WXKG06LM would also be very nice.

Smanar, 2020-07-22:
I have supposed they are same device and working same, so I have just added them in whitelist if someone can make a try

Just change the step 1 in

`git clone --branch Legrand-teleruptor https://github.com/Smanar/deconz-rest-plugin.git`

markuskorbel, 2020-07-23:
Hi Smanar,
Thanks for getting back to me. I cloned the branch and built the plugin.so on the raspbian os of my pi4. But when copying the file into the docker container running deconz it seems to break. Is there an architecture difference or something? Do I need to install the dependencies/git/etc. into the docker container and build it in there?

Smanar, 2020-07-23:
Yep, I think the device will be invisible in Phoscon, but visible on HA.
I think this devive will create only 1 entry, so if you have the battery information on one device, you need to have too buttonevent on it, or I m wrong somewhere.

Do you have the entry/device JSON for lumi.remote.b286acn02 5 ?

markuskorbel, 2020-07-24:
Here you go, sorry got a bit busy in work and had to read myself into using the api directly.

{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true,
        "temperature": 2800
    },
    "ep": 1,
    "etag": "fa3aafa1ed954f635a216028a0b18b3a",
    "lastseen": "2020-07-24T11:51:16.321",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.remote.b286acn02",
    "name": "lumi.remote.b286acn02 5",
    "state": {
        "buttonevent": 1002,
        "lastupdated": "2020-07-23T16:37:54.054"
    },
    "swversion": "20190329",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:04:4e:a8:00-01-0012"
}

markuskorbel, 2020-07-24:
It's just dawning on me that this is actually working fine here and I need to continue this journey in the deconz plugin for HA and get the switch added there. At least my python is a lot more up to date, compared to my C++.

I found the button events listed there for the 01 version of the switch, and they match up perfectly - didn't even know it supported double press and long press, awesome.

So thanks again for your help Smanar, I can confirm that the battery and button events are working fine, just need to get HA to create the binary_sensor entities now.

markuskorbel, 2020-07-24:
Ok final update - I promise - got the switches working. I think I found an old example of how to get an automation with these switches working that relied on binary sensors, I now found the newer/correct? guide using events.

I attached an event listener and tada!, so the events coming in, even found the device_trigger.py of the deconz plugin and added the switch there so the automation can detect the device's event types. I will submit a pull request for this into the home assistant repo myself, if you want to do the same for the deconz rest api. Thanks again, getting quite the proper introduction into the inner workings of these two systems :)

Smanar, 2020-07-24:
^^, thx for tests.

anarro, 2020-07-27:
Hi, 

I have some Aqara switches WXKG07LM. I am also waiting for the support for this device.

Thank you for your work,
Antoni.

fenderle, 2020-08-04:
Same here got the 2020 version which pairs with deconz but is not exported to the rest API

closed 2020-08-12 by manup

fribse, 2020-09-19:
Hi Markus
This looks very promising.
I have both 1 button and two buttons here, and It shows up on deconz, but nothing happens when they are pressed, will this PR handle both kinds merge is done @markuskorbel ?

markuskorbel, 2020-09-19:
Sorry not sure about the one button ones, don't have any to test, it will depend on the device IDs

Vissertje1980, 2021-03-10:
Hello, yesterday i update my RPi and the Phoscon app to the latest version (2.05.67) but i can not pair my Xiaomi switches type WXKG07LM. Are the supported yet or not? Otherwise i heave to by other type of switches.

markuskorbel, 2021-03-10:
I have been using them for months now and they are working perfectly. I'm currently running these software versions:
Phoscon 2.09.03 / 9/14/2020 (there is a newer 2.10 version but that's still in beta)
Conbee II firmware 26660700