dresden-elektronik/deconz-rest-plugin #3067: Blitzwolf bw-is5

pszypowicz, 2020-07-22:
##  Device

- Product name: BlitzWolf BW-IS5
- Manufacturer: _TYZB01_o63ssaah
- Model identifier: TS0207
- Device type : 
  - Sensor

## Screenshots

[image]
[image]
[image]
[image]
[image]

tobus3000, 2020-08-10:
Got two of these exact same devices. It's a 'unbranded' no-name water leakage detector based on the exact same HW platform.
Manufacturer Name: _TYZB01_sqmd19i1
The rest is identical to the above screenshots. The manual does mention in one place that a "Tuya Gateway" is required to make use of the device...
AFAIK it only reports whether it has detected water or not... No internal temperature, light or any other sensors..

Mimiix, 2020-08-10:
@SwoopX You able to whitelist on the TYZB01?

closed 2020-08-12 by manup