dresden-elektronik/deconz-rest-plugin #3070: deCONZ-WIFI2.sh[382]: Error: database disk image is malformed

Paalap, 2020-07-23:
## Describe the bug
Deconz is running fine (excepting the stange behaviour of creating additional group "ALL" on each reboot/start) but the system log ist full of errors regarding malformed deCONZ-WIFI2.sh and deCONZ-update2.sh. "Log is full" meaning upto 5 entries almost each second.

## Steps to reproduce the behavior
Just running the system

## Expected behavior
Clear log

## Environment
 - Host system: Raspberry Pi 4
 - Running method: Raspbian host, headless start (fresh setup, configuration migrated) 
 - Firmware version: 26350500
 - deCONZ version: 2.05.78
 - Device: ConBee I
 - Do you use an USB extension cable: no

Mimiix, 2020-07-23:
@Paalap Thanks for opening another one as per request.

i am still missing: 

### deCONZ Logs

Paalap, 2020-07-23:
@Mimiix 

deCONZ logs dont show any errors, my errors are shown in /var/log/syslog

Of course I cound post also full deCONZ logs but they are pretty "noisy"/long

Mimiix, 2020-07-23:
@Paalap Aah! Then just include a few of those.

Paalap, 2020-07-23:
@Mimiix is updated

Hope somebody can help me

ChrisHae, 2020-07-23:
You could make a backup of .local/share/dresden-elektronik/deCONZ/zll.db and delete the original zll.db to see if the malformed database image error is gone. But this means all lights,groups,etc. in the phoscon app are lost (thats why create a backup first)
You could also send me (or to support@dresden-elektronik.de) the zll.db so I could have a look why it is malformed.

Mimiix, 2020-07-23:
@ChrisHae sharing a zll.db is not something i recommend easily because the password is basically plain text in there. 

@Paalap Please change your Phoscon PW prior sending it.

Paalap, 2020-07-23:
@ChrisHae After deleting the db no errors are shown anymore... I could start from scratch but actually all the logic to connected switches/sensors is in my case in iobroker which is depending on internal deconz sensors ids, i.e. I have to connect everything in the same order or review all my code (both are not really preferred approaches)

I'll send the db to the mentioned email.

ChrisHae, 2020-08-13:
Sorry for the late answer but have you solved the problem already? You are not using WIFI, right? So disabling the deconz-wifi.service could solve your problem. You can keep your old db file.
(Another possible solution would be to confiure a WIFI AP in the phosocon app and then deacivat WIFI in the app if not wanted)

I think the error is because of missing wifi parameters in the db.

P.S. all passwords are encrypted in the zll.db

Paalap, 2020-08-13:
Thnaks a lot!

Yes, WIFI is not in use. The error is gone now for deconz-wifi2.sh I have only each 10 sec "deCONZ-update2.sh[415]: Error: database disk image is malformed".

ChrisHae, 2020-08-13:
I have found an article about this kind of sqlite problem, but I have to investigate further. https://sqliteviewer.com/blog/database-disk-image-malformed/

Disabling the deconz-update.service would disable this error message, but it would also disable the check for updates mechanism in the phosocon app. I think manual updates would still be possible.

Paalap, 2020-09-05:
Because I did not found a solution I'm about to setup deconz installation from scratch. Is there an easy way to delete the current installation (just deconz not the entire system) fully?

Smanar, 2020-09-05:
I think if you delete the database it will be enought, all other files will be same.

closed 2020-10-11 by stale[bot]