dresden-elektronik/deconz-rest-plugin #3071: Ikea trådfri hub + philips hue

ConfusedGuy123, 2020-07-23:
Hey!

I have this problem with my smart lights at home.

I have installed an Ikea trådfri hub, with 9 x philips hue lights. For switching them on and off i have trådfri remotes and a google nest. The problem is that when i turn on or off the light via the remote, the hub doesent seem to get the status of them beeing on or off. So if i turn the lights on via the remote, and the try to turn the lights off (via google nest with speech or from the app) it does nothing because it already thinks that the lights are off.

I dont remember having the same problem when i had trådfri bulbs, so does trådfri and philips hue bulbs send the information about their status somehow differently or is there a way to fix this issue? Its kinda annoying to have to ask google to turn all lights first on and then off so that they turn off.

-Thanks for any help!

ebaauw, 2020-07-23:
The Trådfri hub depends on attribute reporting to get the updated light state, which Hue lights don't support.  There's no way to make Hue lights (or any other lights that don't support attribute reporting) work with the Trådfri hub.

Mimiix, 2020-07-23:
Hi @ConfusedGuy123,

I'm happy to have issues open to help people, but this has nothing to do with deCONZ.

For help on those subjects you can try discord servers or reddit pages. 

Closing this one.

closed 2020-07-23 by Mimiix