dresden-elektronik/deconz-rest-plugin #3072: TUYA Smart Switch (EU) 3-Gang DS-112L-3 is incorrectly specified as a battery device

snador1977, 2020-07-23:
The TUYA Smart Switch (EU) 3-speed DS-112L-3 (230V) is incorrectly specified as a battery device.

https://zigbee.blakadder.com/Tuya_DS-112L-3.html

This could be the reason why the device shows no repeater functionality.

Usually all powered Zigbee devices are available as repeaters. The flush-mounted switches would be perfect as a repeater to ensure a stable Zigbee network.

Can you check this out? Thank you!

The same problem exists with another 3-way wall switch. Apparently a white label product.

https://zigbee.blakadder.com/Lerlink_X713A.html

https://de.aliexpress.com/item/4000369677580.html?spm=a2g0s.9042311.0.0.27424c4dJZFTv9

[image]

SwoopX, 2020-07-23:
There's nothing we can do about it. That info comes from the device and is stored in the firmware to be more precise. You may raise this to Tuya.

closed 2020-07-23 by SwoopX

vegetate7, 2020-08-13:
BTW, "No neutral" switches must not be routers. It's kind of "parasite" power supply can not provide enough for router.

Smanar, 2020-08-13:
No, they can ^^.
I have a switch without neutral that work fine as router, and see as router too.

vegetate7, 2020-08-13:
Hmm. Ok. I'm wrong then.