dresden-elektronik/deconz-rest-plugin #3074: Xiaomi Aqara Opple Switches

SAOPP, 2020-07-24:
Hi!

Guys, today I have been tested my **three switches** from aqara. 2x6buttons and 1x2buttons.

My two tested stends is working **raspbee** and **conbee2** in **docker**. Raspbee firmware is 26350500, and Conbee2 firmware is 264A0700. **Poscon is 2.05.79**.
 
_The behaviour is next:_ 

1. The switches didn't pass through the pairing mode - are not connected in the final.

2. When pairing mode is active and if you press any key on the switch, the switch is linking with (I don't know if it's right to say direct connection?) a "direct connection" to the bulbs on the network (in my case, it was three e27 Aqara at first stend, and onoff relay at the second stand). 

3. Moreover, if the switch communicates directly with the light bulbs (or with any other device on the network directly), it is not in Phoscon, but its node is present on the network map in Deconz GUI.

4. By some miracle, after several attempts to connect, my 6-button switch got into Phoscon, and it was detected correctly, but I couldn't get any events in the Home Assistant from it. Those. the switch was just useless.

5. Guys, developers, quite a long time has passed since the release of these devices, as well as since the official support of these devices in Deconz... What am doing wrong?

Thanks in advance.

ebaauw, 2020-07-24:
Just to be sure, could you please attach GUI screenshots of the node, incl. the endpoints and clusters, of the _Node Descriptor_ and of the _Basic_ cluster attributes?  Xiaomi are known to apply undocumented changes to devices.

SAOPP, 2020-07-24:
@ebaauw Erik I will do it, when I try to connect it again. Thanks.

jdeluyck, 2020-07-24:
I recently aquired 2 Opple 2x1 switches - adding them to the network is fairly easy (you just scan with phoscon and push the button on the back for 5+ seconds) - they show up in the network.

Switch type is WXCJRG12LM

I'm searching though how i can link them to a specific group - right now they're triggering everything :/

[image]
[image]
[image]
[image]
[image]

jdeluyck, 2020-07-24:
Ok. I retried adding them as a switch now in Phoscon, and scanning and then pressing the button for (I guess) around 15 seconds made it show up correctly in Phoscon.

ebaauw, 2020-07-24:
> Ok. I retried adding them as a switch now in Phoscon, and scanning and then pressing the button for (I guess) around 15 seconds made it show up correctly in Phoscon.

And, I hope, stop controlling all lights directly?

Next time before posting screen shot, please read the attribute first.  No need to do that now, since we know the switch is recognised.  Also next time, please also provide the _Node Info_ panel.

jdeluyck, 2020-07-24:
> And, I hope, stop controlling all lights directly?
Yes, this indeed now also stopped occurring. 

> Next time before posting screen shot, please read the attribute first. No need to do that now, since we know the switch is 
recognised. Also next time, please also provide the _Node Info_ panel.

Ah, apologies - I didn't quite think that through.

SAOPP, 2020-07-25:
@ebaauw it's mine:

[image]
[image]
[image]
[image]
[image]

 I read all info of node. Anything else is needed?

ebaauw, 2020-07-25:
That's the same _Model Identifier_, _Manufacturer Name_, and _Date Code_ as mine.  Also the same mac address prefix.  You didn't post the _Node Info_ panel to double check the _Manufacturer Code_, but that's probably 0x1014 as well.   I should pair.

Note that only the first two endpoints are listed: the others should have appeared when pressing the other buttons.  This could be indicative for Zigbee communication issues, which would definitely frustrate the pairing process.

SAOPP, 2020-07-25:
[image]

ebaauw, 2020-07-25:
Yep, that's the one.

SAOPP, 2020-07-25:
So, 0x115f is any news?

SwoopX, 2020-07-25:
No, it ahs been introduced with the support of those switches as I recall.

SAOPP, 2020-07-25:
clear, well then, guys, I don't know what to do, I have 2 independent instances and the same behavior with 3 different switches.

SAOPP, 2020-07-25:
literally 10 minutes ago I tried again to connect the 6-button switch on conbee2. And it all worked out :) Moreover, I didn't press any buttons during the pairing process. Everything turned out as usual.

SAOPP, 2020-07-25:
[image]
All works fine, in HA has the events.

cllatser, 2020-07-25:
I have the six button and it works very well with ConBee II.

Mimiix, 2020-07-25:
So it is solved? @SAOPP

SAOPP, 2020-07-27:
Well, on of my 3 switches are connected to conbee 2 without any clicking buttons through pairing process.

closed 2020-07-27 by SAOPP

mprajsler, 2020-11-20:
Hello guys,

It seems I got similar issue as mentioned here. I use Conbee stick with hassio installed on rpi4. I successfully paired Aqara OPPLE switch 3-gang WXCJKG13LM via Phoscon GUI but it controls nearby devices without any setup from my side. Is there any setting I should disable so I can use it how I want?

Thanks and Regards,
M.

mprajsler, 2020-11-20:
This seems to help me to create a group and bind all buttons there
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2061#issuecomment-567380475

SAOPP, 2021-03-15:
I got to connect the next my opple switch, and again the same behaviour, I have paired 15 times already, all the time it connected like directly paired. I'm turn off all my routers/bulbs, and nothing. I saw posts like a short press on the C button helps to select the correct mode during pairing, but somehow it does not help me. :D

Mimiix, 2021-03-15:
You need to click the C after pairing :)

SAOPP, 2021-03-15:
Yes ... it seems like it worked from the 20th time :)

Lyra1337, 2021-04-16:
Same here, I repaired the switch for dozen of times to my deconz. No matter if I press the C switch on the back. It always controls all my zigbee devices at once. Is there a way to get it into the correct mode?

ebaauw, 2021-04-16:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4589#issuecomment-804348316.

Lyra1337, 2021-04-16:
> 
> 
> See [#4589 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4589#issuecomment-804348316).

Thank you for pointing me there. I tried to do so, but in my new switch (i already have one successfully paired - somehow) doesn't has your described Multistate Input.

(left one works, right one controls ALL my zigbee switches at once)
[image]

ebaauw, 2021-04-16:
The Opples don't poll their parent when you press any of the front buttons.  You need to press the button on the back to make it reachable by the coordinator (e.g. when reading/writing attributes or reading the descriptors).