dresden-elektronik/deconz-rest-plugin #3077: Firmware issues

JimBob11, 2020-07-24:
## Describe the bug
Reset Conbee 2 accidentally and now can't get to reconnect to the firmware. I'm running it on a Raspberry pi 4 via Home Assistant. Can anyone recommend what to try?

[image]

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on, although I've lost my intergration
 - Firmware version: (??)
 - deCONZ version: (2.05.79)
 - Device: ConBee II
 - Do you use an USB extension cable:  no

Mimiix, 2020-07-24:
@JimBob11 https://github.com/home-assistant/hassio-addons/blob/master/deconz/DOCS.md

What is at the configuration page of the addon?

SwoopX, 2020-07-24:
> Reset Conbee 2 accidentally and now can't get to reconnect to the firmware.

What does that mean? How do you do that? As a side note: the Conbee can't establish a serial connection.

JimBob11, 2020-07-24:
🤣🤣 I thought reset meant said restart, I'm blaming it on my dyslexia. What is a serial connection? And how do I get one?

JimBob11, 2020-07-24:
> @JimBob11 https://github.com/home-assistant/hassio-addons/blob/master/deconz/DOCS.md
> 
> What is at the configuration page of the addon?

serial:
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2196037-if00
/dev/ttyACM0
/dev/ttyAMA0

I've tried using all of these but everything I try the results are the same.

JimBob11, 2020-07-24:
Not sure if it helps but when i search for my sensors, they can't be discovered and also the zigbee channel is stuck on 0.

Mimiix, 2020-07-25:
@SwoopX

SwoopX, 2020-07-25:
You should join our Discord channel get some more interactive support on this one. One thing worth to mention is that since you use a Conbee, `/dev/ttyAMA0` should be irrelevant for you.

What are the owner information and permissions for the below devices (e.g. `ls -la /dev/ttyACM0`)?
```
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2196037-if00
/dev/ttyACM0
/dev/ttyAMA0
```

JimBob11, 2020-07-25:
> You should join our Discord channel get some more interactive support on this one. One thing worth to mention is that since you use a Conbee, `/dev/ttyAMA0` should be irrelevant for you.
>
> What are [...]

Thanks @SwoopX do you have a link to the Discord channel? I have very basic knowledge of how to code etc. as I am new to Home Assistant and how to configure addons, so when you say  "What are the owner information and permissions for the below devices" I'm unsure what information you are after. I've also tried connecting the Conbee 2 to my laptop to see if I have any issues with it on that and when I use the Deconz app it recognises there is a Conbee 2 plugged in but when I try to connect it, it just shows as not connected.

SwoopX, 2020-07-25:
The link is in the readme of this repository.

The information I asked has nothing th do with coding, but are operating systems basics.

You should probably just try to reflash the stick with the latest firmware on a bare windows system.

samvanh, 2020-07-26:
Have the same issue after an FW upgrade of the Conbee II.
And no windows pc available to reflash the FW to an older version.

Looks like I'm not the only one: https://github.com/home-assistant/hassio-addons/issues/1376

JimBob11, 2020-07-26:
> Have the same issue after an FW upgrade of the Conbee II.
> And no windows pc available to reflash the FW to an older version.
> 
> Looks like I'm not the only one: [home-assistant/hassio-addons#1376](https://github.com/home-assistant/hassio-addons/issues/1376)

@samvanh thanks for this, I'll have a look into it.

samvanh, 2020-09-13:
FYI: 
-returned my conbee II (within warranty)
-ordered new conbee II
-plugged in and worked out of the box
-restored the latest backup file and all my devices were instantly available.

So the issue really was a bricked stick after a firmware upgrade.

Smanar, 2020-09-13:
But you have tried first to update the firmware again with a real machine ?

samvanh, 2020-09-13:
I don't have a Windows PC available and the only Linux I have is running on HassOS which couldn't 'find' the conbee at all. 
Tried all available USB ports and serial configurations but with no luck.