dresden-elektronik/deconz-rest-plugin #3080: state.alert does not update / represent device state

balu-, 2020-07-26:
## Describe the bug
The value state.alert seems to be always "none" when queried via the API even when the device is on state "select" or "blink".
I sumbled across that fact while testing with a waring device, but also confirmed with a philips lamp.

## Steps to reproduce the behavior

1. send a PUT "{ 'alert':'select' }" Request to a Light resource (egg. /lights/1/state )
2. check that the light is blinking
3. send a GET to that light (egg. /lights/1)
4. observe that the response to 3 has  "state": { "alert": "none" } in it.
5. further the state.alert change is also not transmitted via websocket

This can also be observed with alert "blink"

## Expected behavior
The state.alert value should represent the actual alert state of the light and state.alert changes should be anounced via websocket.

## Environment
 - Host system: Raspberry Pi 
 - Running method: Raspbian 
 - Firmware version: 26350500
 - deCONZ version: 2.05.77
 - Device: ConBee I 
 - Do you use an USB extension cable: yes

ebaauw, 2020-07-26:
No, it shouldn't; `state.alert` is write-only;  Zigbee does not provide a way to query the actual alert state of a device.

Note that the Hue bridge API behaves lightly differently: it echoes the last value set, but still not the actual alert state.

balu-, 2020-07-26:
Interesting. 
In that case I would appreciate if the API doc reflects that.
Correct me but in deconz-Api this field does always have the value "none" then.

ebaauw, 2020-07-26:
> In that case I would appreciate if the API doc reflects that.

Reflect what?

> in deconz-Api this field does always have the value "none" then.

Yes.  It would make more sense not to include `alert` in the `state` at all (cf. e.g. `bri_inc`), but then we have to way to indicate whether a light or sensor supports setting `alert`.

balu-, 2020-07-26:
> > In that case I would appreciate if the API doc reflects that.
> 
> Reflect what?

That `state.alert` is always none and not really a value but an indicator that `alert` can be put for this device.
The [docs](https://dresden-elektronik.github.io/deconz-rest-doc/lights/#getstate) mention in the `get State` part that `state.alert`might also take the values `select` and `lselect` which doesn't seem to be correct from what you wrote, or am i wrong?

closed 2020-08-23 by stale[bot]