dresden-elektronik/deconz-rest-plugin #3081: Updated from 5.3.6 > 6.1.2 now the add-on won't run for longer than a few seconds

Puntoboy, 2020-07-27:
I was running 5.3.6 of the Hassio add-on and I just upgraded to 6.1.2. Now the add-on doesn't run for longer than a few seconds. The logs show this:

`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] firmware.sh: executing... 
/usr/bin/GCFFlasher_internal.bin: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory
[13:27:28] INFO: 
[cont-init.d] firmware.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] vnc.sh: executing... 
[cont-init.d] vnc.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[12:27:29] INFO: Running the deCONZ OTA updater...
[12:27:29] INFO: Running the OSRAM LEdvance OTA updater...
[12:27:29] INFO: Running the IKEA OTA updater...
[12:27:29] INFO: Starting udevd...
[13:27:29] INFO: Waiting for device...

[13:27:35] INFO: Starting VNC server...
[13:27:39] INFO: Waiting for VNC to start

[12:28:29] INFO: Starting Nginx...
2020/07/27 12:28:29 [notice] 350#350: using the "epoll" event method
2020/07/27 12:28:29 [notice] 350#350: nginx/1.14.2
2020/07/27 12:28:29 [notice] 350#350: OS: Linux 3.10.105
2020/07/27 12:28:29 [notice] 350#350: getrlimit(RLIMIT_NOFILE): 524288:1048576
2020/07/27 12:28:29 [notice] 350#350: start worker processes
2020/07/27 12:28:29 [notice] 350#350: start worker process 2611
[13:28:39] INFO: Starting the deCONZ gateway...
deCONZ: error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory
2020/07/27 12:28:39 [notice] 350#350: signal 15 (SIGTERM) received from 338, exiting
2020/07/27 12:28:39 [notice] 2611#2611: exiting
2020/07/27 12:28:39 [notice] 2611#2611: exit
2020/07/27 12:28:39 [notice] 350#350: signal 17 (SIGCHLD) received from 2611
2020/07/27 12:28:39 [notice] 350#350: worker process 2611 exited with code 0
2020/07/27 12:28:39 [notice] 350#350: exit
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.`

What have I done wrong?

SwoopX, 2020-07-27:
I'm afraid you need to raise this in HA as we're not involved in the add-on. This repository is for the deconz REST API plugin.

closed 2020-07-27 by SwoopX

Puntoboy, 2020-07-27:
Thanks I raised it there as well. I wasn't sure as the error seems specific to Deconz :) Any insight even though it's not your issue?

SwoopX, 2020-07-27:
Well, for me, the error messages speak very clear: a required shared library could not be found or was not included.