dresden-elektronik/deconz-rest-plugin #3082: Phoscon Gateway with lost connections to nodes

robertftvos, 2020-07-27:
## Describe the bug

Main issue:
The Phoscon Gateway with RaspBee II on deCONZ 2.05.78 and .79 gets lost connections to multiple nodes in mesh of 100+ Nodes. Those become unreachable when queried and can sometimes, but not always, be paired again. Reading the simple descriptor then fails (red). Devices impacted are for instance Aqara Wireless Switch, Aqara Multisensors, various types of Philips Hue lights, Philips Hue Motion sensor, but also the Phoscon Color Kit. The Configuration Tool does not always connect first on cold reboot and often takes ages to connect. Issues have started since about 60 nodes were added on 2.5.77.

Other current issues: 
- Aqara Multisensors cannot be added anymore. When one is lost in mesh (won't light up blue anymore when polled), and you remove it, it is not found anymore after reset.

- Aqara Smart Curtain gets renamed automatically to a Motion Sensor.

## Steps to reproduce the behavior

After running the deCONZ for some hours, first node connections get lost and in time the whole system becomes unresponsive (e.g. lights cannot be switched through buttons or our app Hue Essentials). Removed and installed deCONZ 2.05.78 as well as 2.05.79 with same results. When all lost nodes are paired again (some unsuccessful), then it is just waiting for deCONZ to drop them again. Few hours later, for instance motion sensors do not respond anymore, switches that worked a minute before do not respond, and lights do not switch off anymore.

## Expected behavior

Restored connectivity:)

## Screenshots
Screenshot of all devices are in a fairly good state how it is supposed to be. A bad situation has hardly any connected lines. Only some devices then have a green line between them. The log has a bad situation when devices do not connect properly anymore.

[image]
[image]

## Environment

DeCONZ build: 2.05.78
Firmware version: 0x26520700

 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - Firmware version: 26520700
 - deCONZ version: 2.05.78
 - Device: RaspBee II
 - Do you use an USB extension cable: n/a

### deCONZ Logs

[debug_phosconGW_failed_connectivity_170720.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4982577/debug_phosconGW_failed_connectivity_170720.txt)

## Additional context

 Network now has slightly over 100 devices (lights, sensors, switches). Mainly Philips Hue (various types like E14, E27, GU10, luminaires, ledstrips), Phoscon Color Kit, Gledopto, innr, tint Mueller Licht, Osram, Aqara, Flair, Tuya, Heiman, Ubisys, Trådfri. Both Aqara Smart Curtain Motor and Trådfri Blind are in network. There are multiple bridges (Trådfri, Philips Hue, ConBee I, ConBee II, RaspBee II) over different Zigbee channels and Google WiFi mesh on 2.4/5GHz.

haraldknoll, 2020-07-28:
I have this problem for years now. The only workaround for me was network segmentation. With all the drawbacks. “Issues have started since about 60 nodes“. I hit the same limit. No solution from dresden-elektronik since mid 2018.

ReX1983, 2020-08-14:
> I have this problem for years now. The only workaround for me was network segmentation. With all the drawbacks. “Issues have started since about 60 nodes“. I hit the same limit. No solution from dresden-elektronik since mid 2018.

Argh... I have almost concluded a massive migration to get rid of 3 Xiaomi gateways, 1 Hue bridge and 1 Tradfri hub and connect all my devices (about 150) to a single Zigbee mesh on Deconz.

Do you have multiple Deconz coordinators / Deconz installations, each with less than 60 devices?

> Aqara Multisensors cannot be added anymore. When one is lost in mesh (won't light up blue anymore when polled), and you remove it, it is not found anymore after reset.

And in this case what do you do to connect it back to the network? 
Recently I had a pairing issue with the multisensors, here what worked for me: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1752#issuecomment-674003183

robertftvos, 2020-08-14:
I have multiple deCONZ installations. The ones with less devices have no issues so far and are able to pair to those aqara multisensors without issues. I was aware of the battery issue you linked to but this does not help either on the affected RaspBee II. 

When a device like a light is lost, often it can be paired again by just resetting it and start the search.

ReX1983, 2020-08-17:
> When a device like a light is lost, often it can be paired again by just resetting it and start the search.

Do you mean by just restarting the device (as I do today with the Trådfri bulbs connected to the Hue Bridge) or by putting each bulb in paring mode?

The latter would be quite painful to achieve in my setup (17 bulbs wired in parallel - they can only be switched on and off together).

robertftvos, 2020-08-18:
I put them in pairing mode. Use a Philips Hue Dimmer Switch by holding on/off button together next to bulb or so.

closed 2020-09-19 by stale[bot]