dresden-elektronik/deconz-rest-plugin #3083: Aqara Smart Wall Switch Single Rocker

jdonofrio728, 2020-07-27:
## Device

- Product name: Aqara Smart Wall Switch Single Rocker
- Manufacturer: LUMI
- Model identifier: lumi.switch.b1naus01
- Device type : Please remove all unrelated device types. 
  - Light
  - Sensor
  - Thermostat

## Screenshots

Simple Metering:
[image]

Electrical Measurement:
[image]

Alarms:
[image]

Device Temperature:
[image]

SwoopX, 2020-07-29:
@jdonofrio728 What switch model is this? Looks quite interesting...

jdonofrio728, 2020-07-29:
@SwoopX From the box Model: WS-USC03

https://www.amazon.com/Aqara-REQUIRES-Automation-Compatible-Assistant/dp/B081ZRND3K/ref=sr_1_1_sspa

SwoopX, 2020-07-29:
Thanks! Apart from the fact, that it looks fugly, I like the features I see from here.

jdonofrio728, 2020-07-29:
lol it looks better than some of the cheaper ones that I have.

closed 2020-08-12 by manup