dresden-elektronik/deconz-rest-plugin #3084: Not recieving double-tap events into home assistant from an Aqara Magic Cube

cooperaj, 2020-07-27:
## Describe the bug
I've been able to successfully pair the cube in phoscon so that I can interpret *all* events except the double-tap. Slide, drop, rotate, 90 and 180 degrees all work fine but no event according to the 'deconz_event' watcher in home assistant ever comes through as a double-tap type.

## Steps to reproduce the behavior
Add cube sensor to deconz, ensuring that you rotate, roll and flip it throughout the process. Follow the 'deconz_event' log in home assistant and see all events come through except for double tap.

## Expected behavior
Would see the double-tap event. As per the docs at https://www.home-assistant.io/integrations/deconz/ it should be tagged as Gesture 6.

## Environment
Conbee II
Phoscon: 2.05.78 22/5/2020
Firmware 2650700
Home Assistant 0.112.0 on RaspberryPi 4

SwoopX, 2020-07-28:
Does the REST API show the correct button event? My cube works fine with that gesture.

cooperaj, 2020-07-28:
Do you have some quick and dirty instructions (or a link to) so that I can do that?

SwoopX, 2020-07-28:
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

cooperaj, 2020-07-28:
Thank, I'm able to get connected and can read out the cube - though i do have two, one that seems to register the analogue events and one for the binary ones. Figuring tap was a binary I did all the things to it and could see the various events, but I can't see the tap one :(

I can't figure out how to connect to the websocket to have the events streamed but I think that does it.

When pairing the cube I had to ensure that I span and rotated it to get it to register with analogue and digital, do I have to tap it more vigorously? I have been doing but maybe not enough

ebaauw, 2020-07-28:
I’m afraid that’s an issue with the cube; all digital events are sent the same way, it’s not possible for one event type to arrive but nor another.  Could be a hardware issue with the sensor registering the double tap.  Could be a firmware issue; Xiaomi have been to to change silently the firmware on later models, without changing the name.  What version/date code does you cube report.  If it just sends a different value on double tap, we can fix it, but if it doesn’t send any message at all, there’s nothing we can do.  Could you try and run deCONZ with debug logging to see what message (if any) the cube sends on double-tap?  Alternatively, use a Zigbee sniffer.

cooperaj, 2020-07-28:
```
    "9": {
        "config": {
            "battery": 95,
            "on": true,
            "reachable": true,
            "temperature": 2200
        },
        "ep": 2,
        "etag": "f0e098479f4eae53cffaace456b67430",
        "lastseen": "2020-07-28T10:16:22.235",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.sensor_cube.aqgl01",
        "name": "Study Cube",
        "state": {
            "buttonevent": 7000,
            "gesture": 0,
            "lastupdated": "2020-07-28T10:16:22.234"
        },
        "swversion": "20160704",
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:04:86:10:d8-02-0012"
    }
```

This is whats coming off the cube. The firmware seems.. old - if its a date. I'll see what debugging comes out but I don't have a sniffer sadly.

cooperaj, 2020-07-28:
Oh, I'm so sorry. 

I have read "double-tap" as tapping the device with my fingers, and have been furiously banging my fingers on it. I just saw on the Aqara website that it's tapping the whole device on a surface.

Doing that works fine. Event 6 comes through repeatedly.

Thanks for the help!

closed 2020-07-28 by cooperaj