dresden-elektronik/deconz-rest-plugin #3085: SR-ZG2835 Zigbee Rotary Switch RGB+CCT RGB CCT Mono 4 in 1

alexdepalex, 2020-07-27:
## Device

- Product name:  Zigbee Rotary Switch RGB+CCT RGB CCT Mono 4 in 1 / 
- Manufacturer: Sunricher
- Model identifier: SR-ZG2835 (ROB_200-018-0)
- Device type : Switch

## Screenshots

[image]
[image]
[image]

Device doesn't show up in Phoscon, but it does in Deconz. Blue button blinks whenever the switch is used. Sending commands like Identify works as well.

SwoopX, 2020-07-28:
Please provide all requested screenshots mentioned in the template, otherwise we'll close the request due to incomplete information. Thanks.

alexdepalex, 2020-07-28:
Think I forgot the clusters of the node screenshot. Anything else missing?

[image]

alexdepalex, 2020-07-29:
So I can't see the device in the rest api. I do see some lines in the deconz logging when pushing and turning the rotary button.

```
23:02:07:146 APS-DATA.indication srcAddr: 0x92ea, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 135, rssi: -77
23:02:08:678 APS-DATA.indication srcAddr: 0x92ea, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 135, rssi: -77
```

Is there any more info I can add to this issue? Anything I can help with?

Mimiix, 2020-08-02:
@SwoopX

Gnaget2, 2020-08-05:
Have the same rotary switch, can see it in VNC but not in deconz or in home assistant, anything i can do/provide to help out to get it working?

zyberzero, 2021-12-28:
I've recently bought this device and I can't get it to work. 

I've repaired it several times, but it doesn't show up in the web UIs - it's shown in the old deconz GUI though.
[image]

Running version 2.13.04 / 12/18/2021 and firmware 26390500.

I asked some questions on the discord channel, and got asked what DDF mode I was in so here it goes:
[image]

The people on the discord channel suggested that I should continue on this issue instead.

Please let me know if I can help you further in any way.

Mimiix, 2021-12-28:
@manup @SwoopX can you check this?

Smanar, 2021-12-29:
Problem with mac adress, not whitelisted

```
{ VENDOR_SUNRICHER, "ZG2835", silabs6MacPrefix }, // SR-ZG2835 Zigbee Rotary Switch

        case VENDOR_SUNRICHER:
            return prefix == emberMacPrefix ||
                   prefix == silabs3MacPrefix ||
                   prefix == silabs6MacPrefix;

const quint64 silabs6MacPrefix    = 0x588e810000000000ULL;
const quint64 silabs3MacPrefix    = 0xec1bbd0000000000ULL;
const quint64 emberMacPrefix    = 0x000d6f0000000000ULL;
```

I think it can be solved using DDF file.

zyberzero, 2021-12-29:
@Smanar , no - this one shows up as 0x100b (If I read it from the right place, node info  `manufacturer code`)

Smanar, 2021-12-29:
Ok so double problem ^^

`#define VENDOR_PHILIPS              0x100B`

I think you can try to make a DDF for deconz include the device, then after remove it (the DDF core don't manage battery switch yet)

Smanar, 2021-12-30:
@zyberzero the better starting point for DDF is https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

So for you if I m right you just need it to force deconz to reconize the device.
The important part was the "basic cluster identification" filled with drag and drop.
As device use a ZHASwitch, with letting all by defaut
After the entry was created in the API (you will see a name instead of 0xXXXX), just disable your DDF and restart deconz. The "defaut code" will finish the inclusion.

And after a "hot reload" try to include your device

zyberzero, 2022-01-01:
@Smanar  thanks.

I've created the DDF like this:
[image]

It is now recognized as a named device, "Switch 79" in my case.

This is how the device info looks like:
```                    
{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "2df8c1416662a5df0803385f5366a7e7",
    "lastannounced": null,
    "lastseen": "2022-01-01T10:35Z",
    "manufacturername": "Sunricher",
    "mode": 2,
    "modelid": "ZG2835",
    "name": "Switch 79",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "type": "ZHASwitch",
    "uniqueid": "bc:33:ac:ff:fe:5d:4f:85-01"
}
```

Here I changed the state of the DDF back to draft, and restarted Deconz.

Now it's back to a `0xXXXX` named device, and nowhere to be found in the web UIs (previuosly it was listed as a Sunricher switch, albeit the wrong model - it had a picture of 4x2 buttons on it, this is a knob).

 No events where emitted (according to API information) regardless if it was using the DDF or not when turning the knob.

Is there something else I can try, or did I misunderstand you?

Thanks!

Smanar, 2022-01-01:
>Is there something else I can try, or did I misunderstand you?

Nope was perfect, but why it s back to 0xXXXX ? One time a device is in the API, he was still used by classic code for me, I will ask to devs to be sure.

And if you try the "hybrid" mode ? With the status bronze, perhaps the legacy code will be used to manage the sensor ?

zyberzero, 2022-01-01:
I changed the status to Bronze and enabled hybrid mode; deleted the node and rejoined it again.

It now shows up as `Switch 79` again, but it seems to be the wrong model (yet again). 

[image]

No events are shown in the PWA API debugger (i see the blue dot flash in Deconz GUI however).

[image]

Smanar, 2022-01-02:
>It now shows up as Switch 79 again, but it seems to be the wrong model (yet again).

Don't worry , this is probably a phoscon issue, nothing to see with deconz.
On "hybrid" mode, the code use DDF and legacy code, so can you check in decnz log if you see something whenn using a button ?
On deconz / help / debug view, with flag "info" and "info_l2"

But the method is probably bad, IDK if its possible to do that, I m asking more information to others devs.

zyberzero, 2022-01-02:
The only line that showed up in the logs that I could connect back to my remote was this (searching  by mac-address and its name)
```
07:16:13:790 ZCL attribute report 0xBC33ACFFFE5D4F85 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
```

There was some stuff around that line as well, but I have a network consisting of 67 nodes so it is really noisy.

Smanar, 2022-01-02:
I m searching line starting by 
`[INFO] - No button map for`

Or perhaps you have something on "DDF" flag as the device is perhaps using the DDF core now.

I m not sure it will change something (you need it to have the device working, the legacy code configure them at start, but as it seem you are using the DDF core only now ...), but can try to configure the device to make bind/report or can make then yourself
You need to bind and configure report for cluster 0x0006 0x0008 and 0x0300, but the 0x0006 is enought to make tries

To configure and set you have an exemple here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2077#issuecomment-554640916

zyberzero, 2022-01-02:
@Smanar , I tried to bind it to the conbee-stick, it says success but no new log events (I searched for `button map`, but no matches).

If I bind it straight to one of my lights it works though, so the device seems to work (I bound `0x6` and `0x8` to an Ikea lamp I had lying around)

Smanar, 2022-01-03:
So binding with a bulb is working but nothing in log if binding with the gateway ? Not logic, it's probably bypassed by the DDF core, nothing to in logs if enabling "ddf" log flag ?