dresden-elektronik/deconz-rest-plugin #3086: Device name

haraldknoll, 2020-07-28:
Non of new (Model End 2019) Neuhaus Lightning Group lights can be integrated in the Phoscon App. Older Neuhaus Lightning Group lights work well. The new light  can not be joined at all. E.g. the Q-VITO Series https://www.paul-neuhaus.de/shop/de/q-vito-led-deckenleuchte-smart-home-d-79-4cm-44942.html

Current Setup:
-	Raspberry 4 with RaspBee II
-	Phoscon Version 2.05.79
-	Firmware RaspBee II – 26610700

## Device

- Product name: Q-VITO LED-Deckenlampe
- Manufacturer: Paul Neuhaus
- Model identifier: 
- Device type : 
  - Light
  

## Screenshots

ebaauw, 2020-07-28:
Do they show up in the GUI?  If so, please provide the screenshots mentioned in the wiki.  If not, there’s nothing we can do in the REST API plugin.

haraldknoll, 2020-07-28:
I run all PI 4 devices headless. Is it possible to list all devices by command line? It is very likely that is will be seen on the network. Seem I have to install a GUI version. BTW: The device joins flawless on a Philips HUE bridge.

ebaauw, 2020-07-28:
> BTW: The device joins flawless on a Philips HUE bridge.

That's good.  Now we just need the screenshots.

> I run all PI 4 devices headless.

I run all my Pi-s headless (without a monitor connected), but I still run the Pi's graphical environment and deCONZ in GUI mode.  Connecting to the VNC server is a piece of cake from macOS, using the mac's native screen sharing client.

> Is it possible to list all devices by command line?

No.  Using deCONZ without the GUI is like driving a Ferrari without using the gearbox.  It kinda works, but won't unlock the full power.

haraldknoll, 2020-07-29:
Well, i made a setup with just one "old" Paul Neuhaus Light. It joins. Non of the new Paul Neuhaus Light e.g. Q-VITO Series appears in the deCONZ node list. The "light 3" is an old Paul Neuhaus light.  

[image]

haraldknoll, 2020-07-29:
After a while a device pops up which reports as HA simple sensor. Looks like a power save issue, but its a full powered light.

[image]

haraldknoll, 2020-07-29:
Here we have the Node Info.

[image]

SwoopX, 2020-07-29:
Can you please read the attributes of the basic cluster and share the output? Thanks!

haraldknoll, 2020-07-29:
Here we go ... 

[image]

SwoopX, 2020-07-29:
As written above, please **READ** the attributes.

haraldknoll, 2020-07-29:
OK, back to start. Th 0x8ad4 is sensor an attribute read is not possible. Dont know what it is. The 0xC607 is the right one.  Here are the values:

[image]

[image]

SwoopX, 2020-07-30:
Alright, thanks. So, there we have it: your light doesn't expose itself as a light and that's the reason, why it is not recognized out of the box. It also does not expose any clusters or capabilities one would expect to see here. Not sure if that's really zigbee compliant.

Your "problem" is now exactly that. It doesn't even have an "on/off" cluster to switch it. My assumption is that all the functionality is hidden in the manufacturer specific 0xFC00 cluster, and that must be reverse engineered with the help of the device at hand and a sniffer. I don't have much hope that any information around that is publicly available.

I observe offloading functionality quite often these days. Probably the intention behind it is to make it harder for free projects to integrate the devices and kinda force the customer to buy their own products/gateways...

haraldknoll, 2020-07-31:
Well, deleted node and did a search again. At  the end of the day the lamp was found with the following procedure. Open network and reset lamp. The other way e.g. lamp reset and open network after lamp reset never worked for me, Should we do a sniffer trace to see why the initial join will not work.  

Control of lamp seems to be OK. Looks like only the initial detection of the lamp is a bit sluggish.

[image]

[image]

[image]

[image]

closed 2020-08-29 by stale[bot]