dresden-elektronik/deconz-rest-plugin #3092: Bitron Home 902010/29 - outside siren

Smarti15, 2020-07-29:
- Product name: Bitron Home 902010/29 outside siren
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type  
  - Light
  - Siren

Manual: http://media.bitronvideo.eu/bitronhome/download/LBT90359-A_902010-29_GB-DE.pdf

I tried to pair the device and in the Phoscon app it does not report it successfully, but om deconz adapter of ioBroker I get states. Only they don't work properly. I would like to script an alarm to a state and trigger the siren. Could you possibly see the device better?

ioBroker states : https://i.imgur.com/ZsjotMg.png

TXH

Mimiix, 2020-07-29:
Hi @Smarti15 !

Can you please add the screenshots as requested [here](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support)? 

We need that information, otherwise we can't add it.

Thank you.

Smarti15, 2020-07-29:
Ok I hope you guys can relate to this?

https://i.imgur.com/UVjIqn4.png

SwoopX, 2020-07-29:
The node info pannel screenshot is missing

Smarti15, 2020-07-29:
sry 

https://i.imgur.com/KAWtmC9.png

SwoopX, 2020-07-29:
Thanks. Please upload the screenshots directly here instead at an external hoster for future reference.

Smarti15, 2020-07-29:
I'll keep that in mind. Now, do you have all the information you need?

closed 2020-08-12 by manup