dresden-elektronik/deconz-rest-plugin #3093: all my Aqara sensors stopped working for motion as of last night

ggg22, 2020-07-29:
I have 3 Aqara sensors that stopped activating the lights as of last night.
I use a plugin in Homeseer to connect to deCONZ.
All of them report fine light levels (luminance) but no motion.
On the deCONZ screen I see there is activity when the Aqara sensor detects motion but there is no action on Homeseer level.
This setup used to work fine for many months.. 
PC running Windows was rebooted, no luck.
Wonder if anybody has any clues..

Mimiix, 2020-07-29:
Hi.

I have no clue how homeseer works 😅  Does Phoscon show if there is motion?

ggg22, 2020-07-29:
Thanks for your reply, Dennis!

 

It seems the problem is with 3 sensors.. I actually have 5 in total of same type.

Experimented with one of the 3 non-working.. it’s on my desk..

So, the first time I opened Phoscon it showed motion and the light got activated.. 

This is the first time it worked since last night so I thought is it possible opening Phoscon did something..

But no, following with further testing it stopped working.

What is happening now is the following:

I have deCONZ and Phoscon opened side by side and then I move my hand in front of the sensor.

I see there is activity in deCONZ for that sensor but it doesn’t show motion in Phoscon.

 

Thanks,

-George

ggg22, 2020-07-29:
In addition.

Just tested changing luminance and it reports it fine in Phoscon.

Also, measured the battery and it’s fine – a bit over 3 Volts.. 

-gh

ggg22, 2020-07-29:
So, 3 out of 5 sensors stopped reporting motion as of last night.

They report luminance fine.

Very strange indeed.

 

Thanks,

-George

Mimiix, 2020-07-29:
Interesting. Any logs you can get?

ggg22, 2020-07-29:
Sorry Dennis. Don’t know how to get logs off deCONZ.

And the Homeseer plugin doesn’t seem to provide logs from deCONZ regarding sensors.

There is some basic logs in Homeseer for deCONZ lights and plugs but not for sensors.

If there is an easy way and you give me a clue how to get them I can look for them.

It’s running on Windows 10.

 

Thanks,

-George

ggg22, 2020-07-29:
Dennis,

 

Any idea how I can get a deeper view on a Windows setup?

I am ok to purchase additional hardware if required.

This seems to be a software issue by the looks of it.

 

Thanks for your help so far.

 

Regards,

-George

wvuyk, 2020-07-30:
Did you try resetting the motion sensor? 
Start a search for new devices in either Phoscon or the configuration page of JowiHue. Then on the motion sensor device, press the little button for 4 seconds until the blue light starts blinking. 
Check on the deCONZ window if you see the dots blink on the motion sensor device and check in HomeSeer if the motion part starts responding again.

ggg22, 2020-07-30:
Wim,

 

You simply nailed it! 😊

I didn’t have much time to play with this the last couple of days but noticed after some time the motion function would work for each of these but then it would stop again.

It would work only once that is.

And then you have hours of not working.

And then you will get a single trigger again after some hours.

But following your procedure fixed all 3 sensors so, thanks very much for your brilliant advice!

I am still very confused as to why 3 sensors stopped working at the same time and 2 others were fine.

 

Best Regards,

-George

Mimiix, 2020-07-31:
Sorry, had some busy days! Great to see it is resolved @ggg22  

Can i close the issue :)?

ggg22, 2020-07-31:
Thanks Dennis. Yes, it can be closed. So far things are working as expected.

 

Regards,

-George

Mimiix, 2020-07-31:
Great to hear that George!

Enjoy your weekend :)

closed 2020-07-31 by Mimiix

ggg22, 2020-07-31:
Thanks very much, Dennis and same to you! 😊

-gh