dresden-elektronik/deconz-rest-plugin #3094: Mijia Light Sensor

PainElemental, 2020-07-29:
- Product name: Mijia Light Sensor
- Manufacturer: LUMI
- Model identifier: lumi.sen_ill.mgl01
- Device type : 
  - Sensor
[image]
[image]
[image]

PainElemental, 2020-07-29:
Unfortunately I was not able to retrieve ClusterInfo, because the Panel is empty and there is no "read" button.

SwoopX, 2020-07-29:
Already supported.

SwoopX, 2020-07-29:
Wait, the manufacturer code is different here.

PainElemental, 2020-07-29:
It is not in the list of supported devices.
It works using OpenHAB and deconz binding, but does not show up in Phoscon.

SwoopX, 2020-07-29:
It is in the list of supported devices

[image]

closed 2020-07-29 by SwoopX

PainElemental, 2020-07-29:
OK, sorry. I was looking for "Mijia Light Sensor"...

What do I have to do to get this show up in Phoscon? Raise an issue there: https://github.com/dresden-elektronik/phoscon-app-beta ?

SwoopX, 2020-07-29:
Either wait for DE to include the latest Phoscon changes in the deconz package or use the phoscon beta app from internet http://phoscon.de/pwabeta/login.html