dresden-elektronik/deconz-rest-plugin #3095: v2.05.79 crashes with SIGSEGV

ebaauw, 2020-07-29:
## Describe the bug
deCONZ v2.05.79 crashes with a segmentation fault (SIGSEGV).  I see nothing in the log that gives a clue why.

This happened three times today, once on my production system, and twice on one of my test systems.

## Steps to reproduce the behavior
Unknown.

## Environment
The production is running natively on a Raspberry Pi 3B+ running stretch, using a RaspBee with 0x26350500.
It's running a self-compiled REST API Plugin, with the changes from my latest PR, and the CLI plugin.

The test is running natively on a Raspberry Pi 4B running buster, using a RaspBee II with 0x26610700.  It's only running the stock plugins.  Not sure if it's related, but I had configured the RaspBee II as a router and had joined it to the network created by an eCozy Central Unit.

I haven't seen any crashes on my other two test systems, once mirroring production, and one mirroring the other test system, but with a ConBee II with 0x26580700.

### Core dumps
[production.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4997347/production.gz)

ebaauw, 2020-07-29:
And test:
[core-deCONZ-sig11-user1000-group1000-pid28580-time1596022376.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4997357/core-deCONZ-sig11-user1000-group1000-pid28580-time1596022376.gz)

ebaauw, 2020-07-29:
[core-deCONZ-sig11-user1000-group1000-pid28984-time1596024251.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4997369/core-deCONZ-sig11-user1000-group1000-pid28984-time1596024251.gz)

ebaauw, 2020-08-06:
Had another crash just now, on production, when trying to read the _IAS Zone_ attributes of a Samsung SmartThings multi sensor.
[core-deCONZ-sig11-user1000-group1000-pid17428-time1596697605.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5033159/core-deCONZ-sig11-user1000-group1000-pid17428-time1596697605.gz)

ebaauw, 2020-08-06:
And it's repeatable?!  Last log message is a report from the sensor's FC02 cluster, probably because I was shaking it to wake it.
```
Aug 06 09:13:17 pi2 deCONZ[7018]: 09:13:17:462 APS-DATA.indication srcAddr: 0x078d, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xFC02, lqi: 255, rssi: -43
Aug 06 09:13:17 pi2 deCONZ[7018]: 09:13:17:462 	asdu: 0c4112460a1400294f0112002916fe13002900fd
Aug 06 09:13:17 pi2 deCONZ[7018]: 09:13:17:463 Delay APS request diff = 376
Aug 06 09:13:18 pi2 systemd[1]: deconz-gui.service: Main process exited, code=dumped, status=11/SEGV
Aug 06 09:13:21 pi2 systemd[1]: deconz-gui.service: Unit entered failed state.
Aug 06 09:13:21 pi2 systemd[1]: deconz-gui.service: Failed with result 'core-dump'.
```

Next time, I got the _IAS Zone_ message:
```
Aug 06 09:18:06 pi2 deCONZ[12755]: 09:18:06:501 APS-DATA.indication srcAddr: 0x078d, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -58
Aug 06 09:18:06 pi2 deCONZ[12755]: 09:18:06:501 	asdu: 1899010000003001010000311500020000192000100000f00000000000000000
Aug 06 09:18:07 pi2 systemd[1]: deconz-gui.service: Main process exited, code=dumped, status=11/SEGV
Aug 06 09:18:10 pi2 systemd[1]: deconz-gui.service: Unit entered failed state.
Aug 06 09:18:10 pi2 systemd[1]: deconz-gui.service: Failed with result 'core-dump'.
```

The crash is on processing the _ZoneState_ value from the _Read Attributes Response_.  According to WireShark, the response is valid.
```
ZigBee Encapsulation Protocol, Channel: 25, Length: 53
IEEE 802.15.4 Data, Dst: 0x3ebd, Src: 0x078d
    Frame Control Field: 0x8861, Frame Type: Data, Acknowledge Request, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
    Sequence Number: 51
    Destination PAN: 0xe8dd
    Destination: 0x3ebd
    Source: 0x078d
    [Extended Source: SAMJIN_00:01:08:89:a5 (28:6d:97:00:01:08:89:a5)]
    [Origin: 177]
    TI CC24xx-format metadata: FCS OK
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x078d
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
    Destination: 0x0000
    Source: 0x078d
    Radius: 30
    Sequence Number: 235
    [Extended Source: SAMJIN_00:01:08:89:a5 (28:6d:97:00:01:08:89:a5)]
    [Origin: 177]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 18
    Frame Control Field: Profile-wide (0x18)
    Sequence Number: 18
    Command: Read Attributes Response (0x01)
    Status Record
        Attribute: ZoneState (0x0000)
        Status: Success (0x00)
        Data Type: 8-Bit Enumeration (0x30)
        ZoneState: Enrolled (0x0001)
```
I can read the attributes if the Heiman Siren's and Develco Smoke Sensor's _IAS Zone_ without any issue.

@manup, can you determine from the core dump whether the crash originates in the REST API plugin?

manup, 2020-08-06:
Thanks for for core dumps, I'll spin up a test pi to see at what point it happens.

Just had a idea: Since it might be in the plugin, it might be cool to add some instructions to compile the plugin in debug or provide debug symbols for the official release.

manup, 2020-08-06:
By running `gdb /usr/bin/deCONZ  core-deCONZ-sig11-user1000-group1000-pid28984-time1596024251`,
and then `bt` to get the backtrace.

It seems to be a null pointer deref in `handleThermostatClusterIndication()`.

[image]

manup, 2020-08-06:
The other core dumps don't load, perhaps because they are created with the self compiled plugin.
Can you please share the .so file?

I've checked the IAS zone code but couldn't find anything suspicious yet.

ebaauw, 2020-08-06:
Thanks, that's the old dump from the test system, isn't it?  The core dump from today shows a different trace:
```
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `/usr/bin/deCONZ --http-port=80 --dbg-info=2 --dbg-aps=2 --dbg-error=1'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x6bcf8e3c in ResourceItem::toNumber() const () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
[Current thread is 1 (Thread 0x76fe3a00 (LWP 12755))]
(gdb) bt
#0  0x6bcf8e3c in ResourceItem::toNumber() const () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#1  0x6bcc5e18 in DeRestPluginPrivate::updateSensorNode(deCONZ::NodeEvent const&) ()
   from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#2  0x6bcc8a0c in DeRestPluginPrivate::nodeEvent(deCONZ::NodeEvent const&) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#3  0x6bdb5ca0 in DeRestPluginPrivate::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
   from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#4  0x76184bfc in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#5  0x76f77fcc in deCONZ::ApsController::nodeEvent(deCONZ::NodeEvent const&) () from /usr/lib/libdeCONZ.so.1
#6  0x0007a1c4 in zmController::onApsdeDataIndication(deCONZ::ApsDataIndication const&) ()
#7  0x000c4880 in zmController::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
#8  0x76184bfc in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#9  0x000c2c0c in zmMaster::apsdeDataIndication(deCONZ::ApsDataIndication) ()
#10 0x00050c80 in zmMaster::processPacked(zm_command*) ()
#11 0x00051fa8 in zmMaster::onPacket() ()
#12 0x000c31fc in zmMaster::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
#13 0x76185a10 in QObject::event(QEvent*) () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#14 0x769ea408 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5
#15 0x769f260c in QApplication::notify(QObject*, QEvent*) () from /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5
#16 0x01646af0 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
I think it's dereferencing a bad `*ResourceItem` for `RConfigPending`:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/dd58a42044cb68bbe057f4bf7cfe2d6e07150a1e/de_web_plugin.cpp#L8070-L8076

manup, 2020-08-06:
Ah yes that could be, if item should be checked for `null` in the if statement.

ebaauw, 2020-08-06:
Added the check and no more crash.

closed 2020-08-06 by ebaauw

ebaauw, 2020-08-07:
> Just had a idea: Since it might be in the plugin, it might be cool to add some instructions to compile the plugin in debug or provide debug symbols for the official release.

@manup yes, seems like a good idea.  You might actually want to publish a version with debug symbols on the bete channel, and one without on the main channel.