dresden-elektronik/deconz-rest-plugin #3096: Failed to start Deconz Service

AustinErck, 2020-07-29:
## Describe the bug
Following [this](https://phoscon.de/en/conbee2/install#raspbian) guide for raspbian setup on a headless Raspberry Pi Zero W with a Conbee 2. The service fails to start.

## Steps to reproduce the behavior
1. Follow conbee2 install steps for raspbian
2. enable deconz systemctl service
3. disable deconz-gui systemctl service
4. start deconz service
5. Get following error
`● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2020-07-29 15:12:35 PDT; 25s ago
  Process: 12647 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 (code=exited, status=217/USER)
 Main PID: 12647 (code=exited, status=217/USER)`

## Expected behavior
Service to start

## Environment

 - Host system: Raspberry Pi Zero W
 - Running method: Raspbian
 - Firmware version: 82f9bb929ce2186eb1824178c1ae82902ad6275c
 - deCONZ version: 2.05.78
 - Device: ConBee II
 - Do you use an USB extension cable: yes

SwoopX, 2020-07-29:
Have you also stopped deconz-gui service. If yes, please ensure both services are stopped and run deconz as mentioned in ExecStart
`/usr/bin/deCONZ -platform minimal --http-port=80`

There should probably be some errors to spot.

AustinErck, 2020-07-29:
@SwoopX 
Both are stopped. Here is the output of the command:

`/usr/bin/deCONZ -platform minimal --http-port=80
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-austinerck'
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()`

SwoopX, 2020-07-30:
Ah, sorry. Made a mistake (it's already very late). The command should have been
`/usr/bin/deCONZ -platform minimal --dbg-error=2 --dbg-info=2 --http-port=80`

Might be a lot. In doubt, pipe the output to a file.

AustinErck, 2020-07-30:
@SwoopX  Thank you so much it was a port issue. How can I edit my service to use a different port? I tried the command below, but when I went to save it seemed to try and save it as a temporary file.
`systemctl edit --full deconz.service`

SwoopX, 2020-07-30:
Thought so :wink:

You need to use drop in files, so stuff doesn't get overwritten on reboot or update. I'll see if we may have some info around that, apart from google.

AustinErck, 2020-07-30:
@SwoopX Ok so the server seems to work, however the gateway has an unknown vendor and the conbee 2 doesn't show up on lsusb. Any wiki articles on how to debug this?

`austinerck@rasp:~ $ ls -l /dev/ttyACM*
ls: cannot access '/dev/ttyACM*': No such file or directory
austinerck@ rasp:~ $ sudo ls -l /dev/ttyACM*
[sudo] password for austinerck: 
ls: cannot access '/dev/ttyACM*': No such file or directory
austinerck@ rasp:~ $ sudo lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

SwoopX, 2020-07-30:
Is you Raspi sufficiently powered? It's also a classic to use the wrong micro USB ports for power and peripherals, so you might want to double check that as well.

Btw, the Conbee would use `/dev/ttyAMA0`.

AustinErck, 2020-07-30:
I'm using the most inner USB port marked "USB" on the pcb. For power I have a 9W amazon charger running to the outer USB port.

SwoopX, 2020-07-30:
Sry, that was garbage previously on the tty device.

Is the stick recognized on a different system? Is any other UUSB device recognized by the Rpi?

closed 2020-09-05 by stale[bot]