dresden-elektronik/deconz-rest-plugin #3097: Get Backup file via API

Jey-Cee, 2020-07-30:
## Feature request type
Extend REST API with capability of backup download

## Description
Actual it is possible to create a backup via the API, but it is only stored local. This makes it hard to store the file on a external Storage without hacks.

SwoopX, 2020-07-30:
@Jey-Cee Hey, wanna join our Discord? I'd like to keep in touch on API matters 😉

Mimiix, 2020-07-30:
@SwoopX You are ahead of me. I was about to Email Jey with a developer invite ;)

@Jey-Cee , if you are in, hit me up with a message.

justin-caldicott, 2023-11-26:
Looking at how Phoscon works, you can download a backup via `{host}/deCONZ.tar.gz`. This is both good and bad. Good to be able to download a backup via the API. Bad that there's no APIKEY validation here.

You need an APIKEY to export and create the backup, but once that's done, the backup download is wide open. Presumably the db contains the APIKEY list too.

Is this a known issue?