dresden-elektronik/deconz-rest-plugin #3098: Swann Home Key Fob: SWO-KEF1PA

crixyd, 2020-07-30:
## Device

- Product name: Smart Home Key Fob Remote Control
- Manufacturer: SwannOne
- Model identifier: SWO-KEF1PA
- Device type : Remote

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]

https://www.swann.com/au/swo-kef1pa
https://www.swann.com/media/wysiwyg/docs/en/Key_Fob_QSG.pdf

Mimiix, 2020-07-31:
This device was already requested apparently. I'll check with the devs if anyone would take it.

For reference: #1085

Smanar, 2020-07-31:
It s realy strange from https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1085 it seem this device is almost working but there is nothing in deconz about the cluster 0x0501

So I have just used information from previous issue to test.

Can you can try the code ? Procedure are on the github page, just chnage the step 1 by
git clone --branch  swann https://github.com/Smanar/deconz-rest-plugin.git

And you can skip "git checkout -b mybranch HEAD"

If I m right, you will have the device in the API
If the issue is right you will have 2 button working
And on log you will have "no button handler"with usefull data for the 2 others buttons.

crixyd, 2020-08-01:
Ah damn, my apologies guys, I didn't think to search the closed issues given that it's a currently unsupported device. 

I'll try running your branch Smanar, thank you.

Smanar, 2020-08-02:
Sorry, I have something wrong in my files (version mistake)
I will update the files tommorow.

Smanar, 2020-08-03:
Ok So I have big problem with my github install, impossible to start on a clean base
So the PR is strange but it s the V 79 base just with the device added in whitelist.

Like I have said before, It s just the line described in the other issue, I can add more, but it seem they are not usefull, if someone can test it ?

crixyd, 2020-08-05:
Hi Smanar, apologies for taking a while... I've had a couple of issues come up. Will be in touch with more info asap. Thanks!

crixyd, 2020-08-24:
Hi Smanar, I have just received my new Conbee II and will begin testing with that. I have tried to clone the swann branch, but receive the following error `fatal: Remote branch swann not found in upstream origin`. How should I proceed?

Smanar, 2020-08-24:
heeee, I have cleaned my fork recently and delete all old code.

So it s ok, I have remake the code.
After instaled the new code, you can include the device, from the other issue you will have 2 workings buttons.

For the 2 other I will explain later, but you will need to use debug.

crixyd, 2020-08-25:
Hi Smanar,

When you say to install the new code, I presume you mean in an upcoming version of the rest plugin? What version should I look out for? Once i've done that I'll debug for the other two buttons...

Thanks!

Smanar, 2020-08-25:
Just follow that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3098#issuecomment-668158588

It will make a lib version V2_05_80 but modified for your device.

closed 2020-09-29 by stale[bot]