dresden-elektronik/deconz-rest-plugin #3101: Aqara Motion Sensor trigger in Home Assistant (rtcgq11lm)

LeVuHoang, 2020-07-31:
Hi,
I don't know how to make this work, I even don't need LUX value, I only need to have a trigger when there is a motion.
This is how I setup in Home Assistant
```
above: 0
below: 1000
device_id: 5dc862c8f6964b699ad303fc087d4974
domain: sensor
entity_id: sensor.living_motion_sensor
for:
  hours: 0
  minutes: 0
  seconds: 0
platform: device
type: illuminance
```
I changed value of Above, Below, For, but even when LUX changed the value, it's not working
I also tried with State but doesn't work neither
```
entity_id: sensor.living_motion_sensor
platform: state
to: 'on'
```
There are two entities for this motion only
```
sensor.living_motion_sensor
sensor.living_motion_sensor_battery_level
```
So, how can I setup a trigger then I'll turn on a light?

This is my device
https://ibb.co/dcKCNzg

Deconz configuration
https://ibb.co/bFQz31J

Thanks

Mimiix, 2020-07-31:
Hi! 

Can you try to re-pair the device ? It seems it didn't pair correctly.

LeVuHoang, 2020-07-31:
Hi Mimiix,
I re-repaired the device and it's working properly.
Thank you

closed 2020-07-31 by LeVuHoang