dresden-elektronik/deconz-rest-plugin #3102: [Profalux shutters] Periodic movement

vanackej, 2020-07-31:
## Describe the bug
Every few minutes (30 to 60, I don't know exactly), all my Profalux shutters are doing a quick up/down movement (less than 1 second).
I have 4 shutters, and they are all doing the same movement at the same time.
If I turn the gateway down, the movement does not occur anymore, so the movement is related to the gateway.

## Expected behavior
No shutters movement when no action performed.

## Environment
 - Host system: Intel core i9 server
 - Running method: Proxmox VM, Debian OS
 - Firmware version: 26580700
 - deCONZ version: 2.05.79
 - Device: ConBee II
 - Do you use an USB extension cable: no

### deCONZ Logs

Provided here : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2739#issuecomment-660878263

## Additional context
To be sure, I also completely shut down my domotic solution (Jeedom) and my Wifi network, and the problem still occurs with the gateway alone.

vanackej, 2020-08-14:
Hello @Smanar, 
Back from holidays, I tried to disable the rest api part as you advised in [this comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2739#issuecomment-667174671).
The periodic movement does not happen when the rest API is disabled, so I'm pretty confident now that the issue comes from it.
In the attached logs, it occurred again with a single shutter in the zigbee network, around 15:04:28 (maybe during the 2 seconds before/after).
[deconz2.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5074759/deconz2.log)

Smanar, 2020-08-14:
I realy see nothing in your log about the profalux device, how many are paired during  ht previous log ?

> 0x20918A00000AFBA1
> 0x20918A00000AFBA0
> 0x20918A00000AFBA2

But It seem you have some upnp traffic, you haven't some device like alexa or google home that can use the API ?

Some other thing you can try.
- wich one parameter are you using for debuging (in the command line) ?
- on deconz, on the node title, on the first circle you can send again some zigbee request.
- try to use the API, but remove the device from it. Or include the device using deconz and not phoscon, like that the device will be in deconz but not in the api.

vanackej, 2020-08-15:
I deleted the 3 other shutters in the webapp, so for this log there is only 1 shutter left.

By the way, what happens when deleting the lights using the webapp ? Because my shutters have been also disassociated from their physical remotes when I removed them, so I think I have to perform a full reset (which is really a pain to do on those devices)

Here are the nodes in the deconz app. I don't understand what the other nodes are.
[image]

I'm pretty sure the issue is related to some internal timer/scheduled task in the gateway, as it occurs exactly every 30 minutes after the deconz app/gateway startup. If I restart the gateway at 12:00, it will happen at 12:30, 13:00, 13:30, ...

My command line is `/usr/bin/deCONZ --http-port=80 --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --dbg-aps=1 --dbg-http=1 > deconz.log &`. I don't know about all the possible dbg options,I just found it in an other issue comments.

Smanar, 2020-08-15:
Ha, I haven't thinked at remote ! But as they aren't in the API, I don't think they are guilties.

Your command line is perfect ^^, not possible having more informations, but I hope you have disabled it ATM ? (or you don't use SD card).

I m checking what can be trigger every 30 mn in deconz.  But about your unpn devices, you haven't something special ?

Smanar, 2020-08-16:
```
15:04:31:686 Idle timer triggered
15:04:01:686 Idle timer triggered
```

But on the code it happen more than every 30 mn, ca you check if you have it ONLY every 30 mn ?

vanackej, 2020-08-17:
No, really nothing specific regarding upnp devices. And the date/time of the problem does not match the upnp logs time.
I will set up an iptables rule in order to block all incoming upnp traffic, in order to be really sure it's not related.
And yes, I can confirm it happens only every 30 minutes, I reproduced several times, with multiple deconz restarts.
What do you mean by changing the time ? the time of what ?

vanackej, 2020-08-17:
I can confirm the problem is still happening after blocking all upnp traffic with iptables

Smanar, 2020-08-17:
So you have this log "Idle timer triggered" only every 30 mn ? and I think in same time than the VR is moving ?

vanackej, 2020-08-18:
I have the log every 30 seconds, not every 30 minutes, so I don't really think it is related.

BTW, there is a new dev version that have been release 2.05.80, maybe I can try to update (I assume all the Profalux-related changes have been included in this version)

Smanar, 2020-08-18:
Yep, the change have been validated.
But I still don't find an even that happen only 30 mn ....

Smanar, 2020-08-18:
@vanackej do you know how to use the API ?
To check the group JSON ? with an url like `http://IP:PORT/api/KEY/groups`

vanackej, 2020-08-19:
Yes, I know of course
I didn't create any group, so the response is empty.
Should I ?

Smanar, 2020-08-19:
Nope, was jusr to be sure the problem was not from an old scene.

Smanar, 2020-08-20:
Ok So they have said me to check the same fonction DeRestPlugin::idleTimerFired()

If you are agree, i can make a special version with 3/4 more debug lines because ATM I realy don't have idea for this bug.

vanackej, 2020-09-01:
Some updates after a few days of intensive testing, with and without your modified version :
- The issue still happen with your version
- I see "debug 1" a lot, but never seen "debug 2" log
- If I change the server time manually, I'm able to reproduce the issue. If I launch the deconz process at 5, and set the clock to 5:29, the shutter make the movement the minute after.
- It stops happening when the day change (i.e. after midnight the day the deconz process was started). Also able to reproduce by changing the server clock

Smanar, 2020-09-01:
Ha, nice, usefull.

vanackej, 2020-09-01:
Yes, if I start the deconz process at 23h, it will occur at 23:30 then no more.

Here are my latest "debug 1" logs. The interval is not always exactly the same, but it looks like it runs every 30 seconds most of the time.

```
01:43:27:830 Profalux debug 1
01:43:57:830 Profalux debug 1
01:44:27:829 Profalux debug 1
01:44:29:829 Profalux debug 1
01:44:30:829 Profalux debug 1
01:45:00:830 Profalux debug 1
01:45:30:830 Profalux debug 1
01:46:00:829 Profalux debug 1
```

vanackej, 2020-09-01:
I think I found a least where the problem is coming from (but still don't get why for now)
Each time the problem occurs, I see the log `send permit join, duration: 0`
The condition around permit join (permitJoin.cpp:245) is `if (!permitJoinLastSendTime.isValid() || (diff > PERMIT_JOIN_SEND_INTERVAL))`, the constant here being 1800 * 1000 ms=> 30 minutes!

vanackej, 2020-09-02:
I can confirm that the problem disappear when I make the permitJoin.cpp:245 contion always false.
I'm not familiar with Zigbee protocol, so what I say here is maybe incorrect. But from my understanding, when the shutter receive the permit join message, it confirms he is already member of the network by doing this quick movement.
It still does not explain why it stops happening after midnight.
Do you know why the REST gateway send a permit join every 30 minutes ? Is it a "standard" Zigbee behavior ? Everything seems to be ok for me with it being disabled

Smanar, 2020-09-02:
Hu ?? I need explanation for the calcul 

> 1800 * 1000 ms=> 30 minutes!

^^

BTW perhaps that can be usefull for you https://github.com/dresden-elektronik/deconz-rest-plugin/commit/e10b53002bda06158f69cc837772c22f9b29c8b0#diff-7eef30a0e3da84572a02ee75c64e5264

And yes shutter make this move everytime you set the permit join, but in my mind it was everytime with value not 0 ....

I m still searching explanation.

vanackej, 2020-09-02:
30 minutes = 30 x 60 seconds = 1800 secondes, then multiply by 1000 to convert into milliseconds ^^
So yes, that was the condition that triggered the permit join after 30 minutes. I even made custom builds with differents values (shorter ones), and it happenned as expected according to the new value.
I tried to build and run master branch (which includes the commit you mentionned), but the problem is the same.
Permit Join with 0 value is still emitted every 30 minutes, and each time the shutter acknowledge by moving up and down.

Smanar, 2020-09-02:
Lol, ha yes I m making reverse calcul, tired ....., but soon on holiday ^^.

I m asking to manup ATM, because I don't see the utility of this feature.

Smanar, 2020-09-04:
Ok so some news

>The periodic sending of permit join (=0) is on purpose and meant to ensure that the network is closed.
>The duration should be the value as given by the API, it's decremented automatically. There were fixes a couple of days back, so hopefully the current alpha should not show problems.

You can try the modification 3 posts above, but I m not sure it change something, your situation is special, and still no information for the reason it stop at midnight.

vanackej, 2020-09-06:
As I said in my previous message, I already tried the fix above, as I made some tests with master branch, that includes the fix we are talking about. Not better.

About why it stops at midnight, I looked at the code and I think it's pretty obvious: Time comparisons are all made using QTime objects, that only represents time elapsed since midnight, not including the date part. So, when  permitJoinLastSendTime is 23:50 and currentTime is 00:00, the diff is a negative value (-10 minutes in my exemple), and will never be more than PERMIT_JOIN_SEND_INTERVAL. 
permitJoinLastSendTime should be kept in a QDateTime object.I made the fix locally, and it's now ok. I can make a merge request with the fix.
The "funny" thing is that now, with the "fixed" version, my bug does not disappear after midnight anymore :sob:

Smanar, 2020-09-06:
Lol yes, It for that I prefer wait for an anwser from manup

vanackej, 2020-09-06:
Is there a channel on the discord server I could be directly involved in the discussion about the issue ?
What is surprising to me, is that according to git history, this bug after midnight is at least present for the last 7 years, and no one noticed. So I think this periodic sending of permit join = 0 is maybe not that usefull...

Smanar, 2020-09-06:
Lol, yes.
The discord server is here https://discord.gg/QFhTxqN

But I have already exposed your problem on dev channel, and 5mn ago have linked your post about the "midnight phase" at manup.

vanackej, 2020-09-06:
Ok, that's already the discord server I was aware of. I suppose there are some private channels for dev ? because I don't have access to the discussion you are speaking about.

Smanar, 2020-09-06:
Yep, it s a private channel.

Smanar, 2020-09-08:
So the "periodic call" is a security, and manup prefer to keep it, because ATM there is no way to be sure at 100% the network is closed.
No answer about the midnight problem, but I think it's better like that. If they correct this bug, it will be hard to use your device, atm you can use it with playing with time after inclusion.

The problem is this fonction affect the whole network, it s not a setting by device, so it mean if an user have 1 profalux and 50 other device, we need to disable it.

vanackej, 2020-09-08:
Well, my problem is that I restart my local services every night at 3am in
order to perform clean backups of my virtual machines, so even if the
date/time comparison issue is not fixed (meaning that the problem would
stop after midnight), I would still have my roller shutters issue because
of the nightly restart.

Would it be possible at least to have a command line option that would
allow to disable this periodic check ? I'm not really happy with the idea
of maintaining my own fork of the gateway just for that purpose.

Smanar, 2020-09-08:
The problem is I don't see what I can do to avoid a too much ugly hack on a sensible part.
There is nothing to do on hardware side ? (Firmware update ?)

Because this command is realy a "permit join off" no reason for the device react even it s a permit join command.

@manup if you have an idea ?

vanackej, 2020-09-08:
The problem is that Profalux choose to use the permit join 0 to acknowledge
that the shutter is a member of the network by doing this shutter movement.
They are not likely to chance it because it is made on purpose to confirm
remotes/shutters association. Anyway, my shutters are only 5 months old,
and there is no firmware update available.

I don't know what the zigbee spec say about it, but other zigbee keys (like
the Zigate one) don't have this issue with those shutters. So they don't
need to send permit join 0 commands periodically.

I think I will have to maintain a fork or maybe sell my Conbee and switch
to Zigate, just let me know if anything can be done. I can make a pull
request any acceptable solution is agreed too!

Smanar, 2020-09-14:
I have some return from profalux (thx to pipiche again ^^)
IDK if you are french ?

> Bonjour,
> 
> Ce comportement des moteurs n'est pas modifiable… Ces acquittements sont utilisés par les poseurs lors de l’installation des produits.
> Je comprends que les acquittements puissent vous poser problème. Mais j’ai plusieurs questions quant à votre application :
> Quel est l'intérêt de fermer un réseau régulièrement ?
> Quel est l'intérêt de fermer un réseau alors qu'il est déjà fermé ?
> 
> Bonne journée,

We can't use a "device detection" so for me the better way is using an option in the setting (and if possible an existing one)

vanackej, 2020-09-14:
Ok, thanks. That's what I was thinking.
I agree with you, the only short term solution is to provide an option to
disable this periodic network closing feature.
In the long term, I wonder why we are doing it, as your Profalux contact.

And, yes, I am french ^^

Smanar, 2020-09-14:
Ok so, it s ok, he is agreed.

So I can create a new setting in the API like "NotSecurisePermitJoin", this value will false by defaut and you can enable it using the API with a simple Curl Request.

The problem is the new version is realy close to go out, and not enought time for it, so If you are agree, I wait for the next version, clean my fork, and make the patch for the next version ?

vanackej, 2020-09-14:
> this value will be false by defaut and you can enable it using the API with

Yeah, if it's not ok for this version, no need to rush. When would the next
version be released ?
I don't really know how settings work in the gateway. Is it the only
possible way to pass parameters (curl) ? Isn't it possible to set it in a
config file or a startup parameter ?
Are those settings persistent across restarts ? Because I usually restart
my local infrastructure everyday to perform clean backups, and it would be
easier if I don't have to perform the action each time ^^
Let me know when you have a branch ready for test !
Thx

Smanar, 2020-09-14:
I think in 2/3 days you can find a new version.
He have proposed the command line, but for exemple in Jeedom or HA you don't have direct acess to the command line. And you need to use it all the time.
Using the API is easy, you are already using it, and yes the setting is permanent, and it will be in your future backup.
You can see for exemple a list of setting in the API at `http://IP:PORT/api/API_KEY/config`

Your modification is easy, you will have it 1/2 days after the version 81 on the github (even this version is not not official)

Smanar, 2020-09-15:
Ok so You can try this branch

```
     git clone --branch permitjoin https://github.com/Smanar/deconz-rest-plugin.git
     cd deconz-rest-plugin
     qmake && make -j2
     sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

You will a new option visible in tha api called "DisablePJSecurisation"

And can chnage it with

`curl -H 'Content-Type: application/json' -X PUT -d '{"DisablePJSecurisation": true}' http://IP:PORT/api/API_KEY/config`

But I haven't tested the code, so I can have mistake.

Smanar, 2020-09-20:
@vanackej Hello, you have tested the patch ?

vanackej, 2020-09-20:
Hi,
I just tested. The code does not compile, as a semicolon is missing in de_web_plugin_private.h
I fixed that, but the parameter does not seem to be updated. I added a log statement in permitJoin.cpp and gwDisablePJSecurisation value is always 0/false. I'm not familiar with config mecanism (API => database storage and live code value update), but something is wrong here

vanackej, 2020-09-20:
When I fetch back the config using the API, the DisablePJSecurisation is always false. It does not seem to be updated successfully, event if I got a success response during config PUT :
`[
    {
        "success": {
            "/config/DisablePJSecurisation": true
        }
    }
]`

Smanar, 2020-09-21:
Ok my bad, it s the problem with making code using only copy/paste.

both error are corrected.

vanackej, 2020-09-21:
Thanks, I'll have a look tomorrow and let you know

Smanar, 2020-10-02:
@vanackej you still need the option ? ^^

vanackej, 2020-10-03:
Yes, I do !
I have issues, the gateway does not seem to start when I replace the original rest plugin lib with your patched version.
Is it compatible with latest stable version (2.05.81) ?

Smanar, 2020-10-03:
I don't remember, I can update the code with the last one ( > 83), but I don't think using an old version change something for you.

But I m checking the code again, and I don't see why the gateway can be blocked ....

On jeedom, you can change the log level in a configuration part, can you take a look if you see something bad ?

vanackej, 2020-10-03:
I'm not using the jeedom managed gateway, I deployed a another one in a
separated VM to be able to manage versions upgrade by myself.
If you can update to latest release version, I can update and test today.

Smanar, 2020-10-03:
Done, I have updated the code with the last deconz version > 83.

You can miss a file, depend off the last beta version you have used

> All available button maps have been outsourced to an external JSON file named button_maps.json, which must reside in the same directory as the deCONZ database zll.db. The purpose is to speed up integration of new switches as users can now create it theirselves (given that some appropriate guidance is provided).

vanackej, 2020-10-04:
Hi,
I only have v 81 available, even installing the deconz-dev package. Where am I supposed to get v 83 ?
I tried to compiled and run the plugin with v81, it does not start

Smanar, 2020-10-05:
You are on the stable branch, you need to switch to the beta one for other versions if you are using packages.
else you have all versions here http://deconz.dresden-elektronik.de/raspbian/beta/

But I don't think lib V 83 need a decone V83 too.

On my side I m on deconz version 82, it have started.
[image]

vanackej, 2020-10-05:
On my side, the process starts, everything seems ok at first, but then unable to log in using the webapp.
If I restore the original libde_rest_plugin file, it works again.

Smanar, 2020-10-05:
I don't have same problem, but yes, I have too some problem with the last master, need to search why.

Smanar, 2020-10-07:
Ok so I have some problem specific to my hardware, other can compile code without problem.
On my side node title are missing, and websocket not working, but can use the webapp, and BTW make a boobkmark after the logging in deconz else you will have a new entry created every time you log in.

So I m using the version 83 beta, with the compiled lib else you need to copy the file button_maps.json in /usr/share/deCONZ/devices

vanackej, 2020-10-11:
Hi,
I've been struggling with my local setup, but it seems ok now. I just disabled periodic permit join in my local setup, I will let you know if it's ok

Smanar, 2020-10-11:
Hello,
Ok tell me if the setting is working, the date limit for PR is realy close.

vanackej, 2020-10-12:
Hi,

When DisablePJSecurisation is set, the behavior is ok.
But the parameter is not persisted across service restart. As soon as I restart the gateway, DisablePJSecurisation is back to false.

Smanar, 2020-10-12:
Ok, I m still not understanding ...

I need to recompile on my side to be sure, but nothing special on code.

You can see something in log ? the database is parsed at plugin start.

vanackej, 2020-10-13:
I can see 2 lines loading the configuration, with differents values :

12:37:01:264 Load config DisablePJSecurisation: false from db.
12:37:01:264 Load config DisablePJsecurisation: true from db.

There is typo : one securisation is with an uppercase, the other one is
with lowercase
The type is located in database.cpp : gwConfig["DisablePJsecurisation"] =
gwDisablePJSecurisation;

The good one is the one with the uppercase name I guess

Smanar, 2020-10-13:
Arf good catch, the code memorise it with the lowercase and the code work with the upper ...
It s working with this modification ?

[5 more replies]

closed 2020-10-15 by manup