dresden-elektronik/deconz-rest-plugin #3106: Aqara EU power plug SP-EUC01 (lumi.plug.maeu01) incorrect power readings

JeffL1, 2020-08-01:
This plug reads 11.000+ watts when a 1000 watt vacuum is turned on.
When I plug in a tiny 1-2 watt fan it reads 17 watt.
I think it is not a product defect because I have 9 of them.
I have tested the same fan and vacuum on other type Deconz powerplugs and I get correct readings. 
I have tested these plugs a couple of months ago on version .78. Today I updated the Deconz firmware to the latest version still wrong power readings.
Also I have been running the Aqara plug for months now on Homey (with build-in Zigbee) here I get correct power readings.
I looks like the readings of this type of plug are in Deconz about 10 times higher then reality.
Hope somebody has a solution. I love to to get all my Zigbee devices on Deconz. 
I have some additional screenshots below. If their is additional information needed, I am happy to help.

(Deconz is directly installed on PI 4B)
[image]
[image]

[image]

[image]

ebaauw, 2020-08-01:
Looking at the multiplier and divisor, the plug reports power in 0.1 mA.  Unfortunately, the REST API plugin currently doesn't consider these, because not all devices implement them, or implement them wrongly.  The conversion factor needs to be hard-coded, whitelisting the plug.

JeffL1, 2020-08-01:
@ebaauw Thank you for your quick reply.
Is that something that can be changed in for example a version .80/.81. Or is this something that only gets done with a major update of the REST API?

SwoopX, 2020-08-01:
I'd recommend to update deconz instead of the firmware and check the release notes on that occasion.

closed 2020-08-06 by SwoopX