dresden-elektronik/deconz-rest-plugin #3107: Busch-Jaeger wall-mounted transmitter 4-channel: Default scenes not created properly

klada, 2020-08-01:
## Describe the bug

- The battery-operated ZLL switch from Busch-Jaeger (6737/01-84) supports 8 scenes to be mapped to it's buttons
- However only 6 scenes are created, two of which have wrong names
- It seems like the scenes for the top rocker are not created. My guess is that deCONZ assumes that the the top rocker directly controls a relay/dimmer like the powered RM01, which is not the case

## Steps to reproduce the behavior

- Add a battery-operated Busch-Jaeger ZLL switch to the network (part number 6737/01-84)
- The following scenes are created in the REST-API:
  - `Scene 1` (second rocker, left button)
  - `Scene %u` (second rocker, right button, **buggy name**)
  - `Scene 3`  (third rocker, left button)
  - `Scene %u` (third rocker, right button, **buggy name**)
  - `Scene 5`  (fourth rocker, left button)
  - `Scene 6`  (fourth rocker, right button)
- The scenes for the top rocker are missing

## Expected behavior

- Scenes for the top rocker should be created
- Scenes should all be numbered from 1 to 8 (without the `%u`).

## Screenshots

- Buggy battery-operated 4 channel BJ switch (on the right) next to a working 2 channel BJ switch (wall-powered, on the left):

[image]

- Phoscon UI:
[image]

## Environment

 - Host system: PC
 - Running method: Debian package
 - Firmware version: 26580700
 - deCONZ version: 2.05.79
 - Device: ConBee II
 - Do you use an USB extension cable: yes

## Additional context

I think the scene behavior was just copied from the BJ ZLL switches which control a directly connected relais/dimmer through their top rocker. For them no scenes should be created. However the battery-operated switch is different. It does not control any device directly through the top rocker (because there is none attached), so the rocker needs to be mappable to scenes.

Probably also applies to the 1 and 2 channel versions of the switch. And maybe even to the powered switches which are not connected to a relais/dimmer, but just to a in-wall power supply.

Possibly related: #463

klada, 2020-08-01:
This is the group created for the switch. You can see that scenes with ID 1 and ID 2 are missing (not created):

```python
{'action': {'alert': 'none',
  'bri': 127,
  'colormode': 'hs',
  'ct': 0,
  'effect': 'none',
  'hue': 0,
  'on': False,
  'sat': 127,
  'scene': None,
  'xy': [0, 0]},
 'devicemembership': ['12', '13', '14', '15'],
 'etag': '4a8e9bc03eecc581c2247d654e9f4c58',
 'id': '9',
 'lights': ['3', '8'],
 'name': 'RB01 12',
 'scenes': [{'id': '3',
   'lightcount': 0,
   'name': 'Szene 1',
   'transitiontime': 10},
  {'id': '5', 'lightcount': 2, 'name': 'Szene 3', 'transitiontime': 10},
  {'id': '7', 'lightcount': 0, 'name': 'Szene 5', 'transitiontime': 10},
  {'id': '8', 'lightcount': 0, 'name': 'Szene 6', 'transitiontime': 10},
  {'id': '4', 'lightcount': 0, 'name': 'Scene %u', 'transitiontime': 0},
  {'id': '6', 'lightcount': 0, 'name': 'Scene %u', 'transitiontime': 0}],
 'state': {'all_on': False, 'any_on': False},
 'type': 'LightGroup',
 'uniqueid': 'd8:5d:ef:11:a1:00:28:xx'}
```

And these are the sensors 12 to 15:

```python
{'config': {'group': '9', 'on': True, 'reachable': True},
 'ep': 10,
 'etag': '4a8e9bc03eecc581c2247d654e9f4c58',
 'lastseen': '2020-08-01T16:59:37.926',
 'manufacturername': 'Busch-Jaeger',
 'mode': 1,
 'modelid': 'RB01',
 'name': 'Wandsender 4-Kanal',
 'state': {'buttonevent': 1002, 'lastupdated': '2020-08-01T16:59:37.923'},
 'swversion': '1.2.0',
 'type': 'ZHASwitch',
 'uniqueid': 'd8:5d:ef:11:a1:00:28:xx-0a-1000'}

{'config': {'group': '9', 'on': True, 'reachable': True},
 'ep': 11,
 'etag': 'e060829c928caebdb25520c2a9fbff0f',
 'lastseen': '2020-08-01T16:56:24.752',
 'manufacturername': 'Busch-Jaeger',
 'mode': 1,
 'modelid': 'RB01',
 'name': 'Wandsender 4-Kanal',
 'state': {'buttonevent': 4002, 'lastupdated': '2020-08-01T16:07:10.487'},
 'swversion': '1.2.0',
 'type': 'ZHASwitch',
 'uniqueid': 'd8:5d:ef:11:a1:00:28:xx-0b-1000'}

{'config': {'group': '9', 'on': True, 'reachable': True},
 'ep': 12,
 'etag': 'e060829c928caebdb25520c2a9fbff0f',
 'lastseen': '2020-08-01T16:56:24.753',
 'manufacturername': 'Busch-Jaeger',
 'mode': 1,
 'modelid': 'RB01',
 'name': 'Wandsender 4-Kanal',
 'state': {'buttonevent': 6002, 'lastupdated': '2020-08-01T16:07:12.405'},
 'swversion': '1.2.0',
 'type': 'ZHASwitch',
 'uniqueid': 'd8:5d:ef:11:a1:00:28:xx-0c-1000'}

{'config': {'group': '9', 'on': True, 'reachable': True},
 'ep': 13,
 'etag': '337ab9565da3b6eabcdc2dde71a8465a',
 'lastseen': '2020-08-01T16:56:45.823',
 'manufacturername': 'Busch-Jaeger',
 'mode': 1,
 'modelid': 'RB01',
 'name': 'Wandsender 4-Kanal',
 'state': {'buttonevent': 7002, 'lastupdated': '2020-08-01T16:56:45.820'},
 'swversion': '1.2.0',
 'type': 'ZHASwitch',
 'uniqueid': 'd8:5d:ef:11:a1:00:28:xx-0d-1000'}
```

The issue here probably is the way how the first color scene controller cluster (`0A`) is handled. My guess is that it is always ignored, which is only correct if a dimmer or on/off light cluster is found. If no light cluster is found the scenes should be created for the first scene controller (`0A`) as well.

SwoopX, 2020-08-03:
Hi. I quickly checked the code and it appears that you're right and wrong. Generally, endpoint `0x0A` is considered on the wireless switches, as endpoint `0x12` is missing there. However, for the scene creation, a `>` instead of `>=` is used, resulting in a skip of endpoint `0x0A` for the scene cluster binding.

klada, 2020-08-14:
@SwoopX Sounds like an easy fix then. Could you point me to the right direction what needs to be changed? I need to get this sorted out before I can fully migrate to DeCONZ.

SwoopX, 2020-08-15:
I've already asked @manup to take a look, since they have a device on-site. I only have the 4-gang weired.

SwoopX, 2020-08-24:
I just got a new control element today and played around with it a bit. Funny fact is that most of the BJ info material is wrong. The top rockers do not support storing and recalling scenes. However, this is described correctly in the manuals.

klada, 2020-08-24:
@SwoopX The battery-powered control element can store scenes for sure. That's it's only purpose :)

For the wall-powered versions I am not 100% sure. My impression was that there are two options for mains-powered control elements:

- If the control elements sits on a Busch-Jaeger relay or dimmer it's top rocker always controls the relay/dimmer. The top rocker cannot be used for scenes in this case.
- If the control element only sits on a Busch-Jaeger power supply the top rocker can also be used for storing/recalling scenes.

Which control element were you using? Did you place it on a plain power supply or a relay/dimmer?

But the issue I have is with the battery-powered one anyway. Is there anybody looking into this?