dresden-elektronik/deconz-rest-plugin #3108: Lost half of my devices

popy2k14, 2020-08-01:
## Describe the bug
I have restarted my raspberry pi4 today multiple times (nice shutdown/reboot with cmd - NO powerloss).
After that i have lost half of my ~100 devices (hue motion sensors, dimmer switch, ...)

## Steps to reproduce the behavior
unkown

## Expected behavior
not loose half of my devices!

## Environment
2.05.78
conbee II with latest firmware

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26580700
 - deCONZ version: 2.05.78
 - Device: ConBee II
 - Do you use an USB extension cable: yes - but never had an issue with that

## Fix
I have pulled a backup prior to restore and saw that the file was almost ~500kB in size!
Normaly my backups are ~50 kB.
i have fixed it by restoring my last backup which was ~50 kB from last week.
Now every device is up and running again.

I can provide the faulty ~500 kB backup if needed. But dont want to upload here because of sensible data in it (unique adresses...)

Mimiix, 2020-08-03:
Hi!

Any logs?

Most of the time, the devices recover in function and it can take up to one hour before it is fully up again.

popy2k14, 2020-08-03:
Sorry, no logs :-(

The downtime was about ~1h because i moved my raspbian installation to an SSD.
Ok, did'nt know that.
I have also pressed every switch, jumped before every motion sensor .... but non of the devices showd up in a few minutes.
After restoring the backup the was fully up in a few seconds.
Maybe just had to wait longer...

Will keep an eye on this, thx

Mimiix, 2020-08-03:
For next time:
- Just wait a few minutes. Also power cycling devices / clicking the pair/reset button helps.
- Aqara devices can take up to 2 hours to re-connect.

I'll close this issue for now. If it happens again, let me know :) I'm happy you solved it.

popy2k14, 2020-08-03:
Just let you know:

- Also power cycling devices / clicking the pair/reset button helps.
  Have done a reset on HUE swtiches (all 4 Buttons) -> did not help!

With my initial post (LOST) i'll mean lost -> the lost devices where'nt reported over api with their ID.
It was like they where never be paired!

When i'll got you right you mean the devices was just not online in the zigbee net for sending/receiving messages.
That was not the case. The where not present in phoscon/fhem (over rest API).

Mimiix, 2020-08-03:
@popy2k14 Aah! That changes things.

I've seen this hapening with a "group" of devices. Let's say: People losing their switches while having lights and sensors.

The backup just simply didn't restore them.  Does this sound suitable for your case?

popy2k14, 2020-08-03:
Yes, this was the case.
Some devices (mostly sensors) are just gone from phoscon/fhem.
There was just missing IDs in between other devices, which where present.
I have tried to repair one HUE dimmer switch and phoscon has created completly new ID,
like the switch with the same zigbee MAC was never paired (in the deconz database).

In my case restoring a backup solved the issue.

I have'nt enabled additional debug logs, but i can look if i can find some.
Can you please point me to the correct path to look for debug logs?

thx

Mimiix, 2020-08-03:
I'll bring this up with @manup in person.  He might request more information. 

Thanks!

popy2k14, 2020-08-04:
thx, i am here if you need any additional informations.