dresden-elektronik/deconz-rest-plugin #3111: 0% battery level on power-plugged Xiaomi Natural Gas Detector

mwhiegl, 2020-08-02:
## Describe the bug
Thanks to #1182 the Xiamoi Natural Gas Detector works with the REST API now. However even though it has no battery in Homebridge and HomeKit it shows as having a battery but with 0% level and not loading.

## Steps to reproduce the behavior
It should be the same for everyone with the sensor.

## Expected behavior
Several options that could behave differently/better. Simplest one actually not having any battery reference if there is no battery. If the battery reference is not avoidable having 100% level fixed when there is no battery would be closest to correct.

## Screenshots

[image]

## Environment

deCONZ
Raspberry Pi 3
Raspbian GNU/Linux Buster (10)

ConBee II on an extension cable
Firmware 26580700
deCONZ 2.05.79 22/05/2020

Homebridge:
Raspberry Pi 4
Raspbian GNU/Linux Buster (10)

### deCONZ Logs
I include the message from Deconz to Hombridge both on the Fire sensor which as I understand served as the template and has a battery and the Natural Gas sensor, which does not.
```
[8/2/2020, 11:39:41] [Deconz] ConBee: /sensors/20: warning: unknown ZHAFire sensor {"config":{"battery":100,"on":true,"pending":[],"reachable":true,"temperature":2800},"ep":1,"etag":"03ceb99a85c6f62eb7ed71b0a9790799","lastseen":"2020-08-02T10:39:30.583","manufacturername":"LUMI","modelid":"lumi.sensor_smoke","name":"lumi.sensor_smoke (2)","state":{"fire":false,"lastupdated":"2020-02-08T17:09:39.000","lowbattery":false,"tampered":false},"swversion":"20170314","type":"ZHAFire","uniqueid":"00:15:8d:00:03:46:5e:86-01-0500"}
[8/2/2020, 11:39:41] [Deconz] ConBee: /sensors/24: warning: unknown ZHACarbonMonoxide sensor {"config":{"battery":null,"on":true,"pending":[],"reachable":true,"temperature":5700},"ep":1,"etag":"787b2ee801998bfc5c597e4568feace6","lastseen":"2020-08-02T10:38:41.988","manufacturername":"LUMI","modelid":"lumi.sensor_natgas","name":"lumi.sensor_natgas","state":{"carbonmonoxide":false,"lastupdated":"2020-08-02T10:33:03.656","lowbattery":false,"tampered":false},"swversion":"10-19-2016","type":"ZHACarbonMonoxide","uniqueid":"00:15:8d:00:03:5b:10:d4-01-0500"}
```
## Additional context
The issue seems related to "battery":null which is not recognised as "not existent" but rather as "0%". 

I am actually not aware that there is a version of that sensor available with battery, but as there is a kind of reference to it in the product documentation, I'm not gonna state it as fact. But maybe other owners of that sensor can chime in.

I tried finding out if home bridge expects something else when battery is non-existent, but I didn't find anything, so my recommendation is to use 100% when it is directly plugged in.

SwoopX, 2020-08-02:
Hm, I've considered that while doing the integration. Apparently, there's yet another place to whitelist/blacklist.

ebaauw, 2020-08-02:
It would seem the natural gas sensor is wrongly exposed as CO sensor?

And both sensors expose `config.pending`, but no any settable `config` attributes?

If memory serves, the Heiman combustable gas sensor was also mains powered with no backup battery.

SwoopX, 2020-08-02:
@ebaauw 

> It would seem the natural gas sensor is wrongly exposed as CO sensor?

It identifies itself as carbon monoxyde sensor

> And both sensors expose config.pending, but no any settable config attributes?

Misused the `config.pending` back in the days to further stabilize IAS enrollment. Better, but not perfect.

ebaauw, 2020-08-02:
> It identifies itself as carbon monoxyde sensor

A Xiaomi device that wrongly identifies itself - that would be a first... not.

closed 2020-08-12 by manup