dresden-elektronik/deconz-rest-plugin #3112: The usb stick began to lose connection

VanGog86, 2020-08-02:
I have a problem. after updating home-assistant to 0.113.2, host_os to HassOS 4.11  and deconz firmware to 2.05.79. The usb stick began to lose connection with all zigbee devices very often. below I attach a log. Tell me how to fix.

18:56:20:644 delay sending request 212 dt 1 ms to 0x00158D0002237EBC, ep: 0x02 cluster: 0x0000 onAir: 1
18:57:03:176 don't close database yet, keep open for 900 seconds
18:57:03:230 Remove websocket 172.30.32.1:46374 after error Unknown error
18:57:03:231 New websocket 172.30.32.1:46546 (state: 3) 
18:57:03:248 Current channel 15
18:57:03:258 Device TTL 6774 s flags: 0x7
18:57:05:244 delay sending request 246 dt 0 ms to 0x00158D0002237EBC, ep: 0x03 cluster: 0x0000 onAir: 1

Smanar, 2020-08-02:
Can you try the USB cordon extension ?
Can you check if it s not USB disconnection ?

VanGog86, 2020-08-02:
this problem persists with both USB cordon and without it, now I have checked USB does not disconnect over time, no it does not disconnect. below I am attaching a picture that shows how often communication with devices is lost

[image]

Mimiix, 2020-08-03:
@VanGog86 This is not related to deconz or anything on our end: Websockets are opened by the client, not the server.

Your issue is: https://github.com/home-assistant/core/issues/37372

closed 2020-08-31 by stale[bot]