dresden-elektronik/deconz-rest-plugin #3115: Skip automatic channel change, TODO warn user

roflmao, 2020-08-04:
## Describe the bug

Log says Skip automatic channel change ever 10 seconds after changing channel from 11 to 15.
Seems like im running channel 15, getting counterwise messages in the logs.
```
 20:38:06:141 Current channel 15                                                                                                                                                                                   
 20:38:16:081 channel is 15 but should be 11, start channel change                                                                                                                                                 
 20:38:16:081 Skip automatic channel change, TODO warn user                                                                                                                                                        
 20:38:26:082 channel is 15 but should be 11, start channel change                                                                                                                                                 
 20:38:26:082 Skip automatic channel change, TODO warn user                                                                                                                                                        
 20:38:36:081 channel is 15 but should be 11, start channel change                                                                                                                                                 
 20:38:36:081 Skip automatic channel change, TODO warn user                                                                                                                                                        
 20:38:46:082 channel is 15 but should be 11, start channel change                                                                                                                                                 
 20:38:46:082 Skip automatic channel change, TODO warn user                                                                                                                                                        
 20:38:56:081 channel is 15 but should be 11, start channel change                                                                                                                                                 
 20:38:56:081 Skip automatic channel change, TODO warn user                                                                                                                                                        
 20:39:06:082 channel is 15 but should be 11, start channel change    
```

## Steps to reproduce the behavior

Unsure if its reproducable when i dont want to change channel more than necessary.

  1. Go to 'Phoscon'
  2. Click on 'Gateway, Advanced'
  3. Scroll down to 'Change zigbee channel', switch to 15.
  4. Observed error

## Expected behavior
Unsure if this is a problem or not, but output should mean something

## Environment

(Please make this copy/pastable by web interface)
 - Host system: PC
 - Running method: Marthoc docker
 - Firmware version: (2635050000)
 - deCONZ version: (2.05.79)
 - Device: (ConBee I )
 - Do you use an USB extension cable: yes

### deCONZ Logs

```
20:42:51:145 don't close database yet, keep open for 900 seconds                                                                                                                                                  
 20:42:56:081 channel is 15 but should be 11, start channel change                                                                                                                                                 
 20:42:56:081 Skip automatic channel change, TODO warn user                                                                                                                                                        
 20:42:57:525 0x7CB2 seems to be a zombie recv errors 6                                                                                                                                                            
 20:42:57:525 Node zombie state changed 0x00178801035617db                                                                                                                                                         
 20:42:58:090 don't close database yet, keep open for 900 seconds                                                                                                                                                  
 20:42:58:703 ZCL attribute report 0x00158D0001D555FF for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000                                                                                           
 20:43:00:159 ZCL attribute report 0x00158D0001F55A51 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000                                                                                           
 20:43:00:191 ZCL attribute report 0x00158D0001F55A51 for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000                                                                                           
 20:43:00:223 ZCL attribute report 0x00158D0001F55A51 for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000                                                                                           
 20:43:05:137 don't close database yet, keep open for 900 seconds                                                                                                                                                  
 20:43:06:082 channel is 15 but should be 11, start channel change                                                                                                                                                 
 20:43:06:082 Skip automatic channel change, TODO warn user                                                                                                                                                        
 20:43:06:110 GW firmware version: 0x26350500                                                                                                                                                                      
 20:43:06:110 GW firmware version is up to date: 0x26350500                                                                                                                                                        
 20:43:06:161 Current channel 15                                                                                                                                                                                   
 20:43:06:193 Device TTL 5380 s flags: 0x7                                                                                                                                                                         
 20:43:10:628 ZCL attribute report 0x00158D0001F55A51 for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000                                                                                           
 20:43:10:628 0x00158D0001F55A51 extract Xiaomi special attribute 0xFF01                                                                                                                                           
 20:43:10:628     01 battery 2965 (0x0B95)                                                                                                                                                                         
 20:43:10:628     04 unknown 5032 (0x13A8)                                                                                                                                                                         
 20:43:10:628     05 RSSI dB (?) 10486 (0x28F6)                                                                                                                                                                    
 20:43:10:628     06 LQI (?) 4294967296 (0x0100000000)                                                                                                                                                             
 20:43:10:628     64 temperature 2293                                                                                                                                                                              
[... cut ...]
 20:43:36:081 channel is 15 but should be 11, start channel change                                                                                                                                                 
 20:43:36:081 Skip automatic channel change, TODO warn user                                                                                                                                                        
```

## Additional context
IKEA Tråfri/Xiaomi sensors, about 15-20 devices total

SwoopX, 2020-08-04:
What does Phoscon say under _Gateway -> Advanced_ for the current channel? I understand channel 15 was not working for you?

roflmao, 2020-08-05:
A couple of weeks ago i changed from channel 11(iirc) to channel 15. 
Since then i have had these messages. Phoscon, Gateway, Advanced says Zigbee channel 15, but the logs keeps telling me something is up. 
I have strange behaviour of old sensors coming back after a while after deleting as well (thought adding this in another bug report, but may be relevant?)

SwoopX, 2020-08-05:
Thanks, that's interesting. I'll see if I can reproduce this somehow. If it's currently just an annoyance, then we're good for the moment.

Devices coming back when previously deleted in deconz GUI is normal. The mechanism behind that is a bit delicate but I also got some tests in this regard on my list. No need for a seperate issue.

roflmao, 2020-08-05:
I also have an issue currently where added devices suddenly disappear as well. I will try to reproduce, what kind of logs do you want?
I use docker marthoc image with the following envs:

```
DEBUG_INFO=1
DEBUG_APS=0
DEBUG_ZDP=0
DEBUG_OTAU=1
DEBUG_ZCL=0
```

Mimiix, 2020-08-05:
@roflmao To keep issues on subject, I'd like you to open a seperate issue for that. 😄

roflmao, 2020-08-05:
@Mimiix Sure thing, just wanted a recommandation for which debug data to enable before reproducing

SwoopX, 2020-08-05:
`DEBUG_INFO=2`

closed 2020-09-03 by stale[bot]

m0wlheld, 2021-01-08:
Error / warning is still existing. I have that message in my deCONZ logs, almost every 10 seconds.

```
deconz        | 22:43:44:092 dev /dev/ttyAMA0
deconz        | 22:43:44:093 GW firmware version: 0x26690700
deconz        | 22:43:49:009 channel is 11 but should be 15, start channel change
deconz        | 22:43:49:009 Skip automatic channel change, TODO warn user
deconz        | 22:43:54:107 dev /dev/ttyAMA0
deconz        | 22:43:54:108 GW firmware version: 0x26690700
deconz        | 22:43:59:010 channel is 11 but should be 15, start channel change
deconz        | 22:43:59:010 Skip automatic channel change, TODO warn user
deconz        | 22:44:04:103 dev /dev/ttyAMA0
deconz        | 22:44:04:104 GW firmware version: 0x26690700
deconz        | 22:44:09:009 channel is 11 but should be 15, start channel change
deconz        | 22:44:09:009 Skip automatic channel change, TODO warn user
deconz        | 22:44:14:100 dev /dev/ttyAMA0
deconz        | 22:44:14:100 GW firmware version: 0x26690700
```

deCONZ running as Docker container on a Raspberry Pi 4B 4GB, using the Raspbee II add-on card. According to the Phoscon web app, the current channel is 11. Version infos are "2.07.01 / 8.12.2020" and firmware 26690700

I am using deCONZ with the Home Assistant integration, since I have recently bought some [Innr RB 265](https://zigbee.blakadder.com/Innr_RB_265.html) bulbs, which work with deCONZ, but not with the "native" ZHA integration for Home Assistant. 

I have ZHA integration used before, it uses channel 15 by [default](https://www.home-assistant.io/integrations/zha/#defining-zigbee-channel-to-use). Also, the non-Innr devices (mostly OSRAM/Ledvance) where assigned to ZHA integration.

Maybe there's something left from ZHA on the Raspbee II memory? Or could it be a connected device, that causes this?

davewood, 2021-02-03:
i am running deCONZ in docker and i see the same error/warning