dresden-elektronik/deconz-rest-plugin #3116: Xiaomi power plug (lumi.plug.mmeu01) not showing power consumption (again?)

ateuchtmann, 2020-08-05:
I have recently bought a Xiaomi power plug (lumi.plug.mmeu01) and I am experiencing the (known) issue ( #1513 ) that the power plug is not showing/recording the power consumption (kWh) correctly anymore. The recording was working fine until I have restarted Hassio for the first time after adding the power plug to the system. Afterwards the power consumption value was showing up as "unknown".

 - Host system: Raspberry Pi (HassOS 4.11)
 - Device: Phoscon RaspBee II
 - Firmware version: (26520700)
 - deCONZ version: (2.05.79)

Any help is appreciated but please be indulgent to me because I am new to Raspberry/Hassio/Deconz things. Thanks in advance.

SwoopX, 2020-08-05:
It's already known and will be fixed in the next version.

Mimiix, 2020-08-05:
@SwoopX Can you assign a pull request to this issue so it closes automatically?

ateuchtmann, 2020-08-05:
> It's already known and will be fixed in the next version.

Oh sorry for the hassle. I thought of another thread mentioning that this was fixed in .79

SwoopX, 2020-08-05:
The error was introduced at a different place in the code and applies for all Xiaomi devices providing consumption values.

cwildfoerster, 2020-08-11:
> > It's already known and will be fixed in the next version.
> 
> Oh sorry for the hassle. I thought of another thread mentioning that this was fixed in .79

Running 2.05.79 noticed the consumption sensor state is unknown since a couple of days....

closed 2020-08-12 by manup

3bbing, 2021-01-14:
Is this supposed to show up in Phoscon too? So far in PhosCon I only see the plug as light "LUMI Adapter Plug" with no power metering. The API reports it as sensor too with metering (although update interval seems to be long 3 minutes)
`{"config":
{"on":true,"reachable":true,"temperature":null},
..,
"manufacturername":"LUMI",
"modelid":"lumi.plug.mmeu01",
"name":"Adapter plug",
"state":{
"current":null,
"lastupdated":"2021-01-14T14:24:22.812",
"power":1658,
"voltage":null},
"swversion":"09-06-2019",
"type":"ZHAPower",
"uniqueid":"..."}`

cleversmart, 2021-03-22:
running deconz 2.9.3, LUMI Adaper Plug don't show power metering. How does it work ? Is it supported jet ?
[image]

lmanoarchilla, 2021-10-14:
Hi!, i have the same problem with two different smartplugs that are supported with Deconz, Xiaomi and philips hue lom001. Xiami smartplug sometimes show power comsuption and sometimes not. Philips never shows comsuption.

{
  "etag": "8f48aa1d5164d1a8cbf81f9e0d26291a",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2021-10-14T10:10Z",
  "manufacturername": "LUMI",
  "modelid": "lumi.plug.mmeu01",
  "name": "Smart plug 8",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": "09-06-2019",
  "type": "Smart plug",
  "uniqueid": "04:cf:8c:df:3c:7b:02:d1-01"
}

{
  "etag": "0057692b26b035975dcd34ab6fc6b2f5",
  "hascolor": false,
  "lastannounced": "2021-10-14T10:10:45Z",
  "lastseen": "2021-10-14T10:10Z",
  "manufacturername": "Philips",
  "modelid": "LOM001",
  "name": "On/Off plug-in unit 7",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": "1.55.10_r29112",
  "type": "On/Off plug-in unit",
  "uniqueid": "00:17:88:01:08:61:34:14-0b"
}