dresden-elektronik/deconz-rest-plugin #3117: Trust ZYCT ‑202 - different Manufacture code

opechunka, 2020-08-05:
Manufacture code  on my device is  0x122d. 
[image]

Mimiix, 2020-08-05:
Hi @opechunka 😄 
I am tempted to close this issue, as you left out vital information as per requested in the bug report template.  Please add this information.

Thanks!

SwoopX, 2020-08-06:
From what I've read, the devices are a little bit stubborn. Try entering sensor search in Phoscon and then read the simple descriptors (from the bullet you've marked). Please make sure the device is awake, so press a button right before reading the descriptors.

The current code shouldn't require any code changes, as there's also a check on the MAC address prefix, which hasn't changed in your case.

opechunka, 2020-08-09:
After I've removed all routers and left just RaspBee, I've managed to add ZYCT-202.  It works, thank you.

closed 2020-08-09 by opechunka