dresden-elektronik/deconz-rest-plugin #3118: LEDVANCE Edge-Lit Undercabinet TW showing up as RGB

ryancasler, 2020-08-05:
This device is just tunable white but it is showing up in the Phoscon app as an RGB light.  Not even RGBW.  It had been working before but now it seems to miss-identify them.

Mimiix, 2020-08-05:
@ryancasler I am closing this instantly. You haven't used the template at all and vital information is missing. Please open a new issue with the requested information.

ryancasler, 2020-08-06:
Really?   You need a screenshot of this?   Okay...I will do it.  But you also better fix it in the next release too.

Mimiix, 2020-08-07:
Missclicked. No, i don't need a screenshot, i do need the version details. 
> But you also better fix it in the next release too.
You might have just gotten yourself in a later release.

closed 2020-08-07 by Mimiix

Mimiix, 2020-08-07:
For others finding this: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3124