dresden-elektronik/deconz-rest-plugin #3124: LEDVANCE Edge-Lit Undercabinet TW Not Pairing as TW

ryancasler, 2020-08-06:
## Describe the bug
This device is just tuneable white but it is showing up in the Phoscon app as an RGB light. Not even RGBW. It had been working before but now it seems to miss-identify them.

## Steps to reproduce the behavior
Add the device. 
Try to click on Color Temperature 
Nothing happens

## Expected behavior
The light show up at tuneable white (color temp) not at RGB.

## Screenshots
[image]

## Environment
Host : Raspbian Buster
Method: Raspbian
Firmware: 26350500
Device: Conbee
deCONZ: 2.05.79
No extention cable

### deCONZ Logs
If you want me to grab the logs when the device is paired, let me know.  
## Additional context

ryancasler, 2020-08-06:
This is how they are identified in the Phoscon App:

[image]

Mimiix, 2020-08-07:
Then it probably is a phoscon issue. If you say the rest api can control it correctly. 

See, with this info it's much better to answer your question ;)

Mimiix, 2020-08-07:
Are you able to confirm if it works properly using the RestAPI? https://dresden-elektronik.github.io/deconz-rest-doc/ 

If that does work, let me know and i move this issue to the Phoscon repo (this saves you from creating _another_ issue 😉).

ryancasler, 2020-08-07:
Actually, it's more complicated than that.  I can change the Color Temp from the rest API but it's backwards.  Setting "ct:":450 results in a yellower light than "ct::270, which, as I understand it should be a cooler or bluer light.

ebaauw, 2020-08-08:
No, it should not.  `ct` is in mired.

Mimiix, 2020-08-08:
So as far as I'm concerned this is a issue with phoscon. I'll move this over to their repo.

@ryancasler Despite your disrespect against me, I forwarded it to their dev to give it priority.

_Be the change you want to see in this world. _