dresden-elektronik/deconz-rest-plugin #3125: Sunricher Remote control ZGRC-KEY-13 / same as iCasa ICZB-RM11S

rweise, 2020-08-07:
i have both black remote controls with one pair Mater ON/OFF , 4 pairs ON/Off and S1 and S2 Button.
the iCasa is not shown in Phoscon after Installation there was only the Message that it was fine.
the sunricher is shown as a switch in Phoscon.
In IOBroker, Icasa has 1 Buttonevent and shows there a different Buttonevent for Pairs 1-4 and S1 S2. 
sunricher has 4 different sensors and will only have a value for keyevent for Pair 1 that is the same as master ON/OFF. 
Both have the same Firmware! 
[image]
[image]

Here are the screenshots
[image]
[image]

alexdepalex, 2020-08-07:
fyi: The remote works perfectly fine in hassio when using events.

SwoopX, 2020-08-07:
Seems to be the same as #2857 and #1256. The icasa and the Sunricher have been implemented differently though. I'll see that we can streamline that.

closed 2020-08-12 by manup