dresden-elektronik/deconz-rest-plugin #3126: Error in Homebridge.log // Phoscon GW: websocket communication error 127.0.0.1:443: cannot connect: ECONNREFUSED

receens2k, 2020-08-08:
Hello folks,

I am experiencing a strange error in homebridge.log. I'm running homebridge on a Raspberry Pi 3B+ with the RaspBee II on itself. The gateway itself is working, but I want to get rid of the error:

```
[2020-8-8 14:26:07] [Phoscon GW] Phoscon GW: websocket communication error 127.0.0.1:443: cannot connect: ECONNREFUSED
[2020-8-8 14:26:37] [Phoscon GW] Phoscon GW: websocket communication error 127.0.0.1:443: cannot connect: ECONNREFUSED
[2020-8-8 14:27:07] [Phoscon GW] Phoscon GW: websocket communication error 127.0.0.1:443: cannot connect: ECONNREFUSED
[2020-8-8 14:27:37] [Phoscon GW] Phoscon GW: websocket communication error 127.0.0.1:443: cannot connect: ECONNREFUSED
[2020-8-8 14:28:07] [Phoscon GW] Phoscon GW: websocket communication error 127.0.0.1:443: cannot connect: ECONNREFUSED
[2020-8-8 14:28:37] [Phoscon GW] Phoscon GW: websocket communication error 127.0.0.1:443: cannot connect: ECONNREFUSED
[2020-8-8 14:29:07] [Phoscon GW] Phoscon GW: websocket communication error 127.0.0.1:443: cannot connect: ECONNREFUSED
[2020-8-8 14:29:37] [Phoscon GW] Phoscon GW: websocket communication error 127.0.0.1:443: cannot connect: ECONNREFUSED
[2020-8-8 14:30:07] [Phoscon GW] Phoscon GW: websocket communication error 127.0.0.1:443: cannot connect: ECONNREFUSED
```

The config for RaspBee II for homebridge looks like following. I also tried the IP address of the Raspi in the host-section as well as 127.0.0.1. 
```
        {
            "platform": "Hue",
            "name": "Phoscon GW",
            "users": {
                "USER": "SUPERSECRET"
            },
            "anyOn": true,
            "forceHttp": true,
            "host": "localhost:8090",
            "nativeHomeKitLights": true,
            "nativeHomeKitSensors": true,
            "lights": true,
            "sensors": true,
            "groups": true,
            "group0": true,
            "scenes": true,
            "nupnp": true,
            "resource": true,
            "excludeSensorTypes": [
                "CLIPPresence",
                "Geofence"
            ]
        }
```

http://192.168.178.62:8090/api returns only:
```[{"error":{"address":"/","description":"unauthorized user","type":1}}]```

Any ideas how to proceed? Thanks a lot in advance!

ebaauw, 2020-08-08:
Please attach homebridge.log and the debug dump file homebridge-hue creates on startup.  See https://github.com/ebaauw/homebridge-hue#troubleshooting.

Are you running a firewall, or another (web) server at port 443?

receens2k, 2020-08-08:
there is CUPS on 631, pihole on 80 and Phoscon on 8090. Nothing running on 443:

```
pi@piforce1:~ $ netstat | grep :443
pi@piforce1:~ $
```

here are the logs: 

[homebridge.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5045289/homebridge.log)
[homebridge-hue.json.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5045290/homebridge-hue.json.gz)

receens2k, 2020-08-08:
iptables config:

```
iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination
REJECT     tcp  --  anywhere             anywhere             tcp dpt:https reject-with tcp-reset
REJECT     udp  --  anywhere             anywhere             udp dpt:80 reject-with icmp-port-unreachable
REJECT     udp  --  anywhere             anywhere             udp dpt:443 reject-with icmp-port-unreachable
ACCEPT     tcp  --  localhost            anywhere             tcp dpt:http
ACCEPT     tcp  --  localhost            anywhere             tcp dpt:https
ACCEPT     tcp  --  localhost            anywhere             tcp dpt:http
ACCEPT     tcp  --  localhost            anywhere             tcp dpt:https

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
```

receens2k, 2020-08-08:
iptables config, if that helps:

```
iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination
REJECT     tcp  --  anywhere             anywhere             tcp dpt:https reject-with tcp-reset
REJECT     udp  --  anywhere             anywhere             udp dpt:80 reject-with icmp-port-unreachable
REJECT     udp  --  anywhere             anywhere             udp dpt:443 reject-with icmp-port-unreachable
ACCEPT     tcp  --  localhost            localhost            tcp dpt:http
ACCEPT     tcp  --  localhost            localhost            tcp dpt:https
```

receens2k, 2020-08-08:
#Update: 
understanding/knowledge of how iptables rules are "prioritized" would be good. 
Fixed the error by rearranging the order of the rules to:

```
iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination
ACCEPT     tcp  --  localhost            localhost            tcp dpt:http
ACCEPT     tcp  --  localhost            localhost            tcp dpt:https
REJECT     tcp  --  anywhere             anywhere             tcp dpt:https reject-with tcp-reset
REJECT     udp  --  anywhere             anywhere             udp dpt:80 reject-with icmp-port-unreachable
REJECT     udp  --  anywhere             anywhere             udp dpt:443 reject-with icmp-port-unreachable

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
``` 

But thanks anyway, have a nice weekend! :) 👍

closed 2020-08-08 by receens2k