dresden-elektronik/deconz-rest-plugin #3128: "Leichen" bei den Objekten - wie loswerden?

Michaelnorge, 2020-08-09:
Ich benutze eine Raspberry mit Conbee-II, Phoscon/Deconz und ioBroker.

Ich hatte Rolläden von IKEA verbunden, war nicht zufrieden und hab sie wieder gelöscht.
In der Phoscon-App sind die Rolläden verschwunden, aber unter ioBroker sehe ich noch Einträge der Rolläden.
Komischer Weise befinden sich diese Einträge bei den Sensoren!!

Ich nutze das System headless und hab wenig Ahnung von Linux.
Kann man per Kommando vielleicht eine Liste abrufen und unerwünschte Einträge manuell löschen?

[image]

Mimiix, 2020-08-09:
Hi there

As this git is English only, I am closing this issue.

Please use Google translate to help out or contact support directly.

closed 2020-08-09 by Mimiix

Jey-Cee, 2020-08-09:
@Michaelnorge https://github.com/iobroker-community-adapters/ioBroker.deconz/issues or forum.iobroker.net is the right place for your question.