dresden-elektronik/deconz-rest-plugin #3129: Samsung GP-U999SJVLAEA SmartThings Multipurpose sensor vibration not working

guytaylor, 2020-08-10:
Sorry if it's poor form to re-open a closed issue but I am getting this bug so I thought I would.
The old bug is here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1848

I set up Samsung GP-U999SJVLAEA SmartThings Multipurpose with deconz and the vibration sensor appears but never seems to trigger, even if I shake the unit.

@ebaauw, did you ever have any luck in getting this working? If there is anything I can do to assist please ask.
Thank you

ebaauw, 2020-08-10:
See #1848, the sensor is working fine for me.

Does the `state` of the `ZHAVibration` `/sensors` resource change when you read the attributes of the 0xFC02 cluster in the GUI while shaking the sensor?

guytaylor, 2020-08-11:
Hi, 
I've included a pic of what I think you asked (sorry, I'm new to deconz!).
When I click the read button on the left, and tapped the sensor, none of the value changed.

[image]

ebaauw, 2020-08-11:
It might take a couple of attempts.

If the sensor really doesn’t return any values here, I’m afraid it’s an issue with the sensor itself and there’s nothing we can do in deCONZ.

guytaylor, 2020-08-13:
I removed the sensors and re-added it. and this time it is now working. thanks

closed 2020-08-13 by guytaylor