dresden-elektronik/deconz-rest-plugin #3130: Connection losses every few days are back since newest beta

Claus1985, 2020-08-11:
Hi Guys,
Yesterday I updated my raspberry to the newest deconz release (beta).
Since then I got a problem back, which I thought had been solved for good in the last release.
Every few days (e.g. I installed the update on Sunday and today morning I got the problem) deconz loses the connection to all sensors (I use mainly Xiaomi and Philips hue motion sensors as well as Xiaomi temperature sensors) and I don´t get updates about the state anymore. When I restart my raspberry, everything works, as it should. I get updates and everything works fast and responsive.
I had this problem for quite a while but it was solved with the howling wolf release that was accompanied by the new conbee 2 firmware.

When I updated my system, I forgot that I had still put in the beta channel for updates.
I will reinstall of the stable howling wolf release to fix my issue but wanted to write the ticket anyway in order to get a permanent fix.

Thx and brgds,

Claus

## Environment
 - Host system: Raspberry Pi 2B+
 - Running method: Raspbian
 - Firmware version: 26580700
 - deCONZ version: 2.05.78/22.5.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes

Mimiix, 2020-08-11:
@Claus1985 Can you provide a view of your network map and logs :)?

Claus1985, 2020-08-11:
Hi Dennis,

Since I´m using the headless version of deconz I don´t know how to get the network map. Can you give me a hint? 
Regardings log, I just activated logging because I thought the issue was fixed so nothing interesting so far.

Thx and brgds,

Claus

Mimiix, 2020-08-11:
sudo systemctl stop deconz-gui
      /usr/bin/deCONZ --dbg-info=2 --dbg-error=2 > debug

Or adding --dbg-info=2 --dbg-error=2 > debug
 to the start line. 

Make sure to know where the debug file is placed.

Claus1985, 2020-08-11:
Hi Dennis,
console tells me could not connect to display > aborted when I enter the command you send me.
Thx,
Claus

SwoopX, 2020-08-11:
Do the following instead:

```
sudo systemctl stop deconz
/usr/bin/deCONZ -platform minimal --dbg-info=2 --dbg-error=2 > debug
```

Claus1985, 2020-08-11:
When doing this I get the following information:
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()

thedannymullen, 2020-08-11:
> since I´m using the headless version of deconz I don´t know how to get the network map. Can you give me a hint?

@Claus1985 if you have a linux box with a graphical interface, you can run the deconz gui by doing the following.  

1. ssh -X user@host
2. sudo systemctl stop deconz
3. type deCONZ at the prompt.  

If the display is mapped and shared properly the deconz gui will launch.  This is how I do it, using ubuntu desktop tied to my headless pi.

Claus1985, 2020-08-27:
Hey Smanar,

thx for the information the function is awesome! 
Anyways since the first incident I had no further problems or at least no severe ones.
Sometimes the devices react slower than usual but besides that everything is fine.
Fingers crossed but this seems to have been a one-time thing.

Claus

Mimiix, 2020-09-23:
@Claus1985 Why is this re-opened?

Claus1985, 2020-09-23:
Hi Guys,

sadly the problem is back. Now it happens nearly every day (at least in the last few days).
Yesterday I created a debug report to identify the problem.
Is there a way to send you the debug report via mail (for data security resons)?

Thx and brgds,

Claus

Mimiix, 2020-09-23:
@Claus1985 You can post most of it here. There is not a lot of info that can harm

If you wish, send it to me in private on discord.

closed 2020-10-25 by stale[bot]