dresden-elektronik/deconz-rest-plugin #3131: Calibrated Aqara Temperature/Humidity/Pressure Sensors

tony563, 2020-08-11:
I’m using Raspbee II, with a Raspberry Pi 4B. I have some Aqara Temperature/Humidity/Pressure Sensors paired with my Raspbee II. I’m also using HASSIO

The readings which I am getting for these sensors, are a bit inaccurate. Is it possible to calibrate these sensors?

When I was using Homebridge, I was able to do it, via either the Eve/Home+ apps; but not since I moved to HASSIO. 

Regards,
Tony. 

## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II

### deCONZ Logs

## Additional context

Mimiix, 2020-08-11:
In HA you can try to make a sensortemplate. In that way you can edit a bit. Please see: 
https://community.home-assistant.io/t/adding-offset-to-sensor-data/20382

In deCONZ, i am not sure if this is possible.

Smanar, 2020-08-11:
We can't use the field "offset" in the JSON device ?

wvuyk, 2020-08-11:
I am using the JSON field offset (for temperature) here actively. It works like a charm indeed.

tony563, 2020-08-12:
Thank you, for all of your replies. I’ve tried setting up the dummy sensor, but I don’t understand which bits of the example, I can change for the my sensors. 

I’ve also used the curl/put command, to set the Offset value, but it says: “Connection Refused”.

I would appreciate, if someone could enlighten me further.

Mimiix, 2020-08-12:
@tony563 On the HA part i'd like to ask you to check with their discord or forums.

Connection refused probably means you don't have the API used or included. Please reffer to the [RestAPI](https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/)

tony563, 2020-08-12:
> @tony563 On the HA part i'd like to ask you to check with their discord or forums.
> 
> Connection refused probably means you don't have the API used or included. Please reffer to the [RestAPI](https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/)

Thank you. I appreciate your help.

tony563, 2020-08-12:
> I am using the JSON field offset (for temperature) here actively. It works like a charm indeed.

Please explain, how you did that?

tony563, 2020-08-13:
**UPDATE:**
I think, that I have discovered an easier way to do it; by using the "Set States" feature, in the Developer Menu. I'm not sure yet, if it just changes the value temporarily or not.

wvuyk, 2020-08-13:
> > I am using the JSON field offset (for temperature) here actively. It works like a charm indeed.
> 
> Please explain, how you did that?

I set the value of offset to the calibration value needed. From then on the reported value in the rest api temperature shows the calibrated value.

tony563, 2020-08-13:
> I set the value of offset to the calibration value needed. From then on the reported value in the rest api temperature shows the calibrated value.

Thank you; but what command(s) or options did you use, to do it?

wvuyk, 2020-08-14:
I have created a plugin for the Homeseer system here. This plugin sends a PUT command to the REST API's /sensors/<ID> setting the config/offset value. This is then set and saved. Guess you could use a curl command as well.

Mimiix, 2020-08-14:
@wvuyk I did not know that. Are you the maintainer of it? Where is it stored?

wvuyk, 2020-08-15:
I am not a maintainer of it, I just started using it with the REST API when it became available. I m pretty sure @ebaauw knows more on that if I look at #1154

MattWestb, 2020-08-16:
The setting Offset for temperature is working great (putting all devices on the same place for some hours and calculating the avrigh and then the offset for etch one from the avrieh) but NOT user friendly.
In Phoscon App it sud being possible making the setting on every sensor with one easy menu enter offset and OK finish.
The next large problem is luck of setting offset for pressure sensors. I have one (of 6) sensors that have pretty lage diff and its irritating. The next is that i'm now not longer living around seelevel and its not possible compensating for that (metrological and aeronautics you always do that) then the offset for pressure is not exiting. Offset for humidity is normally not needed then the sensor dont have so good tolerance.
I knowing that the Phoscon App is on @YKO-de table but the luck of pressure offset is also one deconz-rest-plugin future.

MattWestb, 2020-09-06:
DE Hollydays ;-)

Mimiix, 2020-09-06:
@MattWestb Why are you bumping this the RestAPI works fine for this? 

@tony563 , did you make it work?

MattWestb, 2020-09-06:
Sorry but for my is not working calibrating offset for pressure in the API.
Can you please providing the right syntax for doing that ?
And its on topics if i have reading it right.

Mimiix, 2020-09-06:
@MattWestb If you have a question, please ask instead of putting "DE Hollydays". That's not beneficial at all 😅.

I found some info on it here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1154  But there is no documentation around the offset as far as i can see.  Let me ask around :)

Mimiix, 2020-09-06:
@MattWestb @tony563  I did a request for documentation : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3237

And bring this up with Manup later on :)

tony563, 2020-09-06:
I just created a template sensor, for each sensor, including the pressure sensor. I was then able to edit that, and build an offset into it.

tony563, 2020-09-06:
Also, if you go into Developer Tool, you can calibrate both temperature and humidity, in the States tab. You can kind of do it for pressure as well, but the moment the pressure sensor updates, it reverts back to the uncalibrate reading.

MattWestb, 2020-09-06:
Thanks @Mimiix that's great. 
I have not looking in the docks for sensor offset for one year but i have saved the procedure for using then i then have getting corrupted DB and must redoing all the settings.
But still not possible (what i knowing) putting offset on pressure in API. 
I need having some hundred meters plus and you perhaps some meter minus recalculated for getting it right in the logs and graphics presentations.

tony563, 2020-09-06:
The big problem, which I have had with sensors, when calibrating them, is that they update at random - when the temperature changes. If you miss that moment, the accuracy of the offset will be dubious.

MattWestb, 2020-09-06:
I was putting 4 new Aqaras at one place with  relative stable temperature for some hours (over the night) and looking for the diffs and putting the offset in and waiting some more hours for validating / finetune the offset.
Was little work but all is in separate locations now and have good precision but as you is saying they is normally not reporting so oft if not (large) sensor measurements is changed.
Sad that is not being implanted in the GUI so the user can making easy.

tony563, 2020-09-06:
I agree. I think there is the ability to enter our set numbers in HASS, via a GUI element. I’m new to HASS, so I’m not sure, if that will do what we would like it to do.

MattWestb, 2020-09-06:
My problem is that i saving the "raw data" for making statistik in InfluxDB from different sources and the best is then the "raw data" is correct calibrated from all sources then its easy displaying in Grafana without doing one more recalculation of the data from the sources.
The best wold being if the sensor itself was doing the calibration and sending "true" calibrated data to the system (I think if i remember right that HUE motion sensor can do offset for most parameters).
HASS is for my more or less one easy presentation tool for checking the last 24 h data is OK and knowing if I need putting long underpants on in the winter and gloves then cleaning the frost from the car windscreen ;-)

tony563, 2020-09-06:
😂😂😂 We don’t seem to need long johns anymore during the Winter.

MattWestb, 2020-09-06:
And perhaps lesser days with umbrellas but you never knowing what mother nature is doing ;-)))

tony563, 2020-09-06:
😂😂😂 At this point, I’m tempted to say something; but perhaps it would be appropriate in this forum. 😱🤭😂😂😂

morfei1, 2020-09-17:
How much the difference is, so you need to calibrate?

MattWestb, 2020-10-11:
EU = ´+ 4000 and - 50 m

SwoopX, 2020-10-11:
Jusg FYI, a rather extensive documentatiln update of mine has been comitted.

Mimiix, 2020-10-12:
And with that, i'll close this issue. Kept bumped open for no apparent reason.

closed 2020-10-12 by Mimiix