dresden-elektronik/deconz-rest-plugin #3133: SegFault in v2.05.79 while importing backup (cross-platform)

SwoopX, 2020-08-12:
## Describe the bug

deCONZ crashes with a SegFault while importing a backup of my test env. The backup came from my Raspbee I Rpi4 and was imported to my Ubuntu 20 LTS with Conbee II. Luckily, I got a core dump, but I'm not sure if this was with the latest beta plugin or a self compiled version.

## Steps to reproduce the behavior

Always reproducable when I import the backup file via Phoscon.
## Expected behavior

Just going with the flow.
## Screenshots

[image]

## Environment

 - Host system: (PC)
 - Running method: (Ubuntu / VMware)
 - Firmware version: (264A0700)
 - deCONZ version: (2.05.79)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (no)

Mimiix, 2020-08-13:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3095

Was reported by Erik aswell. Are they related?

SwoopX, 2020-08-13:
No, that's something different.

ebaauw, 2020-08-13:
No.  #3095 where actually two different, unrelated, issues.  First one fixed by @manup in https://github.com/dresden-elektronik/deconz-rest-plugin/commit/dd58a42044cb68bbe057f4bf7cfe2d6e07150a1e, second one fixed by me in PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3087.

manup, 2020-08-13:
Hmm not sure where the SEGV happens, just looked over the code – not pretty but shouldn't crash.
According to `git gui blame de_web_plugin.cpp` the related parts haven't change in a long time.

I need to make some tests.

Smanar, 2020-08-13:
Perhaps same problem here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2975 but not more finformations on this issue.

SwoopX, 2020-08-13:
@Smanar good catches, those reflect the picture.

It isn't a corrupted DB. I know how backup data is composed and what you can do with it.

The historic network configuration is stored in the DB, iirc.

Smanar, 2020-08-14:
I m sorry for you, but I think it s a good thind it s you that have the bug ^^.

BTW, I m agree, I don't think at a corrupted DB, but perhaps a bad or a missing value. If you have the bug using the hidden config, I think it will be easier to check on this side.
You have the crash for all config, or only the last one ?

SwoopX, 2021-02-20:
The root cause here seems to be that `reconnectTimer` is called before it is actually defined.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/5b8d1a64180b98daa17414671a90af7eef3b7df9/de_web_plugin.cpp#L19371-L19387

Adding a `DBG_Assert(reconnectTimer);` results in the following:

```
15:50:15:816 networkState: DisconnectingNetwork
15:50:15:816 void DeRestPluginPrivate::genericDisconnectNetwork(),19320: assertion 'reconnectTimer' failed
```

closed 2021-02-22 by manup