dresden-elektronik/deconz-rest-plugin #3134: Loratap Tuya Smart Life ZigBee 3.0 Curtain Blind Switch for Roller Shutter Electric motor

lukinno1, 2020-08-12:
Hello this curtain switch have Zigbee protocol. Some support for deconz??

https://www.loratap.com/products/tuya-smart-life-zigbee-30-curtain-blind-switch-for-roller-shutter-electric-motor-google-home-alexa-echo-voice-control-diy

https://www.zemismart.com/products/zigbee-eu-curtain-wall-light-switch-compatible-with-smartthing-hub-echo-plus-app-phone-voice-control

janmo0, 2020-08-13:
I'm also interested in the Loratap device working with deconz. It looks like they are pretty much the same. But I think no one knows until someone connects it with deconz.

The Zemismart should work. At least with the next release I think See:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2999 or https://zigbee.blakadder.com/Zemismart_ZM-CSW002-D.html

Mimiix, 2020-08-13:
@lukinno1 I've changed your request to a User question as this is not a device request.  

We need the device in order to add support for it. Without that we can't do anything 😢 

As @janmo0 mentioned, the Zemismart should have a degree on support in the next beta.

janmo0, 2020-08-13:
> We need the device in order to add support for it. Without that we can't do anything 😢

Ok, I ordered one to investigate. But need to wait at least 1 month for arrival.

> As @janmo0 mentioned, the Zemismart should have a degree on support in the next beta.

You mean only basic functionality like up, down and stop?
Was already wondering how and if the set position is working?!

The only thing I could see in #2999 within the window covering cluster ist the 0x0015 and 0x0016 for 'Acceleration Time - Lift' and 'Deceleration Time - Lift' and maybe the mode on 0x0017 (which is used by the Ubisis Shutter J1 for setting the device into a configuration mode)

Mimiix, 2020-08-13:
@janmo0 Great! Feel free to open a new device request issue when you get it!

@SwoopX might be able to answer this as he created the PR.

SwoopX, 2020-08-15:
> The only thing I could see in #2999 within the window covering cluster ist the 0x0015 and 0x0016 for 'Acceleration Time - Lift' and 'Deceleration Time - Lift' and maybe the mode on 0x0017 (which is used by the Ubisis Shutter J1 for setting the device into a configuration mode)

I don't know what you got that idea. The devices are controlled by the corresponding cluster commands, that's it. You'll probably see the lift percentage in the corresponding attribute then.

janmo0, 2020-08-29:
I got the device today. It's working with Deconz via API or GUI.

[image]
[image]

I'm just wondering how to configure the device. Right now full open or close is around 10 seconds. What if the shade needs more or less to fully close or open?

Smanar, 2020-08-29:
You can use the "lift" field to select the level.
Perhaps in phoscon using the brigtness slider.

cobirnm, 2020-09-05:
Maybe there is a way of entering Calibration Mode using the buttons. Has anyone been able to do it? I have the same issue.

Smanar, 2020-09-05:
Right, I have same reaction for 2 other tuya covering. And one of them need a remote to set calibration.
without calibration the setLevel fonction don't work.

cobirnm, 2020-09-05:
> Right, I have same reaction for 2 other tuya covering. And one of them need a remote to set calibration.
> without calibration the setLevel fonction don't work.

But have you been able to  calibrate any of your devices?

janmo0, 2020-09-05:
I can use the set level function without calibration on the Loratap device. But as default the device send power for 10 seconds. Or if I set to 50% then for 5 seconds.
That should be no problem for only open and close if the blinds close within 10 seconds and the motor has an mechanical or electric endstop.
But the set level function could be wrong then.

As said before, in the Ubisis Shutter J1 the 0x0017 setting is used to put the device into a configuration mode. As I read from the manual.

This is from the Loratap manual:

[image]

[image]

cobirnm, 2020-09-05:
> I can use the set level function without calibration on the Loratap device. But as default the device send power for 10 seconds. Or if I set to 50% then for 5 seconds.
> That should be no problem for [...]

I have tried writing the 0x0017 parameter using VNC conection but I always get "writing failed". Does anyone know how can I achieve this?

Smanar, 2020-09-05:
> 
> 
> > Right, I have same reaction for 2 other tuya covering. And one of them need a remote to set calibration.
> > without calibration the setLevel fonction don't work.
> 
> But have you been able to calibrate any of your devices?

Not me, but yes he do https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3161#issuecomment-687270616

cobirnm, 2020-09-06:
> e to calibrate any of your devices?

In this case the calibration was triggered by a RF remote control. In the case of your switch and mine (I have one branded as Losonho) it seems, by the use manual that you sent me (the one that came with mine doesn't even mention it) that the app ( tuya app) can trigger the calibration, so for sure there is some kind ok command that can be sent to do it.

The  Ubisis Shutter J1 uses the 0x0017 for this, but in Deconz implementation it is created a Sensor as a workaround to trigger this. Maybe someone could change the code in order to also create this sensor for these switches and we could test. (unfortunately I don't know how to do it)

Smanar, 2020-09-06:
On my side I have searched and I haven't see on a zigbee project some information about a "calibration" command.
In all devices, it s always build in the hardware or with a remote.

Do you have a link with the method with the app ? it use their gateway ? and it have only zigbee connexion ?

cobirnm, 2020-09-06:
> On my side I have searched and I haven't see on a zigbee project some information about a zigbee "calibration" command.
> In all devices, it s always build in the hardware or with a 433 mhz remote (and the app)

See the pictures of the manual on the above [pictuture](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3134#issuecomment-687631699). The user manual I received only mentions how to pair but the one @janmo0  received shows how to calibrate in Tuya app. So the zigbee command exists. 

Also when I look at [window_covering.cpp](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/window_covering.cpp) I can see it was created using [ZigBee Home Automation Public Application Profile Document 05-3520-29](http://read.pudn.com/downloads703/ebook/2827955/docs-05-3520-29-00ha-home-automation-profile-specification.pdf). In this document you can see on page 183 that it mentions attribute  0x0017 (sorry if I'm not using the correct naming).  The usage of this attribute 0x0017 has been used for the implementation of the "Ubisys Shutter Control J1". This was done by creating a sensor. Maybe this kind of workaround could be applied to all of cover devices. Unfortunately this is an idea but I don't know how to program it. 

No cover devices will work out of the box without calibrating it. So I believe zigbee aliance has inserted this in their code, and I hope deconz will allow this some day!

Smanar, 2020-09-06:
Ha right.
For my device, I don't think it will work because they are using the tuya cluster but for you why not.
Have you try with deconz ? you can set this value in the GUI, but I m not sure for the value
From actual code
```
            <value name="Reversed" value="0"></value>
            <value name="Calibration Mode" value="1"></value>
            <value name="Maintenance Mode" value="2"></value>
            <value name="LED feedback" value="3"></value>
```

But from file
> * 0x0017 bitmap8,    Mode (bit0=if the motor direction is reversed, bit1=the device is in calibration, bit2=maintenance mode)

So for me the calibration mode will be 0/1/0 = 2

janmo0, 2020-09-07:
> I have tried writing the 0x0017 parameter using VNC conection but I always get "writing failed". Does anyone know how can I achieve this?

Same here 😕

Mimiix, 2020-09-07:
@janmo0  Keep clicking the buttons to keep it awake. Battery devices are sometimes annoying.

janmo0, 2020-09-07:
> @janmo0 Keep clicking the buttons to keep it awake. Battery devices are sometimes annoying.

It's no battery driven device. It's always on main voltage.

Mimiix, 2020-09-07:
Aah, My bad then :( @janmo0

Smanar, 2020-09-07:
I m not sure but if the device don't support this attribute, you have another message ? "Unsupported something" no ?

janmo0, 2020-09-07:
After reading the value is greyed out in the table.

Smanar, 2020-09-07:
ok, so bad luck.
There is an user that have asked for tuya documentation, I will ask to him.

cobirnm, 2020-09-07:
I have found this under [tuya documentation](https://translate.google.pt/translate?sl=auto&tl=en&u=https%3A%2F%2Fdeveloper.tuya.com%2Fen%2Fdocs%2Fiot%2Fdevice-development%2Faccess-standard%2Faccess-standard-zigbee%2Fzigbee-curtain-switch-access-standard%3Fid%3DK9ik6zvra3twv)

Maybe it can help!

Smanar, 2020-09-07:
Ha nice perhaps not usefull for you, but it seem there is an attribute 0xF000 in the windows clovering cluster 

0xF000 > Tuya Private Report Attr > Window Cover Status >0/1/2 > defaut(1)

> 
> a) After the device receives the 0x00 (open) command, it will immediately report the 0 (open) state through Tuya’s private reporting attribute 0xF000, and control the curtain switch relay to keep it open; run to the curtain switch stroke position, stop the curtain switch relay, And report 1 (stop) status.
> 
> b) After the device receives the 0x01 (close) command, it immediately reports the 2 (close) state through Tuya’s private reporting attribute 0xF000, and controls the curtain switch relay to keep it off; it runs to the curtain switch stroke position and stops the curtain switch relay. And report 1 (stop) status.
> 
> c) After the device receives the 0x02 (stop) command, it immediately reports the 1 (stop) state through Tuya's private reporting attribute 0xF000.

janmo0, 2020-09-08:
Issue with same problem on Z2M repo: https://github.com/Koenkk/zigbee2mqtt/issues/4257

lukinno1, 2020-10-20:
> Ha nice, not usefull for you, but it seem there is an attribute 0xF000 in the windows clovering cluster
> 
> 0xF000 > Tuya Private Report Attr > Window Cover Status >0/1/2 > defaut(1)
> 
> > a) After the device receives the 0x00 (open) command, it will immediately report the 0 (open) state through Tuya’s private reporting attribute 0xF000, and control the curtain switch relay to keep it open; run to the curtain switch stroke position, stop the curtain switch relay, And report 1 (stop) status.
> > b) After the device receives the 0x01 (close) command, it immediately reports the 2 (close) state through Tuya’s private reporting attribute 0xF000, and controls the curtain switch relay to keep it off; it runs to the curtain switch stroke position and stops the curtain switch relay. And report 1 (stop) status.
> > c) After the device receives the 0x02 (stop) command, it immediately reports the 1 (stop) state through Tuya's private reporting attribute 0xF000.

Hello

Why my device in conbee is lost/ not working..i see this device like heiman but in vnc i can readed cluster only one. Then like not enable. 

Did you have the same problem?

janmo0, 2020-10-20:
I think the guys @ Z2M made some progress with the calibration mode issue. Anybody here can check if this is usable for deconz?

https://github.com/Koenkk/zigbee2mqtt/issues/4257#issuecomment-711031302

Smanar, 2020-10-20:
Ha yep probably

>(0x0102) | Write Attributes (0x02) | 0xf001 | 8-Bit (0x30) | 0 (0x00) | Start Calibration
(0x0102) | Write Attributes (0x02) | 0xf001 | 8-Bit (0x30) | 1 (0x01) | End Calibration

Will try to add thoses command to api to test.

Smanar, 2020-10-21:
To test you try to edit the xml file > https://github.com/Smanar/deconz-rest-plugin/commit/85a76c2ed9668121c1b7f73a82ae586945039883

Then restart deconz, you will have new option in cluster 0x0102.

janmo0, 2020-10-23:
> To test you can try to edit the xml file > [Smanar@85a76c2](https://github.com/Smanar/deconz-rest-plugin/commit/85a76c2ed9668121c1b7f73a82ae586945039883)
> 
> Then restart deconz, you will have new option in cluster 0x0102.
> 
> But I m reading the z2m issue, it seem you still need a switch for that ? and the timer is memorised on the switch.

It's working!
The device is a switch only. Used for controlling rollershades.

Would it be possible to get the light mode setting also working?

> (0x0006) | Write Attributes (0x02) | 0x8001 | 8-Bit (0x30) | 0 (0x00) | Light Mode  1
(0x0006) | Write Attributes (0x02) | 0x8001 | 8-Bit (0x30) | 1 (0x01) | Light Mode  2
(0x0006) | Write Attributes (0x02) | 0x8001 | 8-Bit (0x30) | 2 (0x02) | Light Mode  3

Will the settings be later available in the REST API too?

Smanar, 2020-10-23:
I will add it on deconz GUI, but what is "light mode " on a shutter ?

`Will the settings be later available in the REST API too?`

IDK, the calibration is only needed 1 time, not sure It s usefull to make a command usable by third app for that.

Smanar, 2020-10-24:
And BTW you have a cluster 0x0f41 in deconz ? on the GUI ?

janmo0, 2020-10-26:
> I will add it on deconz GUI, but what is "light mode " on a shutter ?

It is used for the button lighting. I think it's for "always on", "always off" and "only at touch" I think.
 
> IDK, the calibration is only needed 1 time, not sure It s usefull to make a command usable by third app for that.

Ok. Just tought it would be added because the Ubisys J1 device has the calibration available via API: #813

> And BTW you have a cluster 0x0f41 in deconz ? on the GUI ?

I'm running 2.05.78 and don't have this cluster. Or at least. I didn't found.

Smanar, 2020-10-26:
Ok so have added the light mode in the XML.
Have added the calibration command, it s an hidden command, to use it `PUT, PATCH /api/<apikey>/lights/<id> with {"calibration":true}`

Lymfo, 2020-11-23:
> To test you can try to edit the xml file > [Smanar@85a76c2](https://github.com/Smanar/deconz-rest-plugin/commit/85a76c2ed9668121c1b7f73a82ae586945039883)
>
> Then restart deconz, you will have new [...]

Sorry, guys I'm quite new at this. 

I bought this device and got into Deconz, but now I want to calibrate this. But I'm stuck add adding it into the xml file. How do I do that? 

Your help is highly appreciated.

Smanar, 2020-11-23:
You can install the last deconz version ( Beta 2.6.0 ) ?
Else you can just use a file editor the file is in /usr/share/deCONZ/zcl/general.xml

Lymfo, 2020-11-23:
Thank you for your quick response.

it's the stable version (2.05.88).

But if I understand you correct I can just add the extra lines in the general xml file? No specific position is required for the lines in the file?

Outlook voor Android downloaden<https://aka.ms/ghei36>

cobirnm, 2020-11-24:
@Smanar I have the attribute but it's read only. I'm Unable to change it. What am I doing wrong?

[image]

Smanar, 2020-11-24:
Good catch, my bad

Try with
`access="rw"`

For read and write.

Mimiix, 2020-12-30:
@Smanar what's the status on this? Reopening this until calibration is fixed.

Smanar, 2020-12-30:
It's ok, corrected, alreadu corrected and in official https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/general.xml#L1609

arodriguezf, 2020-12-31:
I´m not sure how to calibrate the switch, can someone help me? a list of steps will be ok

robert4y, 2021-01-17:
Where can I find Attribute Editor ?

Smanar, 2021-01-17:
IDK if it s that you are searching, but on the GUI, select a cluster, and you will have all attributes for this cluster on the cluster panel.

robert4y, 2021-01-18:
Calibration works fine by doing as per indicated in this thread
But....this is my issue:
When I push/touch the “up” button, the curtain goes up and after the configured time the curtain stops but then the “down” button goes into disable stage and the “up” button remains enable. Shouldn’t it be vice versa?. At this point, it is not possible to move the curtain down anymore.
Motor reversal does not help.

Smanar, 2021-01-18:
Ha ^^, it s in your third app ?

And yes I know this issue, a direction problem. when you press "up" the device go "up" but it send a "down" notification, so the third app disable the "down" button.

It can be a hardware problem (same motor are natively reversed) but as you are the only one with this problem yet, I think the problem is more on your side.
Are you sure for the wiring, and the calibration ?
If yes you can try the command to reverse the direction, you have it close the "calibration" in the GUI.

robert4y, 2021-01-18:
The connection is correct, the 433 mhz driver was working there earlier.
"You are speaking about the zigbee command, or the hardware wiring ?"
So I mean the zigbee command.
In the post about zigbee2mqtt, they recommend adding invert_cover: true but I don't know where to enter it in the case of deconz.

Smanar, 2021-01-18:
>So I mean the zigbee command.

So even with this setting on or off you have the same problem ?

Can you hsow the JSON you have for this device ?

No-one else have this problem for this device ?

robert4y, 2021-01-18:
Where can I find the json of this device?

Smanar, 2021-01-18:
On the app you are using with deconz ?
Else using the API.

For IP and PORT you can use exactly the same you are using for phosocn.

First you need an API key, if you haven't one :
`curl http://IP:PORT/api/ -X POST -d '{"devicetype": "TestKey"}'`

If you have an error message about autorisation or autentification, go in phoson/gateway/advanced/authenticate app and remake the command

Then to take a look on your light device
http://IP:PORT/api/KEY/lights

Don't worry, no sensibles informations on thoses data.

lukinno1, 2021-01-18:
> Calibration works fine by doing as per indicated in this thread
> But....this is my issue:
> When I push/touch the “up” button, the curtain goes up and after the configured time the curtain stops but [...]

I had problem with this in HA , but in NODE RED works fine. 
I think sdandard solution is 0% open and 100% close,
but with this Loratap is it opposite. You can range 1-99% and will be work fine :)

closed 2021-03-16 by github-actions[bot]

Jasonthefirst, 2021-05-29:
I tried to calibrate my motor as well but the calibration topic only returns "writing failed". If i try to read the state it returns: "unsupported attribute". 
I have no way of calibrating the motor which means it will only open half. Then I have no way of moving it any further down except turning off the power and pressing down again.
Is there anything I can do or try? 
My current version: 2.11.05 / 9.5.2021 on firmware: 26680700.

Smanar, 2021-05-29:
Perhaps https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4597

navterro, 2021-05-30:
I have the same issue when trying to put module into calibrate mode: "writing failed"and then state "unsupported attribute".
The same happens when I hit "read" button for calibration attribute.
The only attribute that i can read and write is "Motor Reversal".

I also tried (PUT and PATCH) REST API method: 
Response:
[
    {
        "success": {
            "/lights/4/calibration": true
        }
    }
]

But motor is still stopping after 10 seconds so its not in calibration mode. 

My Device: 

    "4": {
        "etag": "d881dce3e5fcc86fb5b14b0e11c0c1bc",
        "hascolor": false,
        "lastannounced": "2021-05-29T20:16:08Z",
        "lastseen": "2021-05-30T18:21Z",
        "manufacturername": "_TZ3000_1dd0d5yi",
        "modelid": "TS130F",
        "name": "Screen",
        "state": {
            "bri": 0,
            "lift": 0,
            "on": false,
            "open": true,
            "reachable": true
        },
        "swversion": null,
        "type": "Window covering controller",
        "uniqueid": "60:a4:23:ff:fe:6f:05:61-01"
    }

My configuration:
Home assistant supervised installation.
DeConz add-on v. 6.9.0 ( 2.11.5) 
ConBee II firmware: 0x26660700

Any suggestions will be much appreciated :)

Smanar, 2021-05-31:
And same, nothing on manual ?
Some device need a specail remote, other have special procedure with buttons, and others can be set with the application using the gateway.

Jasonthefirst, 2021-05-31:
There is sadly nothing in the manual. Or well: "use the corresponding hub". I read the two issues you mentioned but still couldn't figure it out #4597 won't work because i can't write the calibration node. #3134 might work but I don't understand how to write the values over the deconz gui? Is there a manual input for values? I try to use the cluster info and write the values there but calibration is greyed out after i try to read it.

Smanar, 2021-05-31:
Both of them are grayed ?

Cluster Covering, Attribute 0xf001
Cluster Covering, Attribute 0x001

your device is not on battery ?

navterro, 2021-05-31:
This is MOES (Tuya) Zigbee+ RF Smart Curtain Switch Module - there is no info regarding calibration in Manual.
This module can also work with some special RF remote or zigbee gateway.

Window Covering Cluster: 

**Attribute 0xf001** (Calibration) -  grayed after read/write operation
**Attribute 0x0017** (Mode) - value 0x00:
- when i select "calibration" and click write - i get "writing failed"
- when i click read - i get "unsupported attribute" 

**0x0017** is not grayed after read/write try.

Motor is for sure calibrated correctly - was working with wall switch before adding this zigbee module.

I can have screen fully open / closed with this workaround:

To Open:
1.  Exec Down / Close (Screen moving DOWN for 10 seconds)
2.  **0xF002** change from off->on
3. Exec Up / Open (Screen moving DOWN for 10 seconds)
4.  **0xF002** change from on->off
5. Exec Down / Close (Screen moving DOWN for 10 seconds)

So it takes about 30 seconds for motor to move screeen from fully closed to fully open

[31 more replies]